zephyrproject-rtos/zephyr #51509: Bluetooth: controller: workaround to use DLE activation with multiple connections

lanwer50, 2022-10-21:
Is your enhancement proposal related to a problem? Please describe.
A clear and concise description of what the problem is.

I am opening this issue to further discuss a workaround here instead of on discord, so that the information is available for others. 

As workaround before the issue #21995 is closed, I want to use the DLE with multiple connections(20) in parallel using zephyr v3.2.0. 

Describe the solution you'd like
A clear and concise description of what you want to happen.

A suggestion from @cvinayak  on discord was to set the max tx and rx time: https://github.com/zephyrproject-rtos/zephyr/blob/183370277b41d1f92fa56fd9c6c56ee864145977/subsys/bluetooth/controller/ll_sw/ull_central.c#L238-L239. 

In Order to do that I had to activate `CONFIG_BT_LL_SW_LLCP_LEGACY=y` in prj.conf. Any problems to expect with this config?
I think it also makes sense to edit the max number tx and rx octes to `PDU_DC_PAYLOAD_SIZE_MAX`. 

So that part of the code looks as follows now: @cvinayak : any comments on this?

`#if defined(CONFIG_BT_LL_SW_LLCP_LEGACY)
#if defined(CONFIG_BT_CTLR_DATA_LENGTH)
	conn_lll->max_tx_octets = PDU_DC_PAYLOAD_SIZE_MAX;
	conn_lll->max_rx_octets = PDU_DC_PAYLOAD_SIZE_MAX;

#if defined(CONFIG_BT_CTLR_PHY)
	/* Use the default 1M packet Tx time, extended connection initiation
	 * in LLL will update this with the correct PHY.
	 */
	conn_lll->max_tx_time = PDU_DC_PAYLOAD_TIME_MAX;
	conn_lll->max_rx_time = PDU_DC_PAYLOAD_TIME_MAX;

#endif /* CONFIG_BT_CTLR_PHY */
#endif /* CONFIG_BT_CTLR_DATA_LENGTH */
#else /* CONFIG_BT_LL_SW_LLCP_LEGACY */`

Describe alternatives you've considered
A clear and concise description of any alternative solutions or features you've considered.

Additional context
Add any other context or graphics (drag-and-drop an image) about the feature request here.

lanwer50, 2022-10-24:
Along with using lower priorities for other SoC peripherals (SERIAL) by adding the line below
`#define NRF_DEFAULT_IRQ_PRIORITY 5` inside the application folder in dts/arm/nordic/override.dtsi file,  I am facing on the central side a few Asserts every few hours (2-24hrs). 

* ASSERTION FAIL [(ticker_status == 0) || (ticker_status == 2)] @ WEST_TOPDIR/zephyr/subsys/bluetooth/controller/ll_sw/ull_central.c:1044

* ASSERTION FAIL [e] @ WEST_TOPDIR/zephyr/subsys/bluetooth/controller/ll_sw/nordic/lll/lll_conn.c:955

**Application description:**
The central has 20s time to connect to as many as 20 peripherals and pull data from them. Then the central disconnects randomly a few of the peripherals. 

**Connection parameters:**
#define SCAN_INTERVAL 0x00d0 /* 130 ms */  
#define SCAN_WINDOW   0x0010 /* 10 ms */
#define INIT_INTERVAL 0x0010 /* 10 ms */
#define INIT_WINDOW   0x0010 /* 10 ms */
#define CONN_INTERVAL 0x0068 /* 130 ms */ 
#define CONN_LATENCY  0
#define CONN_TIMEOUT  400

Any idea what might be the reason?

cvinayak, 2022-10-24:
Please send a draft pull request with your changes to controller code to better understand the consequences of the changes. The suggested workaround is to only change the time reservation for the connections, not to change the parameters (max octets or max time) in the connection context structures.

Add:
```c
	max_tx_time = PDU_DC_PAYLOAD_TIME_MAX; /* 2120 us */
	max_rx_time = PDU_DC_PAYLOAD_TIME_MAX; /* 2120 us */
``` 

before:
https://github.com/zephyrproject-rtos/zephyr/blob/c022dd7756209b9ea20228757829ea13b362f7ee/subsys/bluetooth/controller/ll_sw/ull_central.c#L472-L477

lanwer-en, 2022-10-26:
I created a draft pull request as you suggested (https://github.com/zephyrproject-rtos/zephyr/pull/51648) and removed the previous changes (changing the parameters (max octets or max time) in the connection context structures). 

I will check if only changing the time reservation for the connections would help and I will let you know either way. But for now: could you take a look at the draft PR to check/understand the consequences of the changes?

lanwer-en, 2022-10-27:
@cvinayak : I unfortunately got the following Assert after 34 hours: 
ASSERTION FAIL @ WEST_TOPDIR/zephyr/subsys/bluetooth/host/att.c:794

I have 2 questions:
* Should I try to use CONFIG_BT_LL_SW_LLCP_LEGACY=y. I have no idea if that would help, I have just saw your issue where this config might help https://github.com/zephyrproject-rtos/zephyr/issues/51693

* The changes to the controller code to use DLE with multiple connections are only needed on the central side, right? Or should the peripheral also have these changes.

cvinayak, 2022-10-27:
> I unfortunately got the following Assert after 34 hours:
> ASSERTION FAIL @ WEST_TOPDIR/zephyr/subsys/bluetooth/host/att.c:794

Please create a new github issue for this.

> * Should I try to use CONFIG_BT_LL_SW_LLCP_LEGACY=y. I have no idea if that would help, I have just saw your issue where this config might help #51693

No, #51693 does not occur on target as the h/w radio peripheral truncates the PDU on reception and does not cause any invalid behaviors. 

> * The changes to the controller code to use DLE with multiple connections are only needed on the central side, right? Or should the peripheral also have these changes.

Yes, central tries to scheduling each of n connection with the reserved spacing between them to form a group in the timeline.  Peripherals having only one connection do not need the workaround, the connection event are connection interval apart anyway.

carlescufi, 2022-10-27:
> @cvinayak : I unfortunately got the following Assert after 34 hours: ASSERTION FAIL @ WEST_TOPDIR/zephyr/subsys/bluetooth/host/att.c:794

Can you please create a separate issue (bug report) for this?

lanwer-en, 2022-10-27:
@carlescufi: w

> > @cvinayak : I unfortunately got the following Assert after 34 hours: ASSERTION FAIL @ WEST_TOPDIR/zephyr/subsys/bluetooth/host/att.c:794
> 
> Can you please create a separate issue (bug report) for this? Before that I suggest you bump stack sizes before posting it just to make sure it's not a stack overflow.

will do that. Do you have any particular stack sizes in mind? (20 connections are needed)

carlescufi, 2022-12-05:
> 

@lanwer-en not really, no. Just play around with numbers.

cvinayak, 2025-02-12:
I believe #52012 should have covered this enhancement request.

closed 2025-02-12 by cvinayak