zephyrproject-rtos/zephyr #51521: subsys: blueutooth: shell: gatt.c build fails with CONFIG_DEBUG_OPTIMIZATIONS and CONFIG_BT_SHELL=y

eSTChristopheU, 2022-10-21:
**Describe the bug**
My project uses BLE and CONFIG_BT_SHELL=y.
The build fails when CONFIG_DEBUG_OPTIMIZATIONS=y is selected 
project build succesfull when CONFIG_SIZE_OPTIMIZATIONS=y or CONFIG_SPEED_OPTIMIZATIONS=y or CONFIG_NO_OPTIMIZATIONS=y 

The issue is located into the cmd_write_without_rsp() function. 
int err; should be initialised to 0 per exemple to remove the error. All warnings being treated as errors in my config

**To Reproduce**
CONFIG_BT=y
CONFIG_BT_SHELL=y
CONFIG_DEBUG_OPTIMIZATIONS=y

**Expected behavior**
successful build without error

**Impact**
I can't use the CONFIG_DEBUG_OPTIMIZATIONS when i activate the bluetooth shell.

**Logs and console output**

```
In file included from /home/chris/lpp4_hardware_workspace/platform/zephyr/subsys/bluetooth/shell/gatt.c:24:
/home/chris/lpp4_hardware_workspace/platform/zephyr/subsys/bluetooth/shell/gatt.c: In function 'cmd_write_without_rsp':
/home/chris/lpp4_hardware_workspace/platform/zephyr/include/zephyr/shell/shell.h:1056:2: error: 'err' may be used uninitialized in this function [-Werror=maybe-uninitialized]
 1056 |  shell_fprintf(_sh, SHELL_NORMAL, _ft "\n", ##__VA_ARGS__)
      |  ^~~~~~~~~~~~~
/home/chris/lpp4_hardware_workspace/platform/zephyr/subsys/bluetooth/shell/gatt.c:466:6: note: 'err' was declared here
  466 |  int err;
      |      ^~~
cc1: all warnings being treated as errors
[79/92] Building C object zephyr/kernel/CMakeFiles/kernel.dir/mempool.c.obj
ninja: build stopped: subcommand failed.
FATAL ERROR: re-build in /home/chris/lpp4_hardware_workspace/lpp4_hardware/build failed (no --build-dir given)

```

**Environment (please complete the following information):**
-- Zephyr version: 3.1.99 (/home/chris/lpp4_hardware_workspace/platform/zephyr)
-- Found west (found suitable version "0.13.1", minimum required is "0.7.1")
-- Found host-tools: zephyr 0.13.1 (/opt/zephyr-sdk-0.13.1)
-- Found dtc: /opt/zephyr-sdk-0.13.1/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6")
-- Found toolchain: zephyr 0.13.1 (/opt/zephyr-sdk-0.13.1)

**Additional context**

CONFIG_SHELL_ARGC_MAX=255
CONFIG_ARM_MPU=y
CONFIG_DEBUG_OPTIMIZATIONS=y
CONFIG_SHELL_HARDWARE=y
CONFIG_LOG=y
CONFIG_LOG_MODE_IMMEDIATE=y
CONFIG_SHELL_STACK_SIZE=4096
CONFIG_BT=y
CONFIG_BT_SHELL=y
CONFIG_BT_DEBUG_LOG=y
CONFIG_BT_DEBUG_HCI_CORE=n
CONFIG_BT_DEBUG_HCI_DRIVER=n
CONFIG_BT_PERIPHERAL=y
CONFIG_BT_GATT_CLIENT=y

jori-nordic, 2022-10-26:
I don't have any warnings. I guess you have some more warnings enabled in your flags or something.
But I think I see the issue, the compiler for some reason doesn't understand that `err` will always be set with
```c

	if (!repeat) {
		repeat = 1U;
	}
```
and that the `while(repeat--)` loop will run at least once, setting `err`.

But the line numbers don't match, so could you give me the git revision of your zephyr repo so I can double-check?

jori-nordic, 2022-10-26:
yep got it. so my analysis was correct, the compiler is maybe optimizing that `while(repeat--)` somehow, and that ends up in a situation where it thinks that loop can be skipped (it shouldn't).
basic fix here https://github.com/zephyrproject-rtos/zephyr/pull/51658

commit 4b1585c2 (2022-10-27): Bluetooth: shell: fix compiler warning

Compiler thinks `err` is not initialized when
`CONFIG_DEBUG_OPTIMIZATIONS=y`.

Fixes #51521 .