zephyrproject-rtos/zephyr #51528: spurious warnings when EXTRA_CFLAGS=-save-temps=obj is passed

iperry, 2022-10-21:
When `EXTRA_CFLAGS=-save-temps=obj` is passed, many spurious warnings start showing up, usually something like:
```
/zephyrproject/zephyr/lib/os/cbprintf_packaged.c: In function 'cbprintf_package_convert':
/zephyrproject/zephyr/lib/os/cbprintf_packaged.c:1003:2969: warning: the comparison will always evaluate as 'true' for the address of 'data' will never be NULL [-Waddress]
 1003 |                                 LOG_WRN("(unsigned) char * used for %%p argument. "
```

Please also mention any information which could help others to understand
the problem you're facing:
 - This showed up when I updated to v3.2.0. Was not present in v3.1.0.

**To Reproduce**
Steps to reproduce the behavior:
1. `west build --pristine -b arduino_due zephyr/samples/hello_world -- -DEXTRA_CFLAGS=-save-temps=obj`

**Expected behavior**
No warnings or errors.

**Impact**
Log and compile noise.

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
 - OS: Arch linux
 - Zephyr SDK 0.15.1
 - tag v3.2.0

mbolivar-nordic, 2022-10-24:
I believe it's medium because enabling this option is how you debug macro problems.

yashi, 2023-01-02:
One of the messages is from the line below in `zephyr/sys/cbprintf_internal.h`.

```
size_t _pmax = (buf != NULL) ? _inlen : INT32_MAX; \
```

As the warning says buf is a fixed memory location, `data`, and not a pointer.

This kills it, but I'm not sure this is a right way to fix it.
```
diff --git a/include/zephyr/sys/cbprintf_internal.h b/include/zephyr/sys/cbprintf_internal.h
index 5334bef1937..af5302f1f1e 100644
--- a/include/zephyr/sys/cbprintf_internal.h
+++ b/include/zephyr/sys/cbprintf_internal.h
@@ -402,6 +402,7 @@ do { \
 do { \
        _Pragma("GCC diagnostic push") \
        _Pragma("GCC diagnostic ignored \"-Wpointer-arith\"") \
+       _Pragma("GCC diagnostic ignored \"-Waddress\"") \
        Z_CBPRINTF_SUPPRESS_SIZEOF_ARRAY_DECAY \
        BUILD_ASSERT(!IS_ENABLED(CONFIG_XTENSA) || \
                     (IS_ENABLED(CONFIG_XTENSA) && \
```

yashi, 2023-01-02:
This should always work.
```diff
diff --git a/include/zephyr/sys/cbprintf_internal.h b/include/zephyr/sys/cbprintf_internal.h
index 5334bef1937..a19946b238d 100644
--- a/include/zephyr/sys/cbprintf_internal.h
+++ b/include/zephyr/sys/cbprintf_internal.h
@@ -382,6 +382,11 @@ do { \
                 _name##_buf32)))
 #endif
 
+static inline bool ___is_null(void *p)
+{
+       return p == NULL;
+}
+
 /** @brief Statically package a formatted string with arguments.
  *
  * @param buf buffer. If null then only length is calculated.
@@ -430,7 +435,7 @@ do { \
        Z_CBPRINTF_ON_STACK_ALLOC(_ros_pos_buf, _ros_cnt); \
        uint8_t *_rws_buffer; \
        Z_CBPRINTF_ON_STACK_ALLOC(_rws_buffer, 2 * _rws_cnt); \
-       size_t _pmax = (buf != NULL) ? _inlen : INT32_MAX; \
+       size_t _pmax = ___is_null(buf) ? _inlen : INT32_MAX; \
        int _pkg_len = 0; \
        int _total_len = 0; \
        int _pkg_offset = _align_offset; \
```

`___is_null()` should be somewhere else, I suppose.

henrikbrixandersen, 2023-01-27:
@yashi Could you please submit a PR with the above fix? It seems reasonable.

yashi, 2023-01-31:
Ok.  Give me a few days.

closed 2023-02-06 by carlescufi