zephyrproject-rtos/zephyr #51529: frdm_k64f: tests/net/socket/tls run failed on frdm_k64f

mengxianglinx, 2022-10-22:
**Describe the bug**
tests/net/socket/tls run failed on frdm_k64f

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b frdm_k64f tests/net/socket/tls
2. west flash

**Expected behavior**
test cases run passed

**Logs and console output**

*** Booting Zephyr OS build zephyr-v3.2.0-818-g938b9533c727  ***
Running TESTSUITE net_socket_tls
\===================================================================
START - test_so_protocol
 PASS - test_so_protocol in 1.002 seconds
\===================================================================
START - test_so_type
 PASS - test_so_type in 1.002 seconds
\===================================================================
START - test_v4_dtls_sendmsg

    Assertion failed at WEST_TOPDIR/zephyr/tests/net/socket/tls/src/main.c:530: test_dtls_sendmsg: (rv not equal to 0)
server bind failed
 FAIL - test_v4_dtls_sendmsg in 0.013 seconds
\===================================================================
START - test_v4_msg_trunc

    Assertion failed at WEST_TOPDIR/zephyr/tests/net/socket/tls/src/main.c:421: test_msg_trunc: (rv not equal to 0)
server bind failed
 FAIL - test_v4_msg_trunc in 0.013 seconds
\===================================================================
START - test_v4_msg_waitall

    Assertion failed at WEST_TOPDIR/zephyr/tests/net/socket/tls/src/../../socket_helpers.h:135: prepare_sock_tls_v4: (*sock >= 0 is false)
socket open failed
 FAIL - test_v4_msg_waitall in 0.015 seconds
\===================================================================
START - test_v6_dtls_sendmsg

    Assertion failed at WEST_TOPDIR/zephyr/tests/net/socket/tls/src/../../socket_helpers.h:192: prepare_sock_dtls_v6: (*sock >= 0 is false)
socket open failed
 FAIL - test_v6_dtls_sendmsg in 0.015 seconds
\===================================================================
START - test_v6_msg_trunc

    Assertion failed at WEST_TOPDIR/zephyr/tests/net/socket/tls/src/../../socket_helpers.h:192: prepare_sock_dtls_v6: (*sock >= 0 is false)
socket open failed
 FAIL - test_v6_msg_trunc in 0.015 seconds
\===================================================================
START - test_v6_msg_waitall

    Assertion failed at WEST_TOPDIR/zephyr/tests/net/socket/tls/src/../../socket_helpers.h:154: prepare_sock_tls_v6: (*sock >= 0 is false)
socket open failed
 FAIL - test_v6_msg_waitall in 0.015 seconds
\===================================================================
TESTSUITE net_socket_tls failed.

------ TESTSUITE SUMMARY START ------

SUITE FAIL -  25.00% [net_socket_tls]: pass = 2, fail = 6, skip = 0, total = 8 duration = 2.090 seconds
 - PASS - [net_socket_tls.test_so_protocol] duration = 1.002 seconds
 - PASS - [net_socket_tls.test_so_type] duration = 1.002 seconds
 - FAIL - [net_socket_tls.test_v4_dtls_sendmsg] duration = 0.013 seconds
 - FAIL - [net_socket_tls.test_v4_msg_trunc] duration = 0.013 seconds
 - FAIL - [net_socket_tls.test_v4_msg_waitall] duration = 0.015 seconds
 - FAIL - [net_socket_tls.test_v6_dtls_sendmsg] duration = 0.015 seconds
 - FAIL - [net_socket_tls.test_v6_msg_trunc] duration = 0.015 seconds
 - FAIL - [net_socket_tls.test_v6_msg_waitall] duration = 0.015 seconds

------ TESTSUITE SUMMARY END ------

\===================================================================
PROJECT EXECUTION FAILED

**Environment (please complete the following information):**
 - OS: (Ubuntu Linux)
 - Toolchain (Zephyr SDK 0.15.0)
 - Commit SHA or Version used: 938b9533c727d05e48666decc2c1006b8c34009c

mengxianglinx, 2022-10-24:
git bisect to b279bfd2ddea4ef39bf7844f507feaf554987937.
In drviers/ethernet/eth_mcux.c, set the carrrier state to OFF by default instead of setting the NET_IF_NO_AUTO_START doesn't work well. 
I run this test case on frdm_k64f with ethernet cable connected, failed. Are there any problem with "carrier detection"？

FRASTM, 2022-10-24:
Similar issue found on nucleo_f746zg and stm32_f429zi and stm32h743zi when running 
west build -p auto -b nucleo_f746zg tests/net/socket/ any testcase fails in the same manner:
```
E: HAL_ETH_Init Timed out
*** Booting Zephyr OS build zephyr-v3.2.0-806-g64ea7f61ba2b  ***
Running TESTSUITE net_socket_poll
===================================================================
START - test_poll

    Assertion failed at WEST_TOPDIR/zephyr/tests/net/socket/poll/src/main.c:55: net_socket_poll_test_poll: (res not equal to 0)
bind failed
 FAIL - test_poll in 0.013 seconds
===================================================================
START - test_pollout_tcp

    Assertion failed at WEST_TOPDIR/zephyr/tests/net/socket/poll/src/main.c:197: net_socket_poll_test_pollout_tcp: (res not equal to 0)

 FAIL - test_pollout_tcp in 0.013 seconds
===================================================================
TESTSUITE net_socket_poll failed.

------ TESTSUITE SUMMARY START ------                                                                                                                         
                                                                                                                                                              
SUITE FAIL -   0.00% [net_socket_poll]: pass = 0, fail = 2, skip = 0, total = 2 duration = 0.026 seconds                                                      
 - FAIL - [net_socket_poll.test_poll] duration = 0.013 seconds                                                                                                
 - FAIL - [net_socket_poll.test_pollout_tcp] duration = 0.013 seconds                                                                                         
                                                                                                                                                              
------ TESTSUITE SUMMARY END ------                                                                                                                           
                                                                                                                                                              
===================================================================                                                                                           
PROJECT EXECUTION FAILED 
```

rlubos, 2022-10-25:
> In drviers/ethernet/eth_mcux.c, set the carrrier state to OFF by default instead of setting the NET_IF_NO_AUTO_START doesn't work well.
I run this test case on frdm_k64f with ethernet cable connected, failed. Are there any problem with "carrier detection"？

Carrier detection works fine, the problem here is that some of the tests make use of the `CONFIG_NET_CONFIG_SETTINGS` feature to configure the interface, which IMO is conceptually wrong for the tests.

The test cases we have in Zephyr work either with loopback interface, or dummy interfaces created at the test-suite level. On the other case `CONFIG_NET_CONFIG_SETTINGS` intends to configure **any** network interface with the defaults. This means there's no guarantee the loopback interface will be the one configured. It used to work so far with real hardware, as the `net_config_init_app()` exploited the NET_IF_NO_AUTO_START flag set on certain Ethernet interfaces (https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/net/lib/config/init.c#L420), which basically means such interfaces were ignored. For emulated platforms, it only works because they disable the "real" network interface if `CONFIG_NET_TEST` is enabled.

Given that `CONFIG_NET_CONFIG_SETTINGS` is not a reliable source for interface configuration (it's fine for samples where you typically have a single interface, but no good if there are more interfaces) I think it's use should be banned from the test cases. Instead, the tests should either configure specific interfaces manually, or rely on the default loopback address.

rlubos, 2022-10-25:
@mengxianglinx @FRASTM I've opened https://github.com/zephyrproject-rtos/zephyr/pull/51613, please let me know if this helps in your cases (I've tested with `nucleo_h723zg` where I could reproduce the issue you describe).

I'd also like to know if there are any other test suites that fail for you when running on real hardware an which I might've missed in my PR>

mengxianglinx, 2022-10-26:
Thank you @rlubos , I run test cases in tests/net with #51613 on frdm_k64f, the failed cases passed, no new error occur.

closed 2022-10-26 by carlescufi