zephyrproject-rtos/zephyr #51544: drivers/pwm/pwm_sam.c - update period and duty cycle issue (workaround + suggestions for fix)

tswaehn, 2022-10-23:
Hi, 
I guess I figured out an issue in that file [here](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/pwm/pwm_sam.c#L68):

just for the record duplicated:
```
static int sam_pwm_set_cycles(const struct device *dev, uint32_t channel,
			      uint32_t period_cycles, uint32_t pulse_cycles,
			      pwm_flags_t flags)
{
   ...

	/* Update period and pulse using the update registers, so that the
	 * change is triggered at the next PWM period.
	 */
	pwm->PWM_CH_NUM[channel].PWM_CPRDUPD = period_cycles;
	pwm->PWM_CH_NUM[channel].PWM_CDTYUPD = pulse_cycles;

	/* Enable the output */
	pwm->PWM_ENA = 1 << channel;
   ...
}
```
I used an oscilloscope to see that period and duty cycle are **NOT** updated here in that function. After exit of that function the registers
`PWM_CPRD` and `PWM_CPRD` are still at `0x0`.

I did observe - the registers `PWM_CPRDUPD` and `PWM_CDTYUPD` seem to work only **AFTER** enabling the pwm through `PWM_ENA`. 

**If someone could confirm this behavior - would be great!**

this fixes for me:
* enable `PWM_ENA = 1` **BEFORE** setting  `PWM_CPRDUPD` and `PWM_CDTYUPD`

workaround:
* call `sam_pwm_set_cycles()` twice with the same parameters 

I could fork and create a fix + pull-request if needed. please let me know.

nandojve, 2022-10-24:
Hi @tswaehn ,

Please, open a PR. Maybe it will be necessary check if the PWM_ENA is 0. If that is the case, update PWM_CDTY and PWM_CPRD before enable the PWM. On the next `sam_pwm_set_cycles()` the PWM_CPRDUPD and PWM_CDTYUPD may work. This is my guess!

pdgendt, 2022-10-26:
I bumped into this issue after I submitted my PR (#51626), which introduces polarity.
I discovered the same behavior, and my PR should fix the issue as it will use the `PWM_CDTY` and `PWM_CPRD` registers when setting the PWM for the first time.

tswaehn, 2022-10-30:
@pdgendt from the code - it should do the trick. thanks.

nandojve, 2022-10-30:
Hi @tswaehn ,

Yes, when PR be merged it automatically will close the issue.

commit 8254be21 (2022-11-02): drivers: pwm: pwm_sam: Add inverted polarity support

Set CPOL bit in Channel Mode Register if the PWM polarity is inverted.

Fixes #51544
closed 2022-11-02 by carlescufi