zephyrproject-rtos/zephyr #51566: broken network once lwM2M is resumed after pause

chmielewskiandreas, 2022-10-24:
As already discussed with @jheiskan81 via mail I want to report a bug that I was observing when trying to use LwM2M pause and resume using non-dtls. When I use dtls pause/resume works fine in ma application.

```
[00:00:03.676,116] <dbg> net_lwm2m_message_handling: lwm2m_udp_receive: reply 0x2000c980 handled and removed
[00:00:03.676,269] <dbg> net_lwm2m_message_handling: lwm2m_udp_receive: checking for reply from [10.1.0.240]
[00:00:03.676,330] <dbg> net_lwm2m_message_handling: lwm2m_engine_default_content_format: No accept option given. Assume OMA TLV.
[00:00:03.676,422] <dbg> net_lwm2m_observation: engine_observe_node_init: OBSERVER ADDED 3/0/0/0(2)
[00:00:03.676,483] <dbg> ct: lwm2m_observe_cb: Added observer for 3/0
[00:00:03.676,605] <dbg> net_lwm2m_observation: engine_observe_node_init: token:'0dfbdec49d53d0fb' addr:10.1.0.240
[00:00:03.677,001] <dbg> net_lwm2m_message_handling: lwm2m_udp_receive: checking for reply from [10.1.0.240]
[00:00:03.677,062] <dbg> net_lwm2m_message_handling: lwm2m_engine_default_content_format: No accept option given. Assume OMA TLV.
[00:00:03.677,154] <dbg> net_lwm2m_observation: engine_observe_node_init: OBSERVER ADDED 4/0/0/0(2)
[00:00:03.677,215] <dbg> ct: lwm2m_observe_cb: Added observer for 4/0
[00:00:03.677,276] <dbg> net_lwm2m_observation: engine_observe_node_init: token:'1f48a7cf28172cc7' addr:10.1.0.240
[00:00:08.676,116] <dbg> ct: ciot_suspend_client: Suspending client
[00:00:09.003,265] <inf> net_lwm2m_rd_client: Suspend client
[00:00:09.003,326] <dbg> ct: lwm2m_event_cb: LwM2M engine suspended
[00:00:09.003,326] <inf> app: ************** CIOT_HUB_SUSPENDED
[00:00:09.009,521] <inf> net_lwm2m_engine: LWM2M engine thread paused
[00:00:10.511,383] <inf> app: Modem temperature: 23.0 degrees celsius
[00:00:10.511,444] <dbg> net_lwm2m_registry: lwm2m_engine_set: path:32999/0/27001/0, value:0x2001a3a8, len:8
[00:00:14.003,417] <inf> app: ********** resuming
[00:00:14.003,448] <dbg> ct: ciot_resume_client: Resuming client
[00:00:14.003,540] <inf> net_lwm2m_rd_client: Resume Client state
[00:00:14.013,580] <inf> net_lwm2m_engine: LWM2M engine thread resume
[00:00:14.018,585] <dbg> net_lwm2m_observation: lwm2m_notify_observer_path: NOTIFY EVENT 3/0/13
[00:00:14.018,676] <dbg> net_lwm2m_message_handling: generate_notify_message: [MANUAL] NOTIFY MSG START: 3/0/0(2) token:'0dfbdec49d53d0fb' [10.1.0.240] 14018
[00:00:14.018,951] <dbg> net_lwm2m_message_handling: generate_notify_message: NOTIFY MSG: SENT
[00:00:14.019,561] <err> net_lwm2m_message_handling: Failed to send packet, err 121
[00:00:14.534,454] <dbg> net_lwm2m_registry: lwm2m_engine_get: path:1/0/1, buf:0x200188f8, buflen:4
[00:00:14.534,545] <dbg> net_lwm2m_rd_client: sm_send_registration: registration sent [10.1.0.240]
[00:00:14.535,156] <err> net_lwm2m_message_handling: Failed to send packet, err 121
[00:00:19.531,036] <dbg> ct: at_notification_cb: +CSCON: 0
```

The bug was originally introduced by commit 21e551b6bcd653470c709c7f849ff586cfd2af35. @jheiskan81 asked me to open a ticket. He is going to work on it.

jheiskan81, 2022-10-24:
Thanks @chmielewskiandreas for raising this. Just minor update that issue only happen without DTLS.

jheiskan81, 2022-10-25:
@chmielewskiandreas I tested with nerf-sdk 2.1.1 release without security and i can't re-produce your issue.

chmielewskiandreas, 2022-10-26:
Hi @jheiskan81. I actually can reproduce this issue with lwm2m_client ...

```
diff --git a/samples/nrf9160/lwm2m_client/prj.conf b/samples/nrf9160/lwm2m_client/prj.conf
index 188f4b896..27b3db7ed 100644
--- a/samples/nrf9160/lwm2m_client/prj.conf
+++ b/samples/nrf9160/lwm2m_client/prj.conf
@@ -115,7 +115,7 @@ CONFIG_DOWNLOAD_CLIENT_HTTP_FRAG_SIZE_1024=y
 CONFIG_FOTA_DOWNLOAD=y
 
 # Set LwM2M Server IP address here
-CONFIG_LWM2M_CLIENT_UTILS_SERVER="coaps://leshan.eclipseprojects.io:5684"
+CONFIG_LWM2M_CLIENT_UTILS_SERVER="coap://leshan.eclipseprojects.io:5683"
 
 # Application Event Manager
 CONFIG_APP_EVENT_MANAGER=y
diff --git a/samples/nrf9160/lwm2m_client/src/main.c b/samples/nrf9160/lwm2m_client/src/main.c
index 522d046f2..eb2e5671b 100644
--- a/samples/nrf9160/lwm2m_client/src/main.c
+++ b/samples/nrf9160/lwm2m_client/src/main.c
@@ -266,6 +266,34 @@ static void rd_client_update_lifetime(int srv_obj_inst)
 	update_session_lifetime = false;
 }
 
+static int counter = 0;
+
+static void suspend(struct k_work *work);
+static void resume(struct k_work *work);
+
+K_WORK_DELAYABLE_DEFINE(suspend_work, suspend);
+K_WORK_DELAYABLE_DEFINE(resume_work, resume);
+
+static void suspend(struct k_work *work)
+{
+    if (counter == 0) {
+        LOG_INF("********** suspending");
+        lwm2m_engine_pause();
+        counter++;
+        k_work_schedule(&resume_work, K_SECONDS(5));
+    }
+}
+
+static void resume(struct k_work *work)
+{
+       if (counter == 1) {
+           LOG_INF("********** resuming");
+           lwm2m_engine_resume();
+           counter++;
+       }
+}
+
+
 static void rd_client_event(struct lwm2m_ctx *client, enum lwm2m_rd_client_event client_event)
 {
 	switch (client_event) {
@@ -300,7 +328,7 @@ static void rd_client_event(struct lwm2m_ctx *client, enum lwm2m_rd_client_event
 			/* Read a current server  lifetime value */
 			rd_client_update_lifetime(client->srv_obj_inst);
 		}
-
+        k_work_schedule(&suspend_work, K_SECONDS(20));
 #if defined(CONFIG_APP_LWM2M_CONFORMANCE_TESTING)
 		lwm2m_register_server_send_mute_cb();
 #endif
```

causes this

```
[00:00:04.383,392] <inf> lwm2m_security: Need to write credentials, requesting LTE and GNSS offline...
[00:00:04.453,369] <inf> lwm2m_neighbour_cell: RRC connected, measure when idle
[00:00:07.698,364] <inf> lwm2m_security: Requesting LTE and GNSS online
[00:00:10.007,232] <inf> lwm2m_security: Modem connection restored
[00:00:10.007,873] <inf> net_lwm2m_engine: Connected, sock id 0
[00:00:10.120,880] <dbg> app_lwm2m_client: rd_client_event: Registration complete
[00:00:10.121,032] <inf> app_lwm2m_client: Obtained date-time from modem
[00:00:10.123,596] <dbg> gnss_module: start_gnss: Started GNSS scan
[00:00:10.123,626] <inf> net_lwm2m_rd_client: Registration Done (EP='9QWjIXCaof')
[00:00:10.123,779] <dbg> gnss_module: gnss_event_handler: A-GPS data request not supported.
[00:00:30.121,002] <inf> app_lwm2m_client: ********** suspending
[00:00:30.521,362] <inf> net_lwm2m_rd_client: Suspend client
[00:00:30.521,392] <dbg> app_lwm2m_client: rd_client_event: LwM2M engine suspended
[00:00:30.525,085] <inf> net_lwm2m_engine: LWM2M engine thread paused
[00:00:35.525,177] <inf> app_lwm2m_client: ********** resuming
[00:00:35.525,238] <inf> net_lwm2m_rd_client: Resume Client state
[00:00:35.535,308] <inf> net_lwm2m_engine: LWM2M engine thread resume
[00:00:36.027,374] <err> net_lwm2m_message_handling: Failed to send packet, err 121
[00:00:40.475,189] <err> net_lwm2m_message_handling: Failed to send packet, err 121
[00:00:49.371,093] <err> net_lwm2m_message_handling: Failed to send packet, err 121
[00:01:07.163,085] <err> net_lwm2m_message_handling: Failed to send packet, err 121
[00:01:42.747,070] <err> net_lwm2m_message_handling: Failed to send packet, err 121
[00:02:53.915,039] <wrn> net_lwm2m_rd_client: Registration Update Timeout
[00:02:53.915,069] <dbg> app_lwm2m_client: rd_client_event: Registration update failure!
[00:02:53.915,344] <dbg> app_lwm2m_client: rd_client_event: Disconnected
[00:02:53.915,405] <inf> net_lwm2m_rd_client: Stop LWM2M Client: urn:imei:352656106103065
[00:02:54.165,527] <inf> net_lwm2m_rd_client: Start LWM2M Client: urn:imei:352656106103065
[00:02:55.077,270] <inf> net_lwm2m_rd_client: RD Client started with endpoint 'urn:imei:352656106103065' wit
h client lifetime 30
```

jheiskan81, 2022-10-26:
Thanks. I can now re-produce issue. I was using automatic lte registration status for triggering pause and resume.

jheiskan81, 2022-10-26:
@chmielewskiandreas 
I spot a problem. Socket start call is not happen if DTLS is disabled. Comment those lines like me below It fix problem.
Try modify Function lwm2m_engine_connection_resume:

```
int lwm2m_engine_connection_resume(struct lwm2m_ctx *client_ctx)
{
	int ret;

	if (client_ctx->connection_suspended) {
		lwm2m_close_socket(client_ctx);
		client_ctx->connection_suspended = false;
		ret = lwm2m_open_socket(client_ctx);
		if (ret) {
			return ret;
		}
/*  Changes  */
		// if (!client_ctx->use_dtls) {
		// 	return 0;
		// }
/* end of changes*/
		LOG_DBG("Resume suspended connection");
		return lwm2m_socket_start(client_ctx);
	}

	return 0;
}
```

jheiskan81, 2022-10-26:
Bug fix PR: https://github.com/zephyrproject-rtos/zephyr/pull/51646

chmielewskiandreas, 2022-10-26:
It works :) Thx - I actually had this fix already before writing to you but did not know whether it is correct. It is possible for you to implement that on lwm2m_rd_client_resume a proper event LWM2M_RD_CLIENT_EVENT_ENGINE_RESUMED) is raised as well? I also can implement this if you like ...

jheiskan81, 2022-10-26:
I prioritize bug fix. If you see value for LWM2M_RD_CLIENT_EVENT_ENGINE_RESUMED you can add it.

PR #51646 (2022-10-27): net: lwm2m: Fix connection resume without DTLS
LwM2M connection resume was missing socket connection when DTLS is disabled.

Fixes #51566
closed 2022-10-27 by carlescufi