zephyrproject-rtos/zephyr #51600: Bluetooth assert on flash erase using mcumgr

nordicjm, 2022-10-25:
**Describe the bug**
A bluetooth assert is hit when mcumgr is used to erase the flash for an image upload.

**To Reproduce**
Ensure slot sizes are large, I used a slot size of 0x70000, then build `zephyr/samples/subsys/mgmt/mcumgr/smp_svr` for a target board:
1. `mkdir build`
2. `cmake -GNinja -DBOARD=<BOARD> -DAPP_DIR=.. -DSB_CONFIG_BOOTLOADER_MCUBOOT=y -Dsmp_svr_OVERLAY_CONFIG="overlay-bt.conf" ../../../../../../share/sysbuild/`
3. `ninja`
4. `west flash`
5. Once device is flashed, use mcumgr to upload an application: `mcumgr --conntype ble --connstring ctlr_name=hci0,peer_name='Zephyr' image upload zephyr/zephyr.signed.bin` if this starts uploading right away with no error, hit reboot on the board and try again (erase will only run if there is data in the flash)
6. Board will assert with:
```
ASSERTION FAIL [0] @ WEST_TOPDIR/zephyr/subsys/bluetooth/controller/ll_sw/nordic/lll/lll.c:476 lll_preempt_calc: Actual EVENT_OVERHEAD_START_US = 3906
E: r0/a1:  0x00000003  r1/a2:  0x20004ee8  r2/a3:  0x00000001
E: r3/a4:  0x0003153d r12/ip:  0x00000000 r14/lr:  0x000191eb
E:  xpsr:  0x41000028
E: Faulting instruction address (r15/pc): 0x000191f6
E: >>> ZEPHYR FATAL ERROR 3: Kernel oops on CPU 0
E: Fault during interrupt handling
E: Current thread: 0x20001448 (mcumgr smp)
E: Halting system
```

**Expected behavior**
No assert

**Impact**
Showstopper

**Environment (please complete the following information):**
 - OS: Linux
 - Commit SHA or Version used: b8a957a0ebd955ed65b7f1e64c74dac794a37fb3

carlescufi, 2022-11-02:
@nordicjm could you just double-check this is fixed in `main`?

closed 2022-11-02 by carlescufi

nordicjm, 2022-11-02:
Problem is fixed

PR #51624 (2022-11-02): Bluetooth: Controller: Fix ticker slot reserved check
Fix ticker slot reserved check to handle deferred execution of ticker worker where ticks_elapsed is greater than ticks_slot_previous. In which case, each ticker expiring be checked for overlap with previous slot reservation.

This fixes:
ASSERTION FAIL [0] @ WEST_TOPDIR/zephyr/subsys/bluetooth/ controller/ll_sw/nordic/lll/lll.c:476
lll_preempt_calc: Actual EVENT_OVERHEAD_START_US = 3906

Flash erase operation exposed the assertion, CPU being halted deferred the ticker_worker collision resolution and ticker expiry was not individually checke [... cut]