zephyrproject-rtos/zephyr #51604: doc: is the documentation GPDR compliant since it uses Google Analytics without prompting the user about tracking?

MirkoCovizzi, 2022-10-25:
**Is your enhancement proposal related to a problem? Please describe.**
I'm using uBlock Origin on Firefox and due to the usage of Google Analytics in the documentation, the search feature takes a lot longer than necessary due to tracking. The user should be able to opt out of the tracking. So far I don't see anything like that.

**Describe the solution you'd like**
Add a GPDR compliant windows that lets users opt out of tracking.

MirkoCovizzi, 2022-10-25:
Paging @dbkinder @nashif

stephanosio, 2022-10-25:
cc @gmarull @carlescufi

henrikbrixandersen, 2022-10-25:
Personally I would like to see use move away from using any form of 3rd party analytics in our documentation.

henrikbrixandersen, 2022-10-31:
Also asked here: https://github.com/zephyrproject-rtos/zephyr/discussions/51605

Thalley, 2022-10-31:
> Personally I would like to see use move away from using any form of 3rd party analytics in our documentation.

Indeed. There are plenty of open source self-hosted solutions out there which works well (and can of course be configured to by GDRP compliant).

kartben, 2023-02-08:
I'll submit a PR to add the cookie consent script to all pages.
Note that going forward, it might make sense to gate the generation of analytics-related code behind a build flag, to make it easier for downstream adopters to generate a "clean" infocenter.

thedjnK, 2023-02-08:
> I'll submit a PR to add the cookie consent script to all pages.

Why is the linux foundation allowing this? This is on par with having cookie popups on the linux kernel git website or documentation.

beriberikix, 2023-02-15:
As I mentioned in the TSC, there are several ways companies are addressing GPDR/CCPA compliance so I'd thought I'd share some links.

- Disable analytics via consent (which this PR is doing)
- Enable [privacy controls in Google Analytics](https://support.google.com/analytics/answer/9019185?hl=en#zippy=%2Cin-this-article). Note, this may [not be sufficient](https://www.cookiebot.com/en/google-analytics-gdpr/) in some countries.
- Use a GPDR-compliant analytics service, many of which are OSS like:
  - [Posthog](https://posthog.com/)
  - [Fathom](https://usefathom.com/)
  - [Plausible](https://plausible.io/)
- Implement some custom server-side analytics
- Use server-side search analytics (ex. via Algolia, meilisearch)

stephanosio, 2023-02-16:
No longer relevant after https://github.com/zephyrproject-rtos/zephyr/pull/54875

closed 2023-02-16 by stephanosio

fabiobaltieri, 2023-02-16:
I'm surprised everyone so far argued about cookies and tracking and no one even tried to look at the actual data we've been building up since 2019 and see if it can be used to carve out any information useful to the project.

If anything what I found inadequate about the current system is that maintainers can't easily access the data since it's behind the LF account. I'd certainly like to peek at the data myself.

> How about moving away from 3rd party analytics all together?

> Personally I'd prefer to remove any cookies for tracking users (or tracking specifics users in any way at all) for analytics, unless we have a really good reason. "Because it is easy" is not a good reason.

That would be ideal, but is anyone willing to take the time to deploy and maintain a self hosted system? The project mission is to create a best-in-class RTOS, wondering if investing resources into this is really worth it. Why not self hosting a code review system and bug tracking? GitHub tracks you as well after all.