zephyrproject-rtos/zephyr #51608: [RFC] Do not allow self-merge

gmarull, 2022-10-25:
## Introduction

This proposal is about introducing a new rule to prevent PRs getting merged by their authors.

### Problem description

There's one edge case that could/should be avoided: people with merge rights merging their own PRs. This is a potential path to quickly get code merged without the necessary amount of review. As soon as it gets two approvals (which can be from colleagues), the author can quickly hit "Merge" before someone else comes and blocks. The majority of developers do not have such privilege.

### Proposed change

A PR can't be merged by its author. Exceptions: PRs with `Trivial` or `Hotfix` tags.

## Detailed RFC

See above.

### Proposed change (Detailed)

See above.

### Dependencies

None

### Concerns and Unresolved Questions

N/A

## Alternatives

Do nothing.

gmarull, 2022-10-26:
> Can you please provide examples where this became an issue?

I won't provide names or examples, but I think it has happened in the past. In any case, since this can always be subjective, I think such rule does not harm the project. It will just prevent the discussion whether merge privileges were abused or not, since the author won't be able to do that from the beginning.

mbolivar-nordic, 2022-11-23:
@gmarull do you want to bring this discussion to the process WG next week?

closed 2023-02-22 by mbolivar-nordic