zephyrproject-rtos/zephyr #51622: ESP32 mcuboot not support chip revision 1

sam131208, 2022-10-26:
**Describe the bug**
The mucboot can't run under the chip revision 1.
This is true even if I modify the definition of the chip version，@bootloader\mcuboot\boot\espressif\hal\include\esp32\sdkconfig.h

from
#define CONFIG_ESP32_REV_MIN_3 1
#define CONFIG_ESP32_REV_MIN 3
to 
#define CONFIG_ESP32_REV_MIN_1 1
#define CONFIG_ESP32_REV_MIN 1

**Logs and console output**
rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff8598,len:5296
load:0x40093000,len:13724
load:0x4009b800,len:6656
entry 0x4009341c
[esp32] [INF] [boot] chip revision: 1
[esp32] [ERR] [boot_comm] This chip is revision 1 but the application is configured for minimum revision 3. Can't run.
ets Jun  8 2016 00:22:57

nordicjm, 2022-10-26:
Does this belong here if the issue is in an mcuboot repo? https://github.com/mcu-tools/mcuboot

rftafas, 2022-10-26:
Hi,

It should be an MCUboot issue, IMHO.

That said, as for the problem, this is not a bug, it's intentional. There isn't support for Versions 0 and 1 of ESP32, on both Zephyr and MCUboot. Furthermore, there are no plans to do it on our side, even if one opens a feature request. Main reason is that we joined Zephyr development a lot later than current Chip Version 3 release, so we assume people willing to create new products using Zephyr would find Version 3 to buy. 

Besides, another reason is because there are several rom and silicon fixes, both for security and functional improvements (several Release Notes are out there, they are not a secret, out of scope reviewing it here). Is it impossible to support earlier versions? No, it is possible. Can you hack through it? Yes, you can. Even though, we think it is more important to put our efforts on peripherals and newer chips.

Finally, the best tool for supporting products using earlier ESP32 devices is ESP-IDF. We strongly encourage using Rev 3 for any new project.

closed 2022-12-06 by sam131208