zephyrproject-rtos/zephyr #51629: BLE stack execution fails with CONFIG_NO_OPTIMIZATIONS=y

tpennors, 2022-10-26:
**Describe the bug**
When BLE stack is compiled using CONFIG_NO_OPTIMIZATIONS=y, execution fails.

Reproduced on board STM32U5

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b b_u585i_iot02a samples/bluetooth/hci_uart/
2. west flash
4. See following error:
```
<inf> bt_hci_core: Identity: A0:76:4E:60:EC:16 (public)
<inf> bt_hci_core: HCI: version 5.0 (0x09) <err> os: ***** BUS FAULT *****
<err> os:   Precise data bus error
<err> os:   BFAR Address: 0x0
<err> os: r0/a1:  0x400138c0  r1/a2:  0x20001248  r2/a3:  0x20001248
<err> os: r3/a4:  0x00000000 r12/ip:  0xaaaaaaaa r14/lr:  0x080764d3
<err> os:  xpsr:  0x61000000
<err> os: Faulting instruction address (r15/pc): 0x08075fc4
<err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
<err> os: Current thread: 0x20001248 (BT_LW_WQ)
<err> os: Halting system
```

**Expected behavior**
No bus error

**Impact**
Not absolutely needed but want to make sure it is not hiding something that could happen with other optimisation flags

nordicjm, 2022-10-26:
Did you increase stack and thread sizes to account for the lack of optimisation?

tpennors, 2022-10-26:
No, but it could maybe solve the problem indeed. Do you think it is something that should be corrected for all optimisation to work or is it normal to only check size and debug optimisation ?

nordicjm, 2022-10-26:
It's something you'd only need to change for no optimisation. It does not make sense to increase all stack and thread sizes when the vast majority of devices do not need it and needlessly waste it, since it is only required when building a debug no optimisation build.

tpennors, 2022-10-26:
ok, thank you for the answer