zephyrproject-rtos/zephyr #51663: some LTS2 tests are defficient

npitre, 2022-10-26:
**Is your enhancement proposal related to a problem? Please describe.**

Part of #51094 is a list of CI test improvement backports for issues
that the current set of test configurations don't expose. Those issues
were made obvious while backporting RISC-V support to LTS2, but simply
enabling the right Kconfig option, like TLS support for example, would
make concerned tests to fail on many existing platform too.

**Describe the solution you'd like**

Fixes for those tests were backported to LTS2 and made available in
PR #51584.

nashif, 2024-02-08:
huh, seems like the maintainers file has the wrong label to the edtt module:

```
"West project: edtt":
  status: maintained
  maintainers:
    - aescolar
  collaborators:
    - wopu-ot
    - thoh-ot
  files: []
  labels:
    - "area: Tests"
    - ```

nashif, 2024-02-08:
that PR mentioned above never made it into LTS, so closing.

PR #51584 (2023-01-11): LTS2 test fixes
These are test fixes backported from the main branch needed for
successful CI run of a RISC-V SMP configuration.

- tests/kernel/fpu_sharing: Run test with MP_NUM_CPUS=1
- tests/kernel/smp: Add SMP switch torture test
- tests: kernel/smp: wait for threads to exits between tests
- tests: lifo_usage: make it less susceptible to SMP races
- tests/kernel/mem_protect: Check for thread_userspace_local_data
- tests: mem_protect: enlarge heap size of RISCV64
- tests: mem_protect: avoid allocating K_OBJ_MSGQ in userspace.
- tests: mem_protect: ensure allocated objects are initialized
-  [... cut]
closed 2024-02-08 by nashif