zephyrproject-rtos/zephyr #51713: drivers: flash: spi_nor: init fails when flash is busy

gregersrygg, 2022-10-27:
**Describe the bug**
The spi-nor flash device is automatically initialized during boot, but it does not check if the flash is busy with a write/erase operation. This can happen when a slow erase or page program operation starts before a soft-boot. The result is invalid responses from the flash because it will not decode many instruction when it's busy. Reading the JEDEC ID results in 0x00, 0x00, 0x00 instead of the actual JEDEC ID. So that is probably the most common error unless `CONFIG_SPI_NOR_SFDP_RUNTIME` is configured - then the next command will most likely fail instead.

[image]

[image]

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
   - External flash (MX25R6435F) on the nRF9160 DK and nRF52840 DK.
 - What have you tried to diagnose or workaround this issue?
   - Yes, see #51662.

**To Reproduce**
Steps to reproduce the behavior on an nRF52840 DK:
1. cd samples/drivers/spi_flash
1. unzip [init-busy.patch.zip](https://github.com/zephyrproject-rtos/zephyr/files/9879630/init-busy.patch.zip)
1. git apply init-busy.patch  # attached to this issue
1. enable CONFIG_FLASH_LOG_LEVEL_DBG=y
1. west build -b nrf52840dk_nrf52840
1. open a serial monitor application and connect to the nrf52840 DK
1. west flash
1. wait until you see «Test 1: Flash erase» (it will erase for ~17 seconds)
1. press reset button on the DK to reset during the erase operation
1. result: «mx25r6435f@0: device not ready.»

**Expected behavior**
I would expect the driver to wait until the device is ready or to reset the flash chip instead of failing. I don't know how to recover from this error other than reset the whole system.

**Impact**
It took quite some time to understand why the flash suddenly wouldn't report it's JEDEC id, but respond only zeroes instead of the actual JEDEC id. At first it seemed to fail at random, but later when working on erasing the whole flash (which can take several minutes), it was easier to understand why it was failing.

Handling this will be tricky to work around in the application. I'm not aware of a way to recover other than rebooting the system from the application code if the flash device is not ready. This will cause a reboot loop until the flash is ready. On the nRF9160, the modem will detect multiple ungraceful detaches from the cellular network and prevent the modem to attach to the network for some time.

**Logs and console output**
From the SPI flash sample described above:

```
*** Booting Zephyr OS build v3.1.99-ncs1-4-g97c4a1a6d1bd  ***

mx25r6435f@0 SPI flash testing
==========================

Test 1: Flash erase
*** Booting Zephyr OS build v3.1.99-ncs1-4-g97c4a1a6d1bd  ***
mx25r6435f@0: device not ready.
```

**Environment (please complete the following information):**
 - OS: MacOS
 - Toolchain: Zephyr SDK
 - Commit SHA: a9b223b26b0f19b181e17097d401551c0e6f351c (main)

de-nordic, 2022-10-27:
Nice report.

gregersrygg, 2022-10-28:
I used an nRF52 to reproduce the problem because it's less steps to access the external flash, but I forgot that it uses qspi by default. Have updated the patch to overlay the necessary changes to use spi and the jedec spi-nor driver instead of qspi and nordic qspi-nor driver. However, it seems like this problem exists for the nordic qspi-nor driver as well.

commit c9e07fc3 (2022-11-10): drivers: flash: spi_nor: fix flash busy during init

After a soft-reset of the host controller the flash device is not reset.
This can cause the flash device to still be busy writing or erasing when
the host controller boots and initializes the flash. The often results
in errors and the flash device not being initialized.

This fix polls the status register until the WIP flag is off before
initializing the flash device.

Fixes #51713
closed 2022-11-10 by carlescufi