zephyrproject-rtos/zephyr #51718: refacting testcase.yml meta to be more filterable and meaningful structure

hakehuang, 2022-10-27:
1. current testcase.yml meta keywords only defined as `string` but it is used as list. e.g.
`tags`
we shall move it to list structure for more filterable and proporely controlled

2. testcases attributes shall be added such as what test plan it can be
'test_plan:
    -  Acceptance
    -  ...
'

3. more

hakehuang, 2022-10-27:
this is related to #51288

nashif, 2022-11-17:
Adding hierarchial level of testing makes a lot of sense. We should be able to define the level of testing in such a way where we each level includes all lower levels or have some schema for inclusion defined somewhere else, so just as an example:

so if we take those levels: (borrowed from  [here](https://www.guru99.com/levels-of-testing.html))

- Unit Testing : checks if software components are fulfilling functionalities or not.
- Integration Testing : checks the data flow from one module to other modules.
- System Testing : evaluates both functional and non-functional needs for the testing.
- Acceptance Testing : checks the requirements of a specification or contract are met as per its delivery.

I am sure we can have more levels defined....

We could have tests that have:

```
 level: 
   - unit
```

and some other test having for example

```
level: 
  - system
 ```

which base on some hierarchy definition would include unit?

Some tests might belong to two level, we need to keep things flexible.

So instead of how we do things right now by pointing to some test directory directly or use tags, we just specify the level of testing at the command line of twister and do the filtering based on that.

PerMac, 2022-11-24:
@nashif What you proposed with levels is very useful and will help with understanding the role of a given test and having a better classification.  It also helps with communication with PMs and auditors and can improve QA in general. E.g. when a PM (auditor) ask: do you (what kind of) unit/system/etc tests have, is this area covered? It is a practice to evaluate a master test plan for a given component by the criticality level of it. A certain criticality can have by default a defined requirements for test levels.

hakehuang, 2022-11-24:
-  about level

```
level: (can be one or more below)
     -	unit
     - integration
     - acceptance
     - system
     - smoke
```
e.g.
if filter with 'unit' then only cases with 'unit' tag in level will be run.
if it is blank, then it will run when there is not level filter.
a case can have multiply tags

- about target platform

add a meta like 'target' which we recommend to run

```
target:
     -	simulation/hardware/host
```

nashif, 2024-01-26:
levels are alreadcy suppported in testcase.yaml

closed 2024-01-26 by nashif