zephyrproject-rtos/zephyr #51720: USB mass sample not working for FAT FS

spoerk, 2022-10-27:
**Describe the bug**
The [USB mass sample ](https://docs.zephyrproject.org/latest/samples/subsys/usb/mass/README.html#fat-fs-example) does not work for the FAT filesystem.

I want the **USB mass sample** with the **FAT FS configuration** as a basis for a new project on the **Nordic nRF52840 DK**. However, when I configure the sample to use FAT, the FS cannot be mounted.

**To Reproduce**
Compile the **USB mass sample** for the **nRF52840 DK** platform using the following **prj.conf**:
```
CONFIG_STDOUT_CONSOLE=y

#USB related configs
CONFIG_USB_DEVICE_STACK=y
CONFIG_USB_DEVICE_PRODUCT="Zephyr MSC sample"
CONFIG_LOG=y
CONFIG_USB_DRIVER_LOG_LEVEL_ERR=y
CONFIG_USB_MASS_STORAGE=y
CONFIG_USB_DEVICE_LOG_LEVEL_ERR=y
CONFIG_USB_MASS_STORAGE_LOG_LEVEL_ERR=y

CONFIG_MAIN_STACK_SIZE=1536

CONFIG_APP_MSC_STORAGE_FLASH_FATFS=y
```

**Expected behavior**
The sample should show a successful empty FAT FS over RTT and USB log.

**Impact**
Was showstopper, now fixed due to great help at [Nordic Devzone](https://devzone.nordicsemi.com/f/nordic-q-a/93046/log-to-nrf52-thingy-internal-flash).

**Logs and console output**
RTT log output:
```
[00:00:00.254,333] <err> usb_msc: Storage init ERROR !!!! - Aborting USB init
[00:00:00.258,514] <err> fs: fs mount error (-5)
[00:00:00.258,544] <err> main: Failed to mount filesystem
[00:00:00.258,728] <inf> main: The device is put in USB mass storage mode.
```

**Environment (please complete the following information):**
 - OS: Windows 11
 - Toolchain: Zephyr SDK v2.1.0
 - Commit SHA or Version used

**Additional context**
[Simon over at the Nordic Devzone](https://devzone.nordicsemi.com/f/nordic-q-a/93046/log-to-nrf52-thingy-internal-flash) has already solved the problem. His fix should be added to the repo.

desowin, 2022-10-28:
The simple solution is to change the "MX25R64" to "mx25r6435f@0" and "GD25Q16" to "gd25q16@0" in `samples/subsys/usb/mass/Kconfig`. However, it was deemed inappropriate and much more complex solution, involving defining the flashdisk via devicetree binding is in review. See #50856 for more information.

desowin, 2022-10-28:
> @jfischer-no @desowin Could you propose a multiline output which would validate that full workflow?

Example output when the sample works correctly is:
```
Area 4 at 0x0 on mx25r6435f@0 for 131072 bytes
Mount /NAND:: 0
/NAND:: bsize = 512 ; frsize = 512 ; blocks = 216 ; bfree = 216
/NAND: opendir: 0
End of files
[00:00:00.323,852] <inf> main: The device is put in USB mass storage mode.
```

Looking for "End of files" in addition to "The device is put in USB mass storage mode" would have caught the issue.

jfischer-no, 2023-01-05:
Resolved by #50856 and #51884

closed 2023-01-05 by jfischer-no