zephyrproject-rtos/zephyr #51722: User Need - easily find which features are supported on each board

gregshue, 2022-10-27:
NOTE: This issue is generated as an action item from the 2022 Oct 27 Dev Review meeting.  This issue is to capture/align on the problem statements resulting from #50788.

**Is your enhancement proposal related to a problem? Please describe.**

As a product prototype developer,
I need to easily and quickly determine which features are supported on a specific board,
so that I can efficiently decide which board to purchase and use for my prototype.

**Describe the solution you'd like**
I would like a solution that begins with the input of key feature selections and determines within 5 minutes which boards in my workspace support the selected features.

carlescufi, 2022-11-22:
Architecture WG:

- @nashif mentions that one option is for the board metadata to be "offline", i.e. metadata that is gathered regularly but not as part of the build system, and is kept in a "database" that is not connected to the actual configuration/build process
  - An example of this approach is something like this: https://github.com/zephyrproject-rtos/zephyr/pull/52377

kartben, 2025-02-28:
I am tempted to call victory on that one :)
I guess it can remain open until a majority of / all boards docs have adopted `.. zephyr:board-supported-hw` directive

kartben, 2025-05-26:
calling victory on this one with almost 600 boards (out of 760) using the `.. zephyr:board-supported-hw::`

closed 2025-05-26 by kartben