zephyrproject-rtos/zephyr #51723: User Need - Clear configuration error messages

gregshue, 2022-10-27:
NOTE: This issue is generated as an action item from the 2022 Oct 27 Dev Review meeting.  This issue is to capture/align on the problem statements resulting from #50788.

**Is your enhancement proposal related to a problem? Please describe.**

As a product integrator,
I need to easily and quickly determine the root cause of build configuration issues,
so that I can efficiently fix my build configuration settings.

**Additional context**
This need comes from problems interpreting and decoding "undefined reference to __device_dts_ord" errors.

henrikbrixandersen, 2022-10-27:
I guess this goes beyond build and configuration issues. We should strive for clear and easily understood error messages everywhere. In the build system, in code, ...

gregshue, 2022-10-27:
I've found that "clear" varies with audience.  That is where identifying the role is important.  

I think we have some flexibility in how granular to capture our user needs.  We probably don't have much flexibility on capturing constraints from the organization.

nashif, 2024-02-08:
very broad and inactionable, reopen with more specific enhancement requests.

closed 2024-02-08 by nashif