zephyrproject-rtos/zephyr #51728: soc: xtensa: esp32_net: Remove binary blobs from source tree

microbuilder, 2022-10-27:
**Describe the bug**

#51314 includes binary blobs, and should not have been approved/merged as-is, based on Zephyr's policy around firmware binary blobs.

**Expected behavior**

The firmware image for the second core should be built from source as an `ExternalProject`, similar to what is done for the LPC54114 here: https://github.com/zephyrproject-rtos/zephyr/blob/main/samples/subsys/ipc/openamp/boards/lpcxpresso54114_m4.conf#L3

It can be built during the main build process via some variation of:

```cmake
include(ExternalProject)

ExternalProject_Add(
  openamp_remote
  SOURCE_DIR ${APPLICATION_SOURCE_DIR}/remote
  INSTALL_COMMAND ""      # This particular build system has no install command
  CMAKE_CACHE_ARGS -DBOARD:STRING=${BOARD_REMOTE}
  CMAKE_CACHE_ARGS -DDTC_OVERLAY_FILE:STRING=${DTC_OVERLAY_FILE}
  BUILD_BYPRODUCTS "${REMOTE_ZEPHYR_DIR}/${KERNEL_BIN_NAME}"
  # NB: Do we need to pass on more CMake variables?
  BUILD_ALWAYS True
)
```

microbuilder, 2022-10-27:
@uLipe FYI

stephanosio, 2023-02-16:
Setting high priority and release blocker since this is a violation of the project policy.

PR #54885 (2023-02-16): samples: ipm_esp32: fix build setup and remove binary blob
Current `ipm_esp32` sample code requires `esp32_net_firmware.c` file
as source input, which is generated by `ipm_esp32_net` build.
File `esp32_net_firmware.c` was manually added as binary blob, which needs to
be removed from the repository to meet Zephyr's policy.

This change modifies `ipm_esp32` CMakeLists.txt to include the build
output of `ipm_esp32_net` as input source of `ipm_esp32` build, allowing the
binary blob to be removed.

Current approach shall be updated during next release.

Fixes #51728
closed 2023-02-16 by stephanosio