zephyrproject-rtos/zephyr #51731: Twister has a hard dependency on `west.log`

tristan-google, 2022-10-27:
A recent change to Twister (#51045) began importing a Python module (domains.py) that in turn imports `west.log`. This adds a hard dependency on West from Twister, which previously has not been the case. Before, it was only an optional dependency only needed if using Twister to run on-device tests.

Based on [comments](https://github.com/zephyrproject-rtos/zephyr/blob/main/scripts/zephyr_module.py#L438-L450) in the code and my own previous issue (#45355), my understanding is that Twister is intended to be usable without West. Can anybody confirm whether or not this is still the case?

If Twister is indeed meant to be standalone and not dependent on West, I'd like to propose:

- Removing the `from west import log` import from `domains.py` and simply use Python's standard logging module for the messages that module needs to print (this is what Twister uses for logging, anyways) in order to take out the dependency.
- Clarifying the documentation and maybe even add some kind of stricter dependency checks to the CI.

If not, could we update comments / requirements / documentation to explicitly make West a required dependency?

PerMac, 2022-10-28:
Copy from https://github.com/zephyrproject-rtos/zephyr/issues/50389:
"I think I am convinced with Anas's argumentation (presented in various other places, e.g. https://github.com/zephyrproject-rtos/zephyr/pull/36729#pullrequestreview-752684699) that the core build system scripting shouldn't be part of twister, since twister is a test runner and users should be able to do general building/flashing without it. IMO the fact that twister is doing it's own scripting for building is against this axiom but I think it is a legacy issue, since as Anas pointed out, twister existed before west. In twisterV2 we do building by directly calling west, but will yet to see if this can be the general solution for all configurations."

However, If such hard dependency can be easily removed it might be useful to do this. I don't see a reason why a user should be blocked from picking its own building tool as a backend. It is not possible in the current architecture but should be rather easy in V2. Not that I have any alternative to west in mind, but by following dependency inversion principle we won't need such tight bounding between modules.

nashif, 2022-10-28:
> IMO the fact that twister is doing it's own scripting for building is against this axiom 

In what way? Twister needs to build stuff in someway, it uses the build system. What is wrong with that? West is not the build system, cmake is, and twister just uses that.

> It is not possible in the current architecture but should be rather easy in V2.

Not possible how? Eveything is possible, the question is why do we need to do anything differently? Twister is supposed to work for everyone the same way and supporting different build systems was never a goal or make sense. We only have one build system.

danieldegrasse, 2022-11-01:
I've opened #51867 as a fix for this issue. One concern I have with this approach is that `domains.py` does not use the same logger instance as twister does. I'd prefer to use twister's logging instance, but since `domains.py` is imported before the `"twister"` logging instance gets configured, there isn't a straightforward way for the `domains.py` module to identify it is running using twister, and use the `"twister"` logging module.

tristan-google, 2022-11-01:
Thanks for the fix!

> One concern I have with this approach is...

Could you instead have Twister pass a Logger object when initializing either of those classes? That should also spare you from needing to set up the verbosity and format since it'll just assume the log settings from twister (or whatever is using this script) automatically.

danieldegrasse, 2022-11-01:
>Could you instead have Twister pass a Logger object when initializing either of those classes? That should also spare you from needing to set up the verbosity and format since it'll just assume the log settings from twister (or whatever is using this script) automatically.

We could, but I'd rather not add that parameter. The idea of the `logging` module is that if you call `logging.getLogger('name')`, the same instance should be returned, even if you are making the call from another module. So ideally, we would simply call `logging.getLogger('twister')` and be done. However, since west doesn't configure a logger with this name, we'd be left with a very basic logger. The core problem is how to detect that twister is being run versus west, and only reconfigure the logger when using west.

PR #51867 (2022-11-14): scripts: pylib: remove west log dependency from domains.py
Convert domains.py to use basic python logging module implementation to remove west logging dependency. This removes hard dependency on west from twister.

Fixes #51731
commit c303c6fa (2022-11-14): scripts: pylib: remove west log dependency from domains.py

Convert domains.py to use basic python logging module implementation to
remove west logging dependency. This removes hard dependency on west from
twister.

Fixes #51731
closed 2022-11-14 by fabiobaltieri