zephyrproject-rtos/zephyr #51735: ITE eSPI driver not supporting ESPI_PERIPHERAL_HOST_IO_PVT Host I/O peripheral Private Channel

NagarajuGundimi, 2022-10-28:
ITE eSPI driver not supporting ESPI_PERIPHERAL_HOST_IO_PVT Host I/O peripheral Private Channel.

ESPI_PERIPHERAL_HOST_IO_PVT_PORT_NUM. This is the port number(0x06A00000) used by the Host and EC to communicate over  the private channel.

laurenmurphyx64, 2022-11-01:
CC @Dino-Li @GTLin08

Dino-Li, 2022-11-02:
@NagarajuGundimi @laurenmurphyx64
Does the below change work on your platform?
https://github.com/Dino-Li/zephyr/commit/6afc690086d7d5d9b98e9e5b83ad0fc08255d313
Can you help to check if any data details need to be set before invoking callback?
(Please see pmc3_it8xxx2_ibf_isr function)

Dino-Li, 2022-11-08:
> 
> Integrated these changes but PVT port is not working on our platform.
> 
> I think pmc3_it8xxx2_ibf_isr() function should be similar to pmc1_it8xxx2_ibf_isr() function. Need to update command/data type and PM3STS flags etc.,

I looked into PVT port ISR for other SOCs (eg. ibf_pvt_isr()).
They didn't set flags before calling callback.
I am curious how they work?

Dino-Li, 2025-10-16:
Since the eSPI driver has supported the CONFIG_ESPI_PERIPHERAL_HOST_IO_PVT config option:
https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/espi/espi_it8xxx2.c#L818
I am going to close this issue. Please feel free to re-open this issue if you have any questions.

closed 2025-10-16 by Dino-Li