zephyrproject-rtos/zephyr #51762: Add a way to check if a given LOG_*() call would result in logging

arbrauns, 2022-10-28:
**Is your enhancement proposal related to a problem? Please describe.**
The `LOG_*()` macros transparently decide whether to log or not based on Kconfig settings and possibly runtime configuration, which works great. If, however, some expensive operation needs to be performed in order to create the log message, it would be nice if this operation was only performed if a log message actually ends up being produced, rather than always.

**Describe the solution you'd like**
A function/macro that returns a boolean based on whether or not logging is enabled at a specific log level.

**Describe alternatives you've considered**
One could use the internal macros (`Z_LOG_CONST_LEVEL_CHECK` etc), but those probably don't have any stability guarantees.

nordicjm, 2022-10-31:
Not the best way but can be done already e.g.
```
if (IS_ENABLED(CONFIG_ARCH_LOG_LEVEL_INF) ||| IS_ENABLED(CONFIG_ARCH_LOG_LEVEL_DBG)) {
        ...
        LOG_INF(...);
}
```

arbrauns, 2022-10-31:
That's a good start, thanks, but it's still overeager when source-specific log levels or even runtime-configured log levels are set. Adding those to the check as well would make it quite unwieldy.