zephyrproject-rtos/zephyr #51774: thread safety of adv_new in Bluetooth subsys

foldl, 2022-10-29:
**Describe the bug**

`adv_new` is not thread-safe, while the usage of `atomic_test_bit` and `atomic_set_bit` indicates
that it tries to be thread-safe.

https://github.com/zephyrproject-rtos/zephyr/blob/51af614f88eae21dc4baa3cbed03e1de81b053df/subsys/bluetooth/host/adv.c#L217

```c
static struct bt_le_ext_adv *adv_new(void)
{
        // ...
	for (i = 0; i < ARRAY_SIZE(adv_pool); i++) {
		if (!atomic_test_bit(adv_pool[i].flags, BT_ADV_CREATED)) {
			adv = &adv_pool[i];
			break;
		}
	}

	// now assume this thread is suspended and another thread runs to here
        // then: `adv` in both threads points to the same element of `adv_pool`
	(void)memset(adv, 0, sizeof(*adv));
	atomic_set_bit(adv_pool[i].flags, BT_ADV_CREATED);
	adv->handle = i;

	return adv;
}
```

**To Reproduce**

Code review.

**Expected behavior**

**Impact**

**Logs and console output**

<void>

**Environment (please complete the following information):**
 
<void>

**Additional context**
<void>

carlescufi, 2022-11-02:
Since Bluetooth in general as a feature is not considered thread-safe, this would be part of a larger task. I suggest we document this thoroughly for the time being and that we create a new GitHub issue to track the larger work.

carlescufi, 2022-11-03:
This issue should be closed by adding a "Thread Safety" section in this page: https://docs.zephyrproject.org/latest/connectivity/bluetooth/bluetooth-dev.html

carlescufi, 2022-11-03:
Note: This particular function could be made thread-safe by using `atomic_test_and_set()` instead of `atomic_test()` and then `atomic_set()`.

foldl, 2022-11-04:
First of all, this question should be settled: Is the Bluetooth feature designated to be thread-safe or not?

If it is, then a larger task needs to be done; otherwise, this issue can be closed with a 
cleanup of the `atomic_xxx` stuffs, and I believe that the whole Bluetooth subsys needs a cleanup
of such codes trying to be thread-safe.

Anyway, a "Thread Safety" section in the doc is welcome.

carlescufi, 2022-11-10:
@foldl our goal is to make the Bluetooth stack thread-safe: i.e. callable from preemptible threads. This means that issues like the one you describe here are real bugs and will be treated as such. Would you mind sending a PR to fix this particular issue, and then we will fix the others one by one as we find them.

carlescufi, 2022-11-10:
@theob-pro and @alwa-nordic will send a documentation PR.

closed 2023-05-24 by github-actions[bot]