zephyrproject-rtos/zephyr #51776: POSIX API is not portable across arches

yashi, 2022-10-29:
**Is your enhancement proposal related to a problem? Please describe.**

I believe that mentioning POSIX API in [The POSIX architecture](https://docs.zephyrproject.org/latest/boards/posix/doc/arch_soc.html#the-posix-architecture) is a +1.

Especially the code like this:

```c
while(1) {
    do_some();
    usleep(1);
}
```

This code works with POSIX API but it doesn't on native_posix. It is because, as I assume, that

- on Non POSIX arch with POSIX API, `usleep()` is `k_usleep()` and it's a scheduling point.
- on POSIX arch, `usleep()` is just a native function and it's not a scheduling point for Zephyr.

**Describe the solution you'd like**
Mention `sleep()` and `usleep()` in [POSIX Support](https://docs.zephyrproject.org/latest/services/portability/posix.html#posix-support) or [The POSIX architecture](https://docs.zephyrproject.org/latest/boards/posix/doc/arch_soc.html#the-posix-architecture) with note about scheduling point.

cfriedt, 2022-11-22:
There are a few changes to the POSIX subsystem planned, and yeah, you're right that there is some inconsistency between ARCH_POSIX and POSIX_API w.r.t. usleep. In the latter, it's possible that a short sleep will not trigger the scheduler (by calling `k_busy_wait()`).

The above is partially a workaround for a lack of high-precision timers, rather than being tick-based. The POSIX sleep calls should really lean on hardware timers rather than being tick-based.

The topic of high-precision timers / a real-time clock API will likely be increasingly important in the near term, as we need to resolve some layering violations when Zephyr switches to PicoLibC.

cc @stephanosio @keith-packard

cfriedt, 2022-11-24:
Actually, I'm going to categorize this as a bug. The (Linux) man page is pretty clear that it is broken.

https://man7.org/linux/man-pages/man3/usleep.3.html

PR imminent.

closed 2022-11-25 by cfriedt

cfriedt, 2022-12-22:
Also, when running on ARCH_POSIX we do not have control over POSIX API functions, since they are all native to the host OS. This is something that should be resolved in the LTSv3 POSIX Roadmap #51211

PR #52519 (2022-11-25): lib: posix: make sleep() and usleep() standards-compliant
4 commits:
* lib: posix: `sleep()` should report unslept time in seconds
* lib: posix: update `usleep()` to follow the POSIX spec
* tests: posix: add tests for `sleep()` and `usleep()`
* tests: posix: clock: do not use usleep in a broken way

See individual commit messages for details.

Fixes #51776
Fixes #52517
Fixes #52518
PR #52545 (2022-11-29): [Backport v3.2-branch] lib: posix: make sleep() and usleep() standards-compliant
Backport 027b79ecc4f71d36dbd89b9e0fb1dab7d096f0f7~4..027b79ecc4f71d36dbd89b9e0fb1dab7d096f0f7 from #52519

Fixes #51776
Fixes #52517
Fixes #52518