zephyrproject-rtos/zephyr #51778: stm32l562e-dk: Broken TF-M psa-crypto sample

mimok, 2022-10-29:
**Describe the bug**
Since Zephyr 3.2.0 (everything is fine with 3.1.0), Zephyr + TF-M seems to be broken on target stm32l562e-dk. 
TF-M psa-crypto sample execution hangs and triggers an HardFault before launching Zephyr app.
Hereafter, the console output:

> [INF] BANK 2 secure flash [0, 41] : OB [0, 127]
[ERR] Error while checking secure flash protection : set wmsec2
[INF] BANK 1 secure user flash [0, 30] : OB [0, 0]
[ERR] Error while checking secure user flash protection : set hdp1
[INF] Starting bootloader
[INF] Beginning BL2 provisioning
[WRN] TFM_DUMMY_PROVISIONING is not suitable for production! This device is NOT SECURE
[INF] Swap type: none
[INF] Swap type: none
[INF] Bootloader chainload address offset: 0x19000
[INF] Jumping to the first image slot
[INF] Beginning TF-M provisioning
[WRN] TFM_DUMMY_PROVISIONING is not suitable for production! This device is NOT SECURE
[Sec Thread] Secure image initializing!
Booting TF-M v1.6.0+8cffe127
Creating an empty ITS flash layout.
Creating an empty PS flash layout.

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b stm32l562e_dk_ns -d build/build-psa-crypto/ zephyr/samples/tfm_integration/psa_crypto
2. source build/build-psa-crypto/tfm/regression.sh
3. west flash -d build/build-psa-crypto/ --elf-file tfm_merged.hex
4. reset the board

**Expected behavior**
psa-crypto sample should pass without any issue

**Environment (please complete the following information):**
 - Zephyr 3.2.0
 - Windows 11 with WSL2

erwango, 2022-12-19:
@mimok Tested OK on `zephyr-v3.2.0-2703-g954ffc38d3`. See below.
Can you have a new check ?

```
[WRN] This device was provisioned with dummy keys. This device is NOT SECURE
[Sec Thread] Secure image initializing!
Booting TF-M v1.6.0+8cffe127
*** Booting Zephyr OS build zephyr-v3.2.0-2703-g954ffc38d300 ***
[00:00:00.095,000] <inf> app: att: System IAT size is: 543 bytes.
[00:00:00.095,000] <inf> app: att: Requesting IAT with 64 byte challenge.
[00:00:00.145,000] <inf> app: att: IAT data received: 543 bytes.

          0  1  2  3  4  5  6  7  8  9  A  B  C  D  E  F
00000000 D2 84 43 A1 01 26 A0 59 01 D3 AA 3A 00 01 24 FF ..C..&.Y...:..$.
00000010 58 40 00 11 22 33 44 55 66 77 88 99 AA BB CC DD X@.."3DUfw......
00000020 EE FF 00 11 22 33 44 55 66 77 88 99 AA BB CC DD ...."3DUfw......
00000030 EE FF 00 11 22 33 44 55 66 77 88 99 AA BB CC DD ...."3DUfw......
00000040 EE FF 00 11 22 33 44 55 66 77 88 99 AA BB CC DD ...."3DUfw......
00000050 EE FF 3A 00 01 24 FB 58 20 A0 A1 A2 A3 A4 A5 A6 ..:..$.X .......
00000060 A7 A8 A9 AA AB AC AD AE AF B0 B1 B2 B3 B4 B5 B6 ................
00000070 B7 B8 B9 BA BB BC BD BE BF 3A 00 01 25 00 58 21 .........:..%.X!
00000080 01 FA 58 75 5F 65 86 27 CE 54 60 F2 9B 75 29 67 ..Xu_e.'.T`..u)g
00000090 13 24 8C AE 7A D9 E2 98 4B 90 28 0E FC BC B5 02 .$..z...K.(.....
000000A0 48 3A 00 01 24 FA 58 20 AA AA AA AA AA AA AA AA H:..$.X ........
000000B0 BB BB BB BB BB BB BB BB CC CC CC CC CC CC CC CC ................
000000C0 DD DD DD DD DD DD DD DD 3A 00 01 24 F8 20 3A 00 ........:..$. :.
000000D0 01 24 F9 19 30 00 3A 00 01 24 FD 82 A5 01 63 53 .$..0.:..$....cS
000000E0 50 45 04 65 30 2E 30 2E 30 05 58 20 BF E6 D8 6F PE.e0.0.0.X ...o
000000F0 88 26 F4 FF 97 FB 96 C4 E6 FB C4 99 3E 46 19 FC .&..........>F..
00000100 56 5D A2 6A DF 34 C3 29 48 9A DC 38 06 66 53 48 V].j.4.)H..8.fSH
00000110 41 32 35 36 02 58 20 51 3C FB 28 72 9D 83 44 7E A256.X Q<.(r..D~
00000120 00 82 45 B5 2F 8B 13 49 C6 8D FE E0 59 95 7D BB ..E./..I....Y.}.
00000130 CE 3B 4A 2B 2C 0C E4 A5 01 64 4E 53 50 45 04 65 .;J+,....dNSPE.e
00000140 30 2E 30 2E 30 05 58 20 B3 60 CA F5 C9 8C 6B 94 0.0.0.X .`....k.
00000150 2A 48 82 FA 9D 48 23 EF B1 66 A9 EF 6A 6E 4A A3 *H...H#..f..jnJ.
00000160 7C 19 19 ED 1F CC C0 49 06 66 53 48 41 32 35 36 |......I.fSHA256
00000170 02 58 20 06 57 74 75 FD 42 D6 8E 52 CF A7 6B 55 .X .Wtu.B..R..kU
00000180 23 66 B0 40 46 DE 3A 46 9B 97 D0 C0 4A 88 18 82 #f.@F.:F....J...
00000190 BF 84 3F 3A 00 01 25 01 77 77 77 77 2E 74 72 75 ..?:..%.wwww.tru
000001A0 73 74 65 64 66 69 72 6D 77 61 72 65 2E 6F 72 67 stedfirmware.org
000001B0 3A 00 01 24 F7 71 50 53 41 5F 49 4F 54 5F 50 52 :..$.qPSA_IOT_PR
000001C0 4F 46 49 4C 45 5F 31 3A 00 01 24 FC 70 30 36 30 OFILE_1:..$.p060
000001D0 34 35 36 35 32 37 32 38 32 39 31 30 30 58 40 D2 4565272829100X@.
000001E0 39 FA BE E6 C7 3B 45 5B 74 D2 9D E2 7F 62 A3 05 9....;E[t....b..
000001F0 D5 5E A1 14 AE F4 9B 02 6D 2C 62 F1 A4 DB 9E 95 .^......m,b.....
00000200 26 6C 38 C4 EE 23 C9 67 8C B0 37 8E FA 18 D0 C6 &l8..#.g..7.....
00000210 59 48 AA 56 81 54 79 93 83 4D 87 D7 2F AB 23    YH.V.Ty..M../.#

[00:00:00.398,000] <inf> app: Persisting SECP256R1 key as #1
[00:00:00.448,000] <err> app: Already exists
[00:00:00.448,000] <err> app: Function: 'crp_gen_key_secp256r1'
[00:00:00.448,000] <err> app: Failed to generate key.
[00:00:00.466,000] <inf> app: Calculating SHA-256 hash of value

          0  1  2  3  4  5  6  7  8  9  A  B  C  D  E  F
00000000 50 6C 65 61 73 65 20 68 61 73 68 20 61 6E 64 20 Please hash and 
00000010 73 69 67 6E 20 74 68 69 73 20 6D 65 73 73 61 67 sign this messag
00000020 65 2E                                           e.

          0  1  2  3  4  5  6  7  8  9  A  B  C  D  E  F
00000000 9D 08 E3 E6 DB 1C 12 39 C0 9B 9A 83 84 83 72 7A .......9......rz
00000010 EA 96 9E 1D 13 72 1E 4D 35 75 CC D4 C8 01 41 9C .....r.M5u....A.

[00:00:00.514,000] <inf> app: Signing SHA-256 hash

          0  1  2  3  4  5  6  7  8  9  A  B  C  D  E  F
00000000 5D A2 59 83 D3 DA 74 B2 1F 97 5E 7F 92 55 AC CE ].Y...t...^..U..
00000010 30 69 EA 9C 7E F2 87 BE 27 94 9E E4 28 CD DC 4F 0i..~...'...(..O
00000020 88 11 3A EA 04 45 8E F8 5D 3B AF 2E A5 F4 75 3A ..:..E..];....u:
00000030 FC 8B 40 79 FA A8 4D FD AB B9 CB 58 8E 7A 0A 1A ..@y..M....X.z..
[... 34 lines cut ...]
00000040 DD                                              .

[00:00:01.122,000] <inf> app: Adding subject name to CSR
[00:00:01.127,000] <inf> app: Adding subject name to CSR completed
[00:00:01.134,000] <inf> app: Adding EC key to PK container
[00:00:01.140,000] <inf> app: Adding EC key to PK container completed
[00:00:01.147,000] <inf> app: Create device Certificate Signing Request
[00:00:01.247,000] <inf> app: Create device Certificate Signing Request completed
[00:00:01.254,000] <inf> app: Certificate Signing Request:

-----BEGIN CERTIFICATE REQUEST-----
MIHsMIGQAgEAMC4xDzANBgNVBAoMBkxpbmFybzEbMBkGA1UEAwwSRGV2aWNlIENl
cnRpZmljYXRlMFkwEwYHKoZIzj0CAQYIKoZIzj0DAQcDQgAERDLvru6y7qKi+jI5
hmZIFbLjZ5Ja0nw3Z9tWX/A0FOmBD6tcMMfch2ia1Ctw9CqxX97+y/zMZEKcf9pk
s17u3aAAMAwGCCqGSM49BAMCBQADSQAwRgIhANgB9D2Mrl4O8JWo/qZqg9FEtN+g
WZ1Jj2eij5g8Pq5AAiEA8LECjaXlMH7siIFwPUeZBI+dQVOcWI/P2st/prThOf4=
-----END CERTIFICATE REQUEST-----

[00:00:01.296,000] <inf> app: Encoding CSR as json
[00:00:01.301,000] <inf> app: Encoding CSR as json completed
[00:00:01.307,000] <inf> app: Certificate Signing Request in JSON:

{"CSR":"-----BEGIN CERTIFICATE REQUEST-----\nMIHsMIGQAgEAMC4xDzANBgNVBAoMBkxpbmFybzEbMBkGA1UEAwwSRGV2aWNlIE}

```

mimok, 2022-12-21:
Well, that is weird. I will check again as soon as possible using a fresh zephyr install...

mimok, 2023-01-09:
I still have the same issue on two different computers but with the same board. Did you try on a real stm32l562 board ?

mimok, 2023-01-11:
I already did full mass erase before firmware loading. I will try to perform a full regression by setting TZEN to 0.
Yes, I use WSL2 on both computer but i would be suprised if that is the source of the problem.
I tried using USBIPD and west flash or STM32CubeProgrammer on windows side (with tfm_merged.hex).

I tried to see what is the problem and i believe (don't remember  exactly) that the system enters an ISR (implemented using an infinite loop) related to the watchdog.

mimok, 2023-02-02:
Because last version of zephyr works as expected for me and you can't reproduce the issue on your side, I would say we can close the issue for now.

closed 2023-02-02 by mimok