zephyrproject-rtos/zephyr #51780: windows-curses Python package in requirements.txt can't install if using Python 3.11

john-pritchard, 2022-10-29:
**Describe the bug**
It seems the Python package "windows-curses" (in requirements.txt) can't be found using Python 3.11.
- Running Windows 10, Python 3.11 already installed (not using Chocolatey)
- Reverting to Python 3.10.8 works just fine, that is my current workaround

**To Reproduce**
Steps to reproduce the behavior:
1. Fresh install of Python 3.11 (no other versions installed)
2. Follow latest Zephyr Getting Started Guide - Windows, "Install Globally"
3. Python package installation should fail when getting to the step that runs the code `pip3 install -r %HOMEPATH%\zephyrproject\zephyr\scripts\requirements.txt`
4. See error

**Expected behavior**

**Impact**
Blocker from installing Python packages requried.

**Logs and console output**
When installing required Python packages, I get an error at the end (note I had tried a couple times so the successfully installed packages pulled from cache):

> \`\`\`
> C:\Users\USER>pip3 install -r %HOMEPATH%\zephyrproject\zephyr\scripts\requirements.txt
> Collecting pyelftools>=0.27
>   Using cached pyelftools-0.29-py2.py3-none-any.whl (174 kB)
> Requirement already satisfied: PyYAML>=5.1 in c:\users\john\appdata\local\programs\python\python311\lib\site-packages (from -r \Users\john\zephyrproject\zephyr\scripts\requirements-base.txt (line 11)) (6.0)
> Requirement already satisfied: pykwalify in c:\users\john\appdata\local\programs\python\python311\lib\site-packages (from -r \Users\john\zephyrproject\zephyr\scripts\requirements-base.txt (line 14)) (1.8.0)
> Collecting canopen
>   Using cached canopen-2.0.0-py3-none-any.whl (59 kB)
> Requirement already satisfied: packaging in c:\users\john\appdata\local\programs\python\python311\lib\site-packages (from -r \Users\john\zephyrproject\zephyr\scripts\requirements-base.txt (line 18)) (21.3)
> Collecting progress
>   Using cached progress-1.6.tar.gz (7.8 kB)
>   Preparing metadata (setup.py) ... done
> Collecting psutil
>   Using cached psutil-5.9.3.tar.gz (483 kB)
>   Installing build dependencies ... done
>   Getting requirements to build wheel ... done
>   Installing backend dependencies ... done
>   Preparing metadata (pyproject.toml) ... done
> Collecting pylink-square
>   Using cached pylink_square-0.14.3-py2.py3-none-any.whl (82 kB)
> Collecting anytree
>   Using cached anytree-2.8.0-py2.py3-none-any.whl (41 kB)
> Collecting intelhex
>   Using cached intelhex-2.3.0-py2.py3-none-any.whl (50 kB)
> Requirement already satisfied: west>=0.10.1 in c:\users\john\appdata\local\programs\python\python311\lib\site-packages (from -r \Users\john\zephyrproject\zephyr\scripts\requirements-base.txt (line 30)) (0.14.0)
> ERROR: Could not find a version that satisfies the requirement windows-curses (from versions: none)
> ERROR: No matching distribution found for windows-curses
> \`\`\`

**Environment (please complete the following information):**
 - OS: Windows
 - Zephyr SDK ver 0.15.1
 - Python ver 3.11.0
 - Zephyr docs ver 3.2.99

stephanosio, 2022-10-31:
windows-curses currently lack Python 3.11 support.

Related issue: https://github.com/zephyrproject-rtos/windows-curses/issues/31

stephanosio, 2022-12-13:
[windows-curses 2.3.1](https://github.com/zephyrproject-rtos/windows-curses/releases/tag/v2.3.1) now supports Python 3.11.