zephyrproject-rtos/zephyr #51796: LE Audio: Improve stream coupling for CIS as the unicast client

Thalley, 2022-10-31:
**Is your enhancement proposal related to a problem? Please describe.**
Imagine a TWS scenario with bidirectional audio: Two audio streams for each device, in each direction, giving a total of 4 audio streams on 2 bidirectional CIS. 

In the current implementation, this can only be achieved if all 4 audio streams are codec configured (from calling `bt_audio_stream_config`) before they are grouped (from calling `bt_audio_unicast_group_create`). The reason for this is that if the grouping is done before the codec configuring, then the stack does not know which of the the 4 streams can use the same CIS (as they are not coupled with an ACL yet). 

In the scenario that a unicast client wants to setup the TWS scenario, but there is a delay between the two devices, and the client don't want to wait (bad user experience), then it cannot codec configure all 4 streams first. The current solution for this, is to either
A) Codec configure the first 2 streams and start those. When the second device connects, the 2 streams will need to be terminated, and the group recreated, to do codec configure again before the grouping. 
B) Codec configure the 2 streams for the connected device, but still add the 2 non-configured streams to the group, thus creating a total of 3 CIS. 

Neither of these are optimal. 

**Describe the solution you'd like**
Solution 1) After a stream has been codec configured, we attempt to couple the stream and any other stream that has been codec configured, to see if we can at least reduce the number of CIS connected from 3 to 2 in the above scenario B). This will only be possible if the CIS has been declared as bidirectional, and use the right QoS values, then it was originally put into the unicast group, as the CIG cannot be modified after the first CIS has been connected. This won't give an ideal solution, as it is restrictive and we will still end up with a CIG that has been configured for an additional CIS that will be connected. 

Solution 2) We allow streams to be "coupled" or "paired" before creating unicast group. This will allow us to ensure that 2 streams use the same CIS, and could be something as simple as 
```c
int bt_audio_unicast_stream_pair(struct bt_audio_stream *a, 
				 struct bt_audio_stream *b);
```
With this solution we would of course need to verify that when streams `a` and `b` are codec configured, that they share the same ACL.

Solution 3) Similar to 2), but where the coupling/pairing happens as part of `bt_audio_unicast_group_create`. We can modify the `bt_audio_unicast_group_param` so that we simply add a pointer to a paired stream, such as 
```c
struct bt_audio_unicast_group_param {
	/** Pointer to a stream object. */
	struct bt_audio_stream *stream;

+	/** Pointer to a stream object. */
+	struct bt_audio_stream *paired_stream;

	/** The QoS settings for the @ref bt_audio_unicast_group_param.stream. */
	struct bt_codec_qos *qos;

	/** @brief The direction of the @ref bt_audio_unicast_group_param.stream
	 *
	 * If two streams are being used for the same ACL connection but in
	 * different directions, they may use the same CIS.
	 */
	enum bt_audio_dir dir;
};
```
Alternatively we can modify the `bt_audio_unicast_group_param` to be something like
```c
struct bt_audio_unicast_group_stream_param {
	/** Pointer to a stream object. */
	struct bt_audio_stream *stream;

	/** The QoS settings for the @ref bt_audio_unicast_group_param.stream. */
	struct bt_codec_qos *qos;
};

struct bt_audio_unicast_group_param {
	struct bt_audio_unicast_group_stream_param *sink_stream;

	struct bt_audio_unicast_group_stream_param *source_stream;
};
```
And then couple the `sink` and `source` streams implicitly. Similar as for 2), we would need to ensure that the two streams are codec configured for the same ACL.

**Describe alternatives you've considered**
See alternatives above

**Additional context**
CAP unicast start procedure expects that all streams for the intended usecase to be configured at the same time:
> An Initiator should wait for all Sink ASEs to be used for the intended use case to transition to the Streaming state before sending audio data on any of the associated unicast Audio Streams. The waiting time is implementation specific.

This implies that in the above TWS scenario, CAP would actually expect us to either wait for the second device, or to tear down the existing group and recreate it with all streams once the second device has connected (which is possible with the current API).

rick1082, 2022-11-03:
Regarding to "An Initiator should wait for all Sink ASEs to be used for the intended use case to transition to the Streaming state before sending audio data on any of the associated unicast Audio Streams. The waiting time is implementation specific.

This implies that in the above TWS scenario, CAP would actually expect us to either wait for the second device, or to tear down the existing group and recreate it with all streams once the second device has connected (which is possible with the current API)." 
Seems this would be complex. There second acceptor could be connected by initiator at anytime, could be when initiator just connected with first acceptor, and then initiator needs to teardown the CIG immediately which it just created.

Thalley, 2022-11-03:
> Seems this would be complex. There second acceptor could be connected by initiator at anytime, could be when initiator just connected with first acceptor, and then initiator needs to teardown the CIG immediately which it just created.

I agree. I think it is important to note that this is only a `should`, and it is allowed to either way. We might want to ask the GAWG about this, to see if the following really is the intention:

1. Connect Acceptor A
2. Start audio streams for Acceptor A
3. Wait
4. Connect Acceptor B
5. Stop audio streams for Acceptor A
6. Start audio streams for Acceptor A and Acceptor B

MariuszSkamra, 2022-11-14:
Another option is to make a `bt_audio_stream` object to be related 1:1 to CIS (Connected Isochronous Stream). It would be straightforward to the user that is familiar with Core specification. We would not need to `pair` the streams as the `bt_audio_stream` would represent either unidirectional or bidirectional CIS.
Currently stream somehow wraps the `struct bt_audio_ep` endpoint.

Thalley, 2022-11-28:
@MariuszSkamra Somewhat related: After https://github.com/zephyrproject-rtos/zephyr/pull/51513/commits/f958b3b2bb210d0fc9c5dfdd734b29de34753ca6 was merged, the `stream->ep` is now required before a stream can be added to a unicast group. Since `stream->ep` is not set until after it has been configured, this also means that a device can no longer create a unicast group with streams, before the streams have been Codec Configured. The reasoning is that `unicast_group_add_stream` now uses the `ep` to get the direction, as well as storing the allocated `audio_iso` in `ep` only.

PR #53420 (2023-01-18): Bluetooth: audio: Improve stream coupling for CIS as the unicast client
Allow  the streams to be paired when creating unicast group. This will
allow to reuse the same ISO for the paired streams.

Fixes: #51796
commit 36058480 (2023-01-18): Bluetooth: audio: Improve stream coupling for CIS as the unicast client

Allow  the streams to be paired when creating unicast group. This will
allow to reuse the same ISO for the paired streams.

Fixes: #51796
closed 2023-01-18 by carlescufi