zephyrproject-rtos/zephyr #51814: ARC irq_offload doesn't honor thread switches

andyross, 2022-10-31:
The arc architecture returns unconditionally from the trap invoked by `arch_irq_offload()`, when it should be invoking the scheduler to find the next thread to run as for any exception or interrupt.  See the implementation here; what it should be doing is calling _get_next_switch_handle as you see for regular interrupts higher up in the file:

https://github.com/zephyrproject-rtos/zephyr/blob/e9dde8256517cdf919a904c0bbabc0c49573db4a/arch/arc/core/fault_s.S#L234

In fact we have a case in tests/kernel/threads/thread_apis (test_abort_from_isr) that aborts the current thread from within an irq_offload() handler, which can't possibly work with the current implementation.  Oddly this test **passes** for some hard-to-diagnose reasons in the current tree (it seems like it crashes and enters some kind of nested trap that then exits via the normal interrupt path?).  I stumbled on this with the zync work when something changed.  But you can see this clear as day with a one-line assert:

```patch
diff --git a/arch/arc/core/irq_offload.c b/arch/arc/core/irq_offload.c
index ec96946b1f..056e3bb359 100644
--- a/arch/arc/core/irq_offload.c
+++ b/arch/arc/core/irq_offload.c
@@ -30,4 +30,6 @@ void arch_irq_offload(irq_offload_routine_t routine, const void *parameter)
                :
                : [id] "i"(_TRAP_S_SCALL_IRQ_OFFLOAD) : );

+       /* If _current was aborted in the offload routine, we shouldn't be here */
+       __ASSERT((_current->base.thread_state & _THREAD_DEAD) == 0, "");
 }
```

You will see the panic in the test logs (though, again, it actually doesn't re-enter the thread).

You can see the misbehavior more clearly with a tiny app instead.  This one does indeed try to return back to the current thread and hits a visible crash:

```c
#include <zephyr/kernel.h>
#include <zephyr/irq_offload.h>

/* Build with CONFIG_TEST=y and CONFIG_IRQ_OFFLOAD=y */

struct k_thread thr;
K_THREAD_STACK_DEFINE(tstk, 4096);

void offload_fn(const void *a)
{
	k_thread_abort(&thr);
}

void thread_fn(void *a, void *b, void *c)
{
	irq_offload(offload_fn, NULL);
	printk("SHOULD NOT BE HERE\n");
}

void main(void)
{
	k_thread_create(&thr, tstk, K_THREAD_STACK_SIZEOF(tstk),
			thread_fn, NULL, NULL, NULL, 0, 0, K_NO_WAIT);
	k_msleep(1000);
}
```

ruuddw, 2022-11-03:
What board and config are you testing?
ARC doesn't set/support the `CONFIG_IRQ_OFFLOAD_NESTED`, so the `k_sem_give(&offload_sem)` in `irq_offload()` takes care of the  reschedule. The tests, including yours above, seem to work fine for me.

andyross, 2022-11-03:
I'm seeing this in the zync work, which is a new synchronization backend (among other things, it turns uncontended synchronization calls into pure atomic operations that don't involve the kernel).

But regardless, k_sem_give() has never been documented as a rescheduling point, and you can't rely on that.  And I'm pretty sure there's some other funny stuff going on; in particular, there's a test that tries to **abort** the current thread from an irq_offload() handler, and that's blowing up for me for sure.

ruuddw, 2022-11-03:
Relying on the k_sem_give() to handle the reschedule was probably not the intend when this code was written, but it explains why tests pass on the current tree. I'm not aware of any regular test failing, and the above 'abort' test seem to work as expected as well (I don't see '"SHOULD NOT BE HERE" or exceptions, also not when adding some code after the sleep in main;

Changing the irq_offload to match your intended behavior doesn't look trivial, the irq offload is handled via the exception path and not the interrupt path. Appreciate if you have any suggestions how to fix this, or maybe solve it at a higher layer as irq_offload is only for simplifying some testing.

andyross, 2022-11-03:
Yeah, I have a patch in #51924 that disables test cases that rely on irq_offload allowing preemption.  That's probably good enough for now.

And yes: indeed, irq_offload is expressly not an application API.  It exists only as an oddball hook for tests to exercise interrupt contexts.  But nonetheless, given that requirement, we really should make every effort to make sure it does, in fact, "act like a hardware interrupt" in all ways possible.

I'm not a zealot though.  If this is hard we could push it down to "low" priority, or alternatively come up with a kconfig (in the vein of the existing IRQ_OFFLOAD_NESTED) that allows the platform to say exactly what it can do.

evgeniy-paltsev, 2022-11-03:
> Changing the irq_offload to match your intended behavior doesn't look trivial, the irq offload is handled via the exception path and not the interrupt path. Appreciate if you have any suggestions [...]

My first thought about this was - let's drop this interrupt emulation in exception entirely and just implement it via real interrupts.

Pros:
 * we'll test real existing interrupt path code instead of current exception implementation which exists only for test purposes.
 * we can implement IRQ_OFFLOAD_NESTED
Cons
 * we need at least one free IRQ line. I'd say it's doable on all our testing platform, but it can't be done in generic way.

Still, it require some investigations - maybe it's not so simple :)

ruuddw, 2022-11-04:
Indeed I wondered why it was implemented (long time ago) as an exception and not as a SW interrupt. Let's get the intend and expectations clear, then see how to model. ARC e.g. has two types of interrupts, fast and normal: should both be tested/covered or is any interrupt context enough since it is used for testing kernel stuff, not interrupts.
Maybe something to take along when you rework the (SMP and other) interrupt design, Evgeniy.

"we need at least one free IRQ line" -> for testing, it is probably ok to re-use another interrupt vector not used during the test, not necessarily a 'free' one. Any vector can be SW triggered.

evgeniy-paltsev, 2022-12-11:
Hi @andyross ,

I'm working on proper implementation of irq_offload for ARC and I have few questions about CONFIG_IRQ_OFFLOAD_NESTED, so probably you can answer them as an author of this option :)

On ARC we have HW support of multiple interrupt priority levels - so we can implement IRQ_OFFLOAD_NESTED by triggering IRQ line with priority higher than the interrupt priority of currently active interrupt. That allows to implement IRQ_OFFLOAD_NESTED without adding special code to interrupt path for IRQ_OFFLOAD_NESTED handling (which is good) and test the real nested interrupt code.

Current tests call (nested) irq_offload only from timer handler (which is executed in IRQ context). Given there is no documentation and no other test examples I'm wondering about: 

 * can we call irq_offload from any IRQ handler or only from timer handler?
 * can we call irq_offload from irq_offload handler?
 * what is maximum nesting of irq_offload? I.e.: can we call irq_offload from irq_offload handler from irq_offload handler?

evgeniy-paltsev, 2023-04-05:
Hi @andyross ,

kindly ping about you reply to https://github.com/zephyrproject-rtos/zephyr/issues/51814#issuecomment-1345599805

Thanks!

closed 2024-02-26 by github-actions[bot]