zephyrproject-rtos/zephyr #51820: Longer strings aren't logged

ahedin25, 2022-10-31:
Strings ~400 characters long sometimes fail to print.  Strings 450+ characters long always fail to print.   'log mem' doesn't indicate that the log buffer was too small. 

Am I missing a Kconfig value or LOG2 documentation that discusses this limitation?  I have used printk as a workaround, but I'd prefer not to do this.

Windows, main - b89a3d955ab377270aa2d900f0c21495b6e68cec

Steps to reproduce the behavior:
1. west build -p always -b nrf52dk_nrf52832 samples/log2
2. west flash
3. test size 400
4. test count 10 (some messages not printed)
5. test size 450
6. test count 10
7. log mem
Log message buffer utilization report:
        Capacity: 4092 bytes
        Currently in use: 0 bytes
        Maximum usage: 1488 bytes

[log2.zip](https://github.com/zephyrproject-rtos/zephyr/files/9903875/log2.zip)

ahedin25, 2023-01-10:
Logging JSON messages often requires long strings (~1K).

glarsennordic, 2023-03-27:
Is this actually fixed? It seems like the bot completed it as done solely because it went stale. That seems inaccurate to me, this is a legitimate problem that should probably be addressed at some point

ahedin25, 2023-03-27:
I don't think it was fixed, but the behavior is different.
For strings around the size of 400, 9 out of 10 get printed.
Strings larger than 443 result in "--- 1 messages dropped ---" being logged.

ahedin25, 2023-03-27:
uart:~$ log mem
Log message buffer utilization report:
        Capacity: 4092 bytes
        Currently in use: 0 bytes
        Maximum usage: 1484 bytes

ahedin25, 2023-03-27:
The size doesn't seem to matter much.  Similar results with 4/16/32K.

west build -p -b nrf52dk_nrf52832
```
uart:~$ kernel stacks
0x200087c8 sysworkq                         (real size 1024):   unused  848     usage  176 / 1024 (17 %)
0x20008350 shell_uart                       (real size 2048):   unused  992     usage 1056 / 2048 (51 %)
0x20000258 logging                          (real size 5120):   unused 4624     usage  496 / 5120 ( 9 %)
0x20008638 idle                             (real size  320):   unused  236     usage   84 /  320 (26 %)
0x200086f0 main                             (real size 1024):   unused  680     usage  344 / 1024 (33 %)
0x2000b340 IRQ 00                           (real size 2048):   unused 1648     usage  400 / 2048 (19 %)
uart:~$ log mem
Log message buffer utilization report:
        Capacity: 32764 bytes
        Currently in use: 0 bytes
        Maximum usage: 400 bytes
uart:~$ test size 450
Size changed to 450
[00:00:46.451,599] <inf> log2_test: size: 128->450
uart:~$ test count 10
Count changed to 10
[00:00:52.452,087] <dbg> log2_test: main: .
--- 1 messages dropped ---
[00:00:53.452,362] <dbg> log2_test: main: .
...

uart:~$ log mem
Log message buffer utilization report:
        Capacity: 32764 bytes
        Currently in use: 0 bytes
        Maximum usage: 1040 bytes
uart:~$ kernel stacks
0x200087c8 sysworkq                         (real size 1024):   unused  848     usage  176 / 1024 (17 %)
0x20008350 shell_uart                       (real size 2048):   unused  992     usage 1056 / 2048 (51 %)
0x20000258 logging                          (real size 5120):   unused 4560     usage  560 / 5120 (10 %)
0x20008638 idle                             (real size  320):   unused  236     usage   84 /  320 (26 %)
0x200086f0 main                             (real size 1024):   unused  680     usage  344 / 1024 (33 %)
0x2000b340 IRQ 00                           (real size 2048):   unused 1648     usage  400 / 2048 (19 %)

```

glarsennordic, 2023-03-27:
In that case, IMO this should be reopened, not that I have any power to do so. You might, though, not sure.

ahedin25, 2023-05-04:
Reopen

nordic-krch, 2025-05-06:
Are you using shell as logging backend? If so then deferred logging is more complex because shell is processing/printing from own thread. When logging thread is passing a log message to shell then shell creates own log message copy. Shell log buffer is configurable (512 by default).  `CONFIG_SHELL_BACKNED_SERIAL_LOG_MESSAGE_QUEUE_SIZE` is the setting.

closed 2025-08-15 by github-actions[bot]