zephyrproject-rtos/zephyr #51821: native_posix: Cmake must be run at least twice to find ${CMAKE_STRIP}

marc-hb, 2022-10-31:
**Describe the bug**
`native_posix` makes ${CMAKE_STRIP} non idempotent / non-"deterministic". 

**To Reproduce**
Reproduction with recent Zephyr commit b89a3d955ab3

```shell
$ rm -rf build/
$ cmake -GNinja -Bbuild -Ssamples/hello_world -DCONFIG_BUILD_OUTPUT_STRIPPED=y -DBOARD=native_posix 

$ ninja -C build/ # FAIL cause CMAKE_STRIP is missing in build.ninja

... && /usr/bin/readelf -e zephyr.elf > zephyr.stat && --strip-all zephyr.elf -ozephyr.strip

/bin/sh: 1: --strip-all: not found
```

When running cmake twice or more the build succeeds:

```
$ rm -rf build/
$ cmake -GNinja -Bbuild -Ssamples/hello_world -DCONFIG_BUILD_OUTPUT_STRIPPED=y -DBOARD=native_posix
$ cmake -GNinja -Bbuild -Ssamples/hello_world -DCONFIG_BUILD_OUTPUT_STRIPPED=y -DBOARD=native_posix # twice or more

$ ninja -C build/ # PASS
$ file build/zephyr/zephyr.strip 

  build/zephyr/zephyr.strip: ELF 32-bit LSB executable,
```

With `grep -e CMAKE_STRIP -e zephyr.strip build/*` one can see that  `CMAKE_STRIP` is correctly defined in `build/CMakeCache.txt` even when running cmake only once; however the value is actually used in `build/build.ninja` only when running cmake twice or more.

This seems to affect only `native_posix` which is obviously an "unusual" toolchain.

Same reproduction with GNU Make instead of ninja.

**Expected behavior**

I don't know whether `/usr/bin.strip` is acceptable for `native_posix` or not but for sure `cmake` should always behave the same and always accept it or always reject it no matter how many times it is run. CMake issues are hard enough to debug already even when cmake is idempotent (and CMake expertise is rare); it took me a couple hours of trial and error to isolate the reproduction above.

**Impact**

- Annoyance: impossible to use CONFIG_BUILD_OUTPUT_STRIPPED with `native_posix`
- Who knows: possibly much worse CMake impact with the same, extremely confusing user experience.

**Logs and console output**

See above

**Environment (please complete the following information):**

Ubuntu Linux

cc: @tejlmand , @mbolivar , @aescolar , @stephanosio , @dcpleung , @daor-oti , @mped-oticon

marc-hb, 2022-10-31:
From https://opensource.intel.com/jira/servicedesk/customer/portal/1/DIO-10430
```
From: Gretzinger, Adam R
Sent: Monday, October 31, 2022 09:06
To: domainadmin <domainadmin@intel.com>
Subject: sof-ci.01.org - DOM-6087
```

> sof-ci.01.org did not get its DNS change thru in time for the end of our hosting contract, and was left offline all weekend.
> I did make notes in the ticket about the urgency, but they seemed to have not helped. Can you push this one to immediate configuration?
> 
> https://digitalrequest.intel.com/servicedesk/customer/portal/9/DOM-6087
>

marc-hb, 2022-10-31:
```
From: domainadmin <domainadmin@intel.com>
Sent: Monday, October 31, 2022 12:27
To: Gretzinger, Adam R <adam.r.gretzinger@intel.com>
Cc: Vargo, Paul <paul.vargo@intel.com>; Holland, James <james.holland@intel.com>
Subject: RE: sof-ci.01.org - DOM-6087
```

> Hi Adam,
> 
> Paul from our DNS Engineering team assigned this to himself 2 hours ago. He will get this completed.
>

marc-hb, 2022-10-31:
Workaround submitted as part of
- #51737

tejlmand, 2022-11-01:
Fixed here: #51837

closed 2022-11-01 by carlescufi

marc-hb, 2022-11-01:
While https://github.com/zephyrproject-rtos/zephyr/pull/51837 made the problem "go away", I think this bug exposed a more generic problem where the toolchain abstraction code is interfering and racing with built-in CMake variables. See my last comment at https://github.com/zephyrproject-rtos/zephyr/pull/51837#issuecomment-1299252613

PR #51837 (2022-11-01): cmake: lookup strip tool and set CMAKE_STRIP for host-gnu target
Fixes: #51821

Set CMAKE_STRIP using `find_program(CMAKE_STRIP strip)` to support strip when building on native posix.
commit 1ee86a88 (2022-11-01): cmake: lookup strip tool and set CMAKE_STRIP for host-gnu target

Fixes: #51821

Set CMAKE_STRIP using `find_program(CMAKE_STRIP strip)` to support
strip when building on native posix.