zephyrproject-rtos/zephyr #51825: west: runners: jlink: JLink.exe name collision

JordanYates, 2022-11-01:
**Describe the bug**

The `jlink` binary runner uses the location of `JLink.exe` or `JLinkExe` to determine the location of the JLink libraries, 

https://github.com/zephyrproject-rtos/zephyr/blob/117039ca2b7202fb08ea7cc4bcc35ca5a015241c/scripts/west_commands/runners/jlink.py#L26
https://github.com/zephyrproject-rtos/zephyr/blob/117039ca2b7202fb08ea7cc4bcc35ca5a015241c/scripts/west_commands/runners/jlink.py#L38
https://github.com/zephyrproject-rtos/zephyr/blob/117039ca2b7202fb08ea7cc4bcc35ca5a015241c/scripts/west_commands/runners/jlink.py#L177-L178

Unfortunately the OpenJDK runtime also distributes an executable called `jlink.exe` : https://strontic.github.io/xcyclopedia/library/jlink.exe-15C86A7D8376EC476DD2AE27EAEE31D1.html

Trying to run `west flash` with the runtime on the path results in errors like:
```
packages\pylink\library.py", line 286, in __init__
    self.load(dllpath)
  File "C:\Python310\lib\site-packages\pylink\library.py", line 392, in load
    with open(self._path, 'rb') as inputfile:
FileNotFoundError: [Errno 2] No such file or directory: 'C:\\Program Files\\Eclipse Adoptium\\jdk-11.0.16.101-hotspot\\bin\\JLink_x64.dll'
```

**Expected behavior**

The jlink plugin should ideally be more resilient to finding a SEGGER executable, not any old executable called `jlink.exe`.

**Impact**

Annoyance, the correct path can be selected with `west flash --commander /path/to/JLink.exe`

**Environment (please complete the following information):**
 - OS: Windows

ehughes, 2024-06-21:
@JordanYates FYI. Thanks for this. I ran into this as well.

Xentax5, 2024-08-14:
@JordanYates Also thanks. Same issue here.

dleach02, 2024-08-15:
We need a different assignee. Does someone on this ticket want to take this on?

henry-202308, 2024-09-05:
I have also encountered this problem, and a temporary solution is: 
Add the required Jlink.exe path to the first line of the environment variable PATH.

PR #83562 (2025-01-08): runners: jlink: win32: search for valid `JLink.exe`
Since "JLink.exe" is also an executable distributed with JDK's, do an explicit search on the standard SEGGER install directories for a JTAG "JLink.exe" before falling back to whatever is first on PATH.

Fixes #51825.
commit 5f0c8005 (2025-01-08): runners: jlink: win32: search for valid `JLink.exe`

Since "JLink.exe" is also an executable distributed with JDK's, do an
explicit search on the standard SEGGER install directories for a JTAG
"JLink.exe" before falling back to whatever is first on PATH.

Fixes #51825.
closed 2025-01-08 by kartben