zephyrproject-rtos/zephyr #51827: picolibc heap lock recursion mismatch

andyross, 2022-11-01:
picolibc's heap implementation has an apparent lock nesting bug.  When malloc() fails, it looks like it tries to unlock its internal lock twice.  You can see this by applying the patch below to instrument the locking hooks, then building the following app:

```c
#include <zephyr/kernel.h>
#include <stdlib.h>

void main(void)
{
	void *p = malloc(0x10000000);

	printk("malloc returned %p\n", p);
}
```

Note that you'll see the same lock acquired, then an attempt to release it twice.  This is benign with current k_mutex (because it returns an error that the libc hooks ignore!), but with zync it hits an __ASSERT and panics.  Ex.:

```
*** Booting Zephyr OS build zephyr-v3.2.0-507-g69e628a6628f  ***
__retarget_lock_acquire_recursive(0x2000004c)
__retarget_lock_release_recursive(0x2000004c)
__retarget_lock_release_recursive(0x2000004c)
malloc returned 0
```

Instrumentation patch:

```patch
diff --git a/lib/libc/picolibc/libc-hooks.c b/lib/libc/picolibc/libc-hooks.c
index 4bdce81d3d..64005147cb 100644
--- a/lib/libc/picolibc/libc-hooks.c
+++ b/lib/libc/picolibc/libc-hooks.c
@@ -454,6 +454,7 @@ void __retarget_lock_close(_LOCK_T lock)
 /* Close dynamic recursive lock */
 void __retarget_lock_close_recursive(_LOCK_T lock)
 {
+       printk("%s(%p)\n", __func__, (void *)lock);
        __ASSERT_NO_MSG(lock != NULL);
 #ifndef CONFIG_USERSPACE
        free(lock);
@@ -465,6 +466,7 @@ void __retarget_lock_close_recursive(_LOCK_T lock)
 /* Acquiure non-recursive lock */
 void __retarget_lock_acquire(_LOCK_T lock)
 {
+       printk("%s(%p)\n", __func__, (void *)lock);
        __ASSERT_NO_MSG(lock != NULL);
        k_sem_take((struct k_sem *)lock, K_FOREVER);
 }
@@ -472,6 +474,7 @@ void __retarget_lock_acquire(_LOCK_T lock)
 /* Acquiure recursive lock */
 void __retarget_lock_acquire_recursive(_LOCK_T lock)
 {
+       printk("%s(%p)\n", __func__, (void *)lock);
        __ASSERT_NO_MSG(lock != NULL);
        k_mutex_lock((struct k_mutex *)lock, K_FOREVER);
 }
@@ -479,6 +482,7 @@ void __retarget_lock_acquire_recursive(_LOCK_T lock)
 /* Try acquiring non-recursive lock */
 int __retarget_lock_try_acquire(_LOCK_T lock)
 {
+       printk("%s(%p)\n", __func__, (void *)lock);
        __ASSERT_NO_MSG(lock != NULL);
        return !k_sem_take((struct k_sem *)lock, K_NO_WAIT);
 }
@@ -486,6 +490,7 @@ int __retarget_lock_try_acquire(_LOCK_T lock)
 /* Try acquiring recursive lock */
 int __retarget_lock_try_acquire_recursive(_LOCK_T lock)
 {
+       printk("%s(%p)\n", __func__, (void *)lock);
        __ASSERT_NO_MSG(lock != NULL);
        return !k_mutex_lock((struct k_mutex *)lock, K_NO_WAIT);
 }
@@ -493,6 +498,7 @@ int __retarget_lock_try_acquire_recursive(_LOCK_T lock)
 /* Release non-recursive lock */
 void __retarget_lock_release(_LOCK_T lock)
 {
+       printk("%s(%p)\n", __func__, (void *)lock);
        __ASSERT_NO_MSG(lock != NULL);
        k_sem_give((struct k_sem *)lock);
 }
@@ -500,6 +506,7 @@ void __retarget_lock_release(_LOCK_T lock)
 /* Release recursive lock */
 void __retarget_lock_release_recursive(_LOCK_T lock)
 {
+       printk("%s(%p)\n", __func__, (void *)lock);
        __ASSERT_NO_MSG(lock != NULL);
        k_mutex_unlock((struct k_mutex *)lock);
 }
```

keith-packard, 2022-11-01:
Thanks; https://github.com/picolibc/picolibc/pull/366 should fix this.

andyross, 2022-11-01:
Indeed, looks great.  What's the process for getting the zephyr west manifest updated?  Is there a regular cadence I can wait for?  Right now I'm carrying a workaround in the zync tree, so it's not urgent, but would be good to not have to merge that.

keith-packard, 2022-11-01:
We don't have a regular cadence, and picolibc releases are pretty much a 'when things seem useful to get released' time frame. There are a bunch of minor fixes that get picolibc support working for all of the Zephyr architectures, both in the module and the SDK, so getting those merged might be useful even before the changes are included in a picolibc release?

PR #53337 (2023-01-27): modules: Update picolibc to 1.8
Closes: #51827
closed 2023-01-27 by cfriedt