zephyrproject-rtos/zephyr #51829: qemu_x86: upgrading to q35 breaks networking samples.

hakehuang, 2022-11-01:
**Describe the bug**
with this commit, samples/net/socket/echo_server fails to run
```
commit a2aa462f7d099bf30c1e054a1f38d34359d83418 (refs/bisect/bad)
Author: Andrei Emeltchenko <andrei.emeltchenko@intel.com>
Date:   Thu Oct 27 18:54:13 2022 +0300

    qemu_x86: Use Qemu Q35 machine feature (ICH9)

    Using very old machine does not make sense anymore. Switch to the new
    Q35 (https://wiki.qemu.org/Features/Q35). Among other things it allows
    to test SMBus and watchdog.

    Signed-off-by: Andrei Emeltchenko <andrei.emeltchenko@intel.com>
```

diff --git a/boards/x86/qemu_x86/board.cmake b/boards/x86/qemu_x86/board.cmake
index b4f7c03639..934b078154 100644
--- a/boards/x86/qemu_x86/board.cmake
+++ b/boards/x86/qemu_x86/board.cmake
@@ -60,6 +60,7 @@ endif()
 set(QEMU_FLAGS_${ARCH}
   -m ${QEMU_MEMORY_SIZE_MB}
   -cpu ${QEMU_CPU_TYPE_${ARCH}}${QEMU_CPU_FLAGS}
+  -machine q35
   -device isa-debug-exit,iobase=0xf4,iosize=0x04
   ${REBOOT_FLAG}
   -nographic
```

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. west build -b qemu_x86 -- -DCONF_FILE=prj.conf -DOVERLAY_CONFIG="overlay-e1000.conf"
3. west build -t run
4. See error
```
-- west build: running target run
[0/1] To exit from QEMU enter: 'CTRL+a, x'[QEMU] CPU: qemu32,+nx,+pae
SeaBIOS (version zephyr-v1.0.0-0-g31d4e0e-dirty-20200714_234759-fv-az50-zephyr)

iPXE (http://ipxe.org) 00:02.0 CA00 PCI2.10 PnP PMM+00392120+002F2120 CA00

Booting from ROM..
FAILED: zephyr/CMakeFiles/run_qemu /home/shared/disk/zephyr_project/zephyr_test/zephyr/samples/net/sockets/echo_server/build/zephyr/CMakeFiles/run_qemu
cd /home/shared/disk/zephyr_project/zephyr_test/zephyr/samples/net/sockets/echo_server/build && /home/ubuntu/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/qemu-system-i386 -m 4 -cpu qemu32,+nx,+pae -machine q35 -device isa-debug-exit,iobase=0xf4,iosize=0x04 -no-reboot -nographic -no-acpi -nic tap,model=e1000,script=no,downscript=no,ifname=zeth -pidfile qemu.pid -chardev stdio,id=con,mux=on -serial chardev:con -mon chardev=con,mode=readline -kernel /home/shared/disk/zephyr_project/zephyr_test/zephyr/samples/net/sockets/echo_server/build/zephyr/zephyr.elf
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/local/bin/cmake --build /home/shared/disk/zephyr_project/zephyr_test/zephyr/samples/net/sockets/echo_server/build --target run

```

**Expected behavior**
the qemu_x86 can start successfully

**Impact**
showstopper

**Environment (please complete the following information):**
 - OS: (e.g. Linux, , )
 - Toolchain (e.g Zephyr SDK, )
 - Commit SHA or Version used: a2aa462f7d099bf30c1e054a1f38d34359d83418 

**Additional context**
this impact the maxpro networking tcpip testing, as we testing tcpip stack on qemu_x86 platforms

hakehuang, 2022-11-01:
@dleach02, @rlubos , please help to highlight this issue as this is a showblocker for MAX Pro TCPIP testing.

finikorg, 2022-11-01:
@hakehuang , I will investigate this, sadly our automatic tests did not notice the issue.

PR #51845 (2022-11-03): e1000: Fix hardcoded BDF
Correct e1000 hardcoded BDF, making it possible to use the same driver when Qemu parameters change.

Fixes #51829
commit b5fd1b22 (2022-11-03): drivers: e1000: Change PCIE BDF for Q35 qemu

Since moving to Qemu Q35 machine the drivers with hardcoded BDF stop
working. Correct e1000 BDF.

Fixes #51829
closed 2022-11-03 by MaureenHelm