zephyrproject-rtos/zephyr #51831: Hardware model v2

mbolivar-nordic, 2022-11-01:
# Is your enhancement proposal related to a problem?

There is no way to define a single [board directory](https://docs.zephyrproject.org/latest/hardware/porting/board_porting.html#create-your-board-directory) or SoC directory for [AMP](https://en.wikipedia.org/wiki/Asymmetric_multiprocessing) SoCs that have CPUs with different architectures.

This is because of the artificial file system hierarchy restriction that requires SoC and board porters to provide the architecture via the file system, e.g.:

- `<SOC_ROOT>/soc/<ARCH>/<SOC_FAMILY>/...`
- `<BOARD_ROOT>/boards/<ARCH>/<BOARD_NAME>/...`

This [hardware model](https://docs.zephyrproject.org/latest/hardware/porting/board_porting.html#boards-socs-etc) has served us well for years when SoCs with Zephyr support were simple microcontrollers with a single CPU or single SMP cluster. The industry is increasingly moving towards heterogeneous AMP SoCs, however, and Zephyr needs to keep up if we wish to continue to serve our users well.

# Describe the solution you'd like

A backwards-compatible "hardware model v2". That is, a file system hierarchy standard supported by Zephyr's build system that lets you:

- support an SoC family etc. in a single directory regardless of the number of architectures (1, 2, ...) involved
- support a board in a single directory for such SoCs

This would look something like this (details in the `...` TBD):

- `<SOC_ROOT>/soc/v2/<VENDOR>/<SOC_FAMILY>/...`
- `<BOARD_ROOT>/boards/v2/<VENDOR>/<BOARD_NAME>/...`

That is, we allow both in-tree and out-of-tree users to define `v2` subdirectories of both `soc` and `boards`. These respectively contain SoC and board definitions which support as many simultaneous architectures as are required.

Hardware model v2 will support **both** "multi-architecture" and "single-architecture" SoCs to allow for an eventual migration path if the project ever wishes to deprecate the current hardware model. However, deprecation and migration is a non-goal for this initial effort: as stated above, the current model has served us well for years, and there seems to be no reason to force satisfied users away from it.

The `<VENDOR>` path components shown will be namespace controlled using the existing [vendor-prefixes.txt](https://github.com/zephyrproject-rtos/zephyr/blob/main/dts/bindings/vendor-prefixes.txt) file format already supported by Zephyr's devicetree tooling. Users may already provide their own vendor-prefixes.txt files in individual modules, so this mechanism gives us full control over what we support in the zephyr repository itself and allows for extensibility.

# Describe alternatives you've considered

None exist (prove me wrong 😄 ).

# Change history

- 2022-10-31: initial proposal describing top level hierarchies and namespacing. Many details TBD.

tejlmand, 2024-01-16:
Slides presented at arch wg meeting
[Zephyr_arch_wg_HWMv2.pptx](https://github.com/zephyrproject-rtos/zephyr/files/13953224/Zephyr_arch_wg_HWMv2.pptx)

mmahadevan108, 2024-01-31:
@carlescufi, can you please confirm if we plan to have both v1 and v2 for 3.7 LTS, this is stated as a `may`

`Zephyr 3.7.0 LTS may ship with a combination of v1 and v2 SoCs and boards, since it is possible that not all of them will have transitioned before its release. Much of the SoC and board code and files are very similar between the two revisions, so this should not be a big issue for backporting fixes, even with SoCs and boards that transition to v2 after LTS release`

carlescufi, 2024-03-03:
Merged here: https://github.com/zephyrproject-rtos/zephyr/pull/69607

PR #50305 (2023-12-22): Introduction of board and SoC scheme v2.
This PR introduces a board and SoC scheme v2.

Fixes #51831

The benefits of board and SoC scheme v2 are:
- Inherit `BOARD` setting from CMake
    -  https://github.com/zephyrproject-rtos/zephyr/pull/49260#issuecomment-1220381626
    - https://github.com/zephyrproject-rtos/zephyr/pull/44654#discussion_r909335964
    - This aligns how `BOARD` is passed from CMake --> Kconfig to the way `BOARD_REVISION` is handled
    - this removes any inconsistency in names
    - Removes the need to keep names in sync, and updating them in two places.
- Remove the need  [... cut]
closed 2024-03-03 by carlescufi