zephyrproject-rtos/zephyr #51832: Sysbuild support for system DT and hardware model v2

mbolivar-nordic, 2022-11-01:
# Is your enhancement proposal related to a problem?

Zephyr's [sysbuild](https://docs.zephyrproject.org/latest/build/sysbuild/index.html) feature lets users set up and develop with multiple interrelated Zephyr- and non-Zephyr application directories. If my application directory has sysbuild integration, I can use it to build, flash, and debug all of the applications at once with command lines like this example:

```
$ west build --sysbuild -b board my-top-level-app-dir # builds multiple applications
$ west flash # flashes all the application images to my board
```

If Zephyr adopts system devicetree (see  #51830) and a hardware model "v2" (see #51831), I will still not have a convenient way to build multiple interrelated applications using these features.

# Describe the solution you'd like

Sysbuild awareness of system devicetree, especially the ability to apply system devicetree overlays at sysbuild's CMake time, to hardware (socs / boards) supported via hardware model v2.

At a high level, I would like sysbuild to be able to manage the portions of this diagram with a white background, creating and managing the individual zephyr build systems in the shaded boxes:

[image]

I would like this enhancement to be supported in a backwards compatible way: no existing sysbuild use cases with hardware model v1 and "regular" devicetree should be broken.

I would like this sysbuild enhancement to work with boards that use system devicetree and boards that do not require system devicetree. I would also like this to work with boards in hardware model  v1 and v2.

# Describe alternatives you've considered

Continuing to use multiple regular `west build` commands to generate individual build systems, making sure to apply the same system devicetree overlays to each one, and flashing the individual zephyr application images to my board via multiple `west flash` commands.

# Change history

- 2022-10-31: Many details are TBD given the lack of implementation details being settled in system devicetree and hardware model v2 at the present time, so this is mainly a placeholder issue for now.

mbolivar-nordic, 2023-05-05:
Split the system DT portions out. I will file a separate issue for these.

carlescufi, 2024-06-06:
This is now complete

closed 2024-06-06 by carlescufi