zephyrproject-rtos/zephyr #51833: Better support for multi-core AMP SoCs

mbolivar-nordic, 2022-11-01:
# Is your enhancement proposal related to a problem?

Zephyr's support for [AMP](https://en.wikipedia.org/wiki/Asymmetric_multiprocessing) SoCs... works, but has several well-known usability problems due to workarounds for deficiencies in our hardware support features at the following layers:

- [devicetree](https://docs.zephyrproject.org/latest/build/dts/index.html)
- [hardware model](https://docs.zephyrproject.org/latest/hardware/porting/board_porting.html#boards-socs-etc) 
- [build system](https://docs.zephyrproject.org/latest/build/cmake/index.html)

This is an umbrella issue that tracks the combined usage of several related enhancements to Zephyr's hardware support which are described further in individual issues listed below.

# Describe the solution you'd like

I would like Zephyr to:

- support system devicetree: see #51830
- define a new, future-proof hardware model for AMP SoCs: see #51831
- allow using sysbuild as a convenient entry point to the above: see #51832

I would like the implementation to be backwards compatible in the sense that merging it shall not break existing users of our current hardware support features, and shall be opt-in at the SoC (and therefore board) level.

# Describe alternatives you've considered

Nordic previously tried to solve these problems in #13672 in 2019. This was rejected upstream for (valid) reasons. We have learned from this initial work and believe the above proposal can address the limitations and issues that led to its rejection.

# Change history

- 2022-10-31: initial proposal

carlescufi, 2023-03-28:
Architecture WG:

- Presentation by @mbolivar-nordic and @tejlmand
- @nashif worries about a 2-tier system for both Devicetree and boards: System DT vs DT and hw model v1 vs hw model v2
- @galak agrees with @nashif that it might be confusing for users to have to deal with multiple systems
- @henrikbrixandersen thinks that both (old and new) hw models and sysdt vs dt should coexist

mbolivar-nordic, 2023-05-05:
I've moved this to v3.5 and made a note that system devicetree is not expected as part of the initial implementation.