zephyrproject-rtos/zephyr #51834: Support for "mixins" (formerly "snippets")

mbolivar-nordic, 2022-11-01:
# Is your enhancement proposal related to a problem?

Zephyr developers frequently need to provide interrelated devicetree overlays and Kconfig fragments when configuring applications. These interrelated configuration overrides are often specific to the target hardware, even when they perform the same "logical" change.

There is currently no way other than [shields](https://docs.zephyrproject.org/latest/hardware/porting/shields.html) to pair these related configuration overrides together in a way that is searchable across modules, customizable on a per-board basis, and able to be passed to e.g. `west build` with a single name.  However, shields are not an appropriate feature for overriding the default configuration on an existing board without involvement of daughter boards. This feature gap creates usability problems.

## Example use case: CDC-ACM console

As a user, I want to override the default UART console configuration on my board with a CDC-ACM console.

As a part of this, I need to both:

- provide a devicetree overlay overriding the default `zephyr,console` [chosen node](https://docs.zephyrproject.org/latest/build/dts/api/api.html#zephyr-specific-chosen-nodes)
- ensure related Kconfig settings, such as setting `CONFIG_USB_DEVICE_STACK=y`

I want to keep these related settings in  a single place within my west workspace, so that I can build multiple applications with the default console overridden. I want this "single place" to support storing the same logical configuration overrides for different boards.

I can currently do this "manually" with command lines like this:

```
$ west build -b board-1 app-1 -- \
    -DDTC_OVERLAY_FILE=/path/to/cdc-acm-console/board-1.overlay \
    -DOVERLAY_CONFIG=/path/to/cdc-acm-console/board-1.conf
$ west build -b board-2 app-1 -- \
    -DDTC_OVERLAY_FILE=/path/to/cdc-acm-console/board-2.overlay \
    -DOVERLAY_CONFIG=/path/to/cdc-acm-console/board-2.conf

$ west build -b board-1 app-2 -- \
    -DDTC_OVERLAY_FILE=/path/to/cdc-acm-console/board-1.overlay \
    -DOVERLAY_CONFIG=/path/to/cdc-acm-console/board-1.conf
$ west build -b board-2 app-2 -- \
    -DDTC_OVERLAY_FILE=/path/to/cdc-acm-console/board-2.overlay \
    -DOVERLAY_CONFIG=/path/to/cdc-acm-console/board-2.conf
```

This is very annoying to me as a user. (It is also difficult to document for other personas like "Zephyr maintainer" or "downstream Zephyr distributor".)

A proposal to handle this using SHIELD was raised in https://github.com/zephyrproject-rtos/zephyr/pull/40645 but rejected at dev-review for the reason described above: this is not a "shield" in the sense that term is usually defined in this context.

# Describe the solution you'd like

A way to place related Kconfig fragments and DT overlays within my workspace:

- as named entities (like `cdc-acm-console`)
- in a way the build system can search for and discover
- in a way that can be specialized for individual boards as needed

We have previously called these entities "snippets", see e.g. #40669. Following some discussion in `#architecture` on Discord, I am provisionally renaming these "mixins".

I am intentionally going to keep the implementation details out of this description for now and keep it high level.

## Example solution usage: CDC-ACM

As a user, I would like to be able to do the following instead of the above, and have the same effects:

```
$ west build -b board-1 --mixin cdc-acm-console app-1
$ west build -b board-2 --mixin cdc-acm-console app-1

$ west build -b board-1 --mixin cdc-acm-console app-2
$ west build -b board-2 --mixin cdc-acm-console app-2
```

## Extensible via modules

As a user, I would like to be able to define "mixins" in my own modules, and have them added to the search path used by the build system via existing [module](https://docs.zephyrproject.org/latest/develop/modules.html) mechanisms, and have those "mixins" picked up by default by the build system when I use [west build](https://docs.zephyrproject.org/latest/develop/west/index.html).

I additionally want to be able to extend existing "mixins" defined elsewhere. For example, if the zephyr repository defines a standard `cdc-acm-console` mixin, I want to be able to extend it with support for my board by adding files to a module I control, without any changes to the zephyr repository itself.

## Multiple mixins at once

As a user, I want to be able to combine multiple "mixins" in a single application build. For example, if I want to both use a CDC-ACM console and override the default settings partition for my board to one on an external flash IC, I should be able to do this:

```
$ west build -b board --mixin cdc-acm-console --mixin my-settings-partition app
```

## West not required

As a user, I want to be able to use the "mixins" feature even if I am not using west. I should be able to do this by defining a CMake variable that declares the mixins I want to use. For example:

```
cmake ... -DBOARD=board -DMIXINS=cdc-acm-console;my-settings-partition
ninja ...
```

## Future directions: per-SoC configuration

As a user, I often want to specify "mixins" on a per-SoC basis, not a per-board basis. This occurs whenever the needed overrides are always the same for any two boards with the same SoC. The CDC-ACM console is a good example here too: it is common for microcontroller SoCs to only have a single USB device peripheral, so the "mixin" for any two boards with the same such SoC is going to be the same.

This is difficult to do within the current build system, but should be possible to address as follow-on work if the Zephyr project adopts build system changes to make SoCs more "first class citizens". This is intentionally left out of scope of the initial work to avoid muddying the waters in this issue.

# Describe alternatives you've considered

- #40669
- #40645

# Additional context

The original discussion was started by @JordanYates at Zephyr Developers Summit 2021. The community at the time agreed the idea was solid, but we haven't been able to agree on specifics.

# Change history:

- 2022-10-31: initial proposal

jfischer-no, 2022-11-07:
I assume "mixins" is just a project name, temporary until something better is found.

aborisovich, 2023-03-26:
Reason for implementing "Snippets" seems perfectly clear - end user should have a possibility to gather a set of needed configurations that alter a board definition under single name. However, I would like some concerns.

### The naming
Personally do not agree with the term you are trying to use. I will not elaborate how important is the matter of giving proper names to abstractions we create, as every developer understands this. So, the name "snippets" does not indicate in any way what it actually is:
* name "snippet" does not indicate that it is a set (a group of configurations given common name)
* name "snippet" does not indicate that it is related to pre-build configuration
* name "snippet" misleadingly indicates that is it something lightweight  and temporary (one of the things that comes to my mind is cut out part of text or code to be reused/pasted somewhere else), while it is actually something very static and of key importance 

I would suggest some alternative names:
1. a "config_group" - this name perfectly clear indicates what it is - a set of configurations that were given a name. However, it may lead to confusion whether a "group" is a "collection" or maybe something we "assign to".
2. a "config_set" - this name also indicates clearly what it is, but may be confused with the meaning of "set" part (somebody can think of it as _setting_ something).
3. a "config_bundle" -   looks intuitive. "Bundle" word seems not to have too much alternative meanings and the verb has quite similar meaning to a noun. I would go for this one personally.

### Feature of inclusion & exclusion
CMakePresets you mentioned suffer from the [combinatory explosion](https://gitlab.kitware.com/cmake/cmake/-/issues/22538) issue. [SOF](https://github.com/thesofproject/sof) project I contribute to also suffers from lack of this (snippets) feature. Let's think about a concrete example:
_"User would like to build a firmware for MTL board, to be run on FPGA, with profiling features enabled and disabled logging"_.
Some context:
* MTL is a board (`intel_adsp_ace15_mtpm.conf`) that contains set of configurations to build production fw for this hardware (for silicon).
* We use overlay `fpga_overlay.conf` file that changes fws clock speed to be run on FPGA
* We use an overlay `profiling_overlay.conf` file to perform performance tests and include some special profiling code in fw and disable logging to have more accurate performance results.

So in this scenarion we have 2 overlays added to the build.
Lets say, that we use the "Snippets" you are proposing and give it a name "mtl_fpga_profiling.snippet" and this file contains information that we want to use `fpga_overlay.conf` and `profiling_overlay.conf` files for the build.
After some later time, I'm developing a new driver version that is supposed to replace old driver version. I would like a build fw with my new driver enabled, old one disabled and run performance tests on FPGA. Let's say I've created a new overlay `driverA_overlay.conf` (that contains `CONFIG_DRIVER_A=n`, `CONFIG_NEW_DRIVER_A=y`).
Next, I would create a "Snippet" to build my set using a single name, lets call it "mtl_fpga_new_driver_a_profiling.snippet".
And here we came to my:
* **feature request**: snippet files may include (or exclude too!) other snipper files (resolved by the order of occurrence) to avoid combinatory explosion. So in my example I would like my "mtl_fpga_new_driver_a_profiling.snippet" to include "mtl_fpga_profiling.snippet" and use additional `driverA_overlay.conf` overlay.

nordicjm, 2023-03-27:
> feature request: snippet files may include (or exclude too!)

I don't personally see the merit in excluding files, if you want a different set of files, create a new snippet. Likewise in a project with build files, if you want to exclude files, create a new project or create a separate configuration directory and set APP_CONFIGURATION_DIR to point to those files instead of hacking around cmake files trying to alter zephyr variables.

aborisovich, 2023-03-27:
> I don't personally see the merit in excluding files, if you want a different set of files, create a new snippet. Likewise in a project with build files, if you want to exclude files, create a new [...]

Hmmm yes, maybe excluding file does not have that much sense indeed. Inclusion would suffice.

mbolivar-nordic, 2023-03-27:
@aborisovich thanks for your input.

I see your point that people might initially think snippets are more trivial than they really are, but I don't think this is going to be a real problem in practice once we have several nontrivial built-in examples. I personally find the name "config_bundle" too verbose and too similar to other names. I wanted a name that clearly stood out and would not be easy to mistake for some other build system feature.

That said, we haven't made any zephyr release with this feature as it is currently named, so if this is a serious issue for you, please escalate to architecture WG or dev-review to see if others agree with you.

Regarding your feature request, it seems like a reasonable extension. I guess the main things to get right would be to detect or avoid circular includes, deciding on semantics if snippets A and B both include C, etc. Perhaps you can propose an extension to the YAML syntax and associated semantics that would resolve these types of issues.

gregersrygg, 2023-05-25:
I have a question regarding «Future work: board-revision-specific snippets»

The proposed syntax is:

```
name: foo
boards:
  bar:
    revision: 1.2.3
      DTC_OVERLAY_FILE: bar-1.2.3.overlay
```

Today it's possible for an application to specify a board overlay for a revision defined in revision.cmake for that board. The user can build for any board revision, and cmake will resolve to the last revision before or equal to that revision. So if the revision.cmake defines `VALID_REVISIONS 0.5.0 1.2.3`, then a user building with `west build -b bar@1.3.7` would resolve `1.2.3` as the last valid revision and look for an overlay named `bar_1_2_3.overlay`. This is very useful when there are only a few revisions where hardware have changed and requires an overlay, but makes it easy for the user when they only need to specify the version of their board without having to know what are the available versions defined as overlays in the board folder.

My question is wether the board revision logic for snippets will work in a similar manner? Where the user specified board revision is resolved to one of the cmake valid revisions before checking for a snippet board revision.

nashif, 2024-05-18:
snippets feature already implemented, closing.

closed 2024-05-18 by nashif