zephyrproject-rtos/zephyr #51855: twister: targeting wrong serial port / device

leocencetti, 2022-11-01:
**Describe the bug**

We have a Linux machine with two connected targets for our CI validation pipeline:
- A `Nucleo H743ZI` board.
- An `OpenMV H7` board, connected via STLink V3 programmer.

We use custom `udev` rules to ensure that the serial ports are always enumerated as follows:
- `Nucleo H743ZI` --> `/dev/nucleo_h743zi`
- `OpenMV H7` (STLink) --> `/dev/openmv_h7`

These rules rely on the serial number of the connected device and are therefore bijective.

Now, to the issue: when the two devices are connected to the same USB port through an external USB hub, `twister` seems unable to select the correct device.

For example, running tests for `OpenMV H7`: 
```
zephyr/scripts/twister --ninja -p openmv_h7 -T zephyr/tests/kernel/common/ --device-testing --device-serial /dev/openmv_h7
```
always results in `twister` flashing the `Nucleo` board instead (as if the `/dev/openmv_h7` was discarded). Targeting the `Nucleo` board will (correctly) flash the device at `/dev/nucleo_h743zi`.
The devices are correctly enumerated according to the `udev` rules.

This issue is not present if the two devices are connected to different USB ports (we are able to flash either one correctly).

**To Reproduce**
Steps to reproduce the behavior:
1. Connect two devices to the same USB port via an external USB hub
2. Build unit tests with `twister` and target one of the two devices.
3. `twister` will default to either of the two devices.

**Expected behavior**

Targeting a specific serial port will result in `twister` flashing said port.

**Impact**

Major, disrupted CI workflow

**Environment (please complete the following information):**
 - OS: Linux (Ubuntu 22.04 LTS)
 - Toolchain: Zephyr SDK 0.14.1
 - Zephyr Version: 3.0.0

henrikbrixandersen, 2022-11-01:
Can you reproduce this by using e.g. `west flash` without using twister?

leocencetti, 2022-11-02:
@henrikbrixandersen yes, I have tried to run:
```
west flash -r openocd --serial <serial-number>
```
The same device is selected no matter was is passed as `<serial-number>`, even with completely random values.

henrikbrixandersen, 2022-11-02:
When using `--serial` with the `openocd` west runner, the serial number is passed on to the openocd via the `_ZEPHYR_BOARD_SERIAL` variable. The board `openocd.cfg` file needs to support this.

At least `boards/arm/nucleo_h743zi/support/openocd.cfg` does not support this variable. I don't see the `OpenMV H7` board in-tree.

henrikbrixandersen, 2022-11-04:
> @henrikbrixandersen Ok this explains it. Then selection is done based on order? Do you have a reference for a board configuration that DOES support the `--serial` option?

One example is the NEORV32 board definition: https://github.com/zephyrproject-rtos/zephyr/blob/main/boards/riscv/neorv32/support/openocd.cfg

> It's not there indeed, we created our own configuration based on the `nucleo_h743zi` (they share the same MCU). Nevertheless, from what I understand from the above, this issue would be reproducible for all boards that do not support individual targeting via `--serial` option.

Correct.