zephyrproject-rtos/zephyr #51857: [RFC] mgmt: mcumgr: transport: Replace get MTU function with get info function that returns a struct

nordicjm, 2022-11-01:
### Problem description

There is a function that can be called for SMP transports to get the MTU size, this file is not overly useful by itself, but would become more useful if it was expanded to return a structure instead of just the MTU (this new struct can include the MTU). A use-case for this would be so it is possible to get SMP details from e.g. command handlers, if asynchronous responses were added then it would make sense to get certain information from the transport - is async supported? What is the size of the UD that needs to be stored for an async command handler? This could also be used alongside dynamic plugins (if support is ever added) to query things like version the plugin is designed for.

### Proposed change

Changing the get MTU function to instead return a structure (the full contents TBD) of (read only) information for an SMP transport, this will at a minimum include:
 * MTU size
 * UD (user data) size
 * Maximum number of concurrent connections (for UART this would be 1, but for bluetooth or UDP this can be multiple)
 * If async is supported or not

This will require a stable API change.

## Alternatives

The alternative is to add additional function pointers but this would bloat the size of the application much more than just returning a struct, and it would get unwieldy each time a new parameter was added, requiring downstream applications to be changed each time.

nordicjm, 2022-11-01:
@de-nordic any objections for this RFC?

de-nordic, 2022-11-08:
There is no need to make forward implementation of something we do not know how it will be used, or whether will be used.

These can be added when use case for them is provided, otherwise we are just blindly trying to provide functionality that will require redesign or may be useless in the future, but will be haunting us from the corner of the flash.

nordicjm, 2022-11-09:
> There is no need to make forward implementation of something we do not know how it will be used, or whether will be used.

It would be used for async messaging, a function handler needs to know if async is supported if it uses it, and instead of having to go through stable API changes to add new function pointers every time a new setting needs to be read, it just makes sense to move it to a struct instead, that way additional parameters can be added without impacting users or needed out of tree code changes

closed 2026-09-30 by nordicjm