zephyrproject-rtos/zephyr #51870: Nucleo_h743zi fails to format storage flash partition

jpowen898, 2022-11-01:
**Describe the bug**
The file system fails to format the storage partition on the nucleo_h743zi board. This is failing on both the `samples/subsys/shell/fs` and the `samples/subsys/fs/littlefs`. Also one thing to note is the docs for the littlefs example say that it was created using the nucleo_h743zi board which us the same board I am attempting to test with.

**To Reproduce**

***littlefs Example***
1. `west build -b nucleo_h743zi zephyr/samples/subsys/fs/littlefs -p auto`
2. Build fails with the following error code:
```
[4/16] Building C object CMakeFiles/app.dir/src/main.c.obj
FAILED: CMakeFiles/app.dir/src/main.c.obj 
ccache /home/jowen/Downloads/zephyr-sdk-0.15.1/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -DCORE_CM7 -DHSE_VALUE=8000000 -DKERNEL -DLFS_CONFIG=zephyr_lfs_config.h -DSTM32H743xx -DUSE_FULL_LL_DRIVER -DUSE_HAL_DRIVER -D_FORTIFY_SOURCE=2 -D__PROGRAM_START -D__ZEPHYR__=1 -I/home/jowen/dev/zephyrproject/zephyr/include -I/home/jowen/dev/zephyrproject/build/zephyr/include/generated -I/home/jowen/dev/zephyrproject/zephyr/soc/arm/st_stm32/stm32h7 -I/home/jowen/dev/zephyrproject/zephyr/drivers -I/home/jowen/dev/zephyrproject/zephyr/soc/arm/st_stm32/common -I/home/jowen/dev/zephyrproject/modules/hal/cmsis/CMSIS/Core/Include -I/home/jowen/dev/zephyrproject/modules/hal/atmel/include -I/home/jowen/dev/zephyrproject/zephyr/modules/hal_nxp/. -I/home/jowen/dev/zephyrproject/modules/hal/stm32/stm32cube/stm32h7xx/soc -I/home/jowen/dev/zephyrproject/modules/hal/stm32/stm32cube/stm32h7xx/drivers/include -I/home/jowen/dev/zephyrproject/modules/hal/stm32/stm32cube/stm32h7xx/drivers/include/Legacy -I/home/jowen/dev/zephyrproject/modules/hal/stm32/stm32cube/common_ll/include -I/home/jowen/dev/zephyrproject/modules/fs/littlefs -I/home/jowen/dev/zephyrproject/zephyr/modules/littlefs -isystem /home/jowen/dev/zephyrproject/zephyr/lib/libc/minimal/include -isystem /home/jowen/Downloads/zephyr-sdk-0.15.1/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.1.0/include -isystem /home/jowen/Downloads/zephyr-sdk-0.15.1/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.1.0/include-fixed -fno-strict-aliasing -Og -imacros /home/jowen/dev/zephyrproject/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -gdwarf-4 -fdiagnostics-color=always -mcpu=cortex-m7 -mthumb -mabi=aapcs -mfp16-format=ieee --sysroot=/home/jowen/Downloads/zephyr-sdk-0.15.1/arm-zephyr-eabi/arm-zephyr-eabi -imacros /home/jowen/dev/zephyrproject/zephyr/include/zephyr/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -Werror=implicit-int -fno-pic -fno-pie -fno-asynchronous-unwind-tables -fno-reorder-functions --param=min-pagesize=0 -fno-defer-pop -fmacro-prefix-map=/home/jowen/dev/zephyrproject/zephyr/samples/subsys/fs/littlefs=CMAKE_SOURCE_DIR -fmacro-prefix-map=/home/jowen/dev/zephyrproject/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/home/jowen/dev/zephyrproject=WEST_TOPDIR -ffunction-sections -fdata-sections -std=c99 -nostdinc -MD -MT CMakeFiles/app.dir/src/main.c.obj -MF CMakeFiles/app.dir/src/main.c.obj.d -o CMakeFiles/app.dir/src/main.c.obj -c /home/jowen/dev/zephyrproject/zephyr/samples/subsys/fs/littlefs/src/main.c
In file included from /home/jowen/dev/zephyrproject/zephyr/include/zephyr/arch/arm/aarch32/arch.h:20,
                 from /home/jowen/dev/zephyrproject/zephyr/include/zephyr/arch/cpu.h:19,
                 from /home/jowen/dev/zephyrproject/zephyr/include/zephyr/kernel_includes.h:33,
                 from /home/jowen/dev/zephyrproject/zephyr/include/zephyr/kernel.h:17,
                 from /home/jowen/dev/zephyrproject/zephyr/samples/subsys/fs/littlefs/src/main.c:12:
/home/jowen/dev/zephyrproject/zephyr/include/zephyr/devicetree.h:190:36: error: 'DT_N_NODELABEL_storage_partition_PARTITION_ID' undeclared here (not in a function)
  190 | #define DT_NODELABEL(label) DT_CAT(DT_N_NODELABEL_, label)
      |                                    ^~~~~~~~~~~~~~~
/home/jowen/dev/zephyrproject/zephyr/include/zephyr/devicetree.h:3714:24: note: in definition of macro 'DT_CAT'
 3714 | #define DT_CAT(a1, a2) a1 ## a2
      |                        ^~
/home/jowen/dev/zephyrproject/zephyr/include/zephyr/storage/flash_map.h:292:35: note: in expansion of macro 'DT_FIXED_PARTITION_ID'
  292 | #define FIXED_PARTITION_ID(label) DT_FIXED_PARTITION_ID(DT_NODELABEL(label))
      |                                   ^~~~~~~~~~~~~~~~~~~~~
/home/jowen/dev/zephyrproject/zephyr/include/zephyr/devicetree.h:190:29: note: in expansion of macro 'DT_CAT'
  190 | #define DT_NODELABEL(label) DT_CAT(DT_N_NODELABEL_, label)
      |                             ^~~~~~
/home/jowen/dev/zephyrproject/zephyr/include/zephyr/storage/flash_map.h:292:57: note: in expansion of macro 'DT_NODELABEL'
  292 | #define FIXED_PARTITION_ID(label) DT_FIXED_PARTITION_ID(DT_NODELABEL(label))
      |                                                         ^~~~~~~~~~~~
/home/jowen/dev/zephyrproject/zephyr/samples/subsys/fs/littlefs/src/main.c:271:32: note: in expansion of macro 'FIXED_PARTITION_ID'
  271 |         .storage_dev = (void *)FIXED_PARTITION_ID(storage_partition),
      |                                ^~~~~~~~~~~~~~~~~~
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /home/jowen/dev/zephyrproject/build
```

***shell/fs Example***
1. `west build -b nucleo_h743zi zephyr/samples/subsys/shell/fs -p auto`
2. `west flash`
3. Send the following command over the uart console to the nucleo. `fs mount littlefs /lfs`
4. Mount command fails:
```
*** Booting Zephyr OS build zephyr-v3.2.0  ***

[00:00:00.000,000] <dbg> fs: fs_register: fs register 1: 0
uart:~$ fs mount littlefs /lfs
Error mounting as littlefs: -28
[00:00:19.609,000] <inf> littlefs: LittleFS version 2.4, disk version 2.0
[00:00:19.609,000] <dbg> littlefs: littlefs_flash_init: FS area 1 at 0x20000 for 131072 bytes
[00:00:19.610,000] <inf> littlefs: FS at flash-controller@52002000:0x20000 is 1 0x20000-byte blocks with 512 cycle
[00:00:19.610,000] <inf> littlefs: sizes: rd 16 ; pr 16 ; ca 64 ; la 32
[00:00:19.610,000] <wrn> littlefs: can't mount (LFS -84); formatting
[00:00:19.610,000] <err> littlefs: WEST_TOPDIR/modules/fs/littlefs/lfs.c:567: No more free space 1
[00:00:19.610,000] <err> littlefs: format failed (LFS -28)
[00:00:19.610,000] <err> fs: fs mount error (-28)
uart:~$ 
```

**Environment (please complete the following information):**
OS: Ubuntu 20.04
Toolchain Zephyr SDK 0.15.1
Commit SHA: https://github.com/zephyrproject-rtos/zephyr/commit/4256cd41df6c60f1832fd2deb14edc30ac7debab

FRASTM, 2022-11-02:
The sample is looking for a  "storage_partition" node.

jpowen898, 2022-11-02:
I am using the default `nucleo_h743zi.dts` to compile this project and it looks like it does have a storage partition. Are you wanting me to test with something different than this?

```
&flash0 {
	partitions {
		compatible = "fixed-partitions";
		#address-cells = <1>;
		#size-cells = <1>;

		/* 128KB for bootloader */
		boot_partition: partition@0 {
			label = "mcuboot";
			reg = <0x00000000 DT_SIZE_K(128)>;
			read-only;
		};

		/* storage: 128KB for settings */
		storage_partition: partition@20000 {
			label = "storage";
			reg = <0x00020000 DT_SIZE_K(128)>;
		};

		/* application image slot: 256KB */
		slot0_partition: partition@40000 {
			label = "image-0";
			reg = <0x00040000 DT_SIZE_K(256)>;
		};

		/* backup slot: 256KB */
		slot1_partition: partition@80000 {
			label = "image-1";
			reg = <0x00080000 DT_SIZE_K(256)>;
		};

		/* swap slot: 128KB */
		scratch_partition: partition@c0000 {
			label = "image-scratch";
			reg = <0x000c0000 DT_SIZE_K(128)>;
		};
	};
};
```

FRASTM, 2022-11-03:
Since the nucleo_h743zi.overlay is deleting the node (of the flash0) ` /delete-node/ &storage_partition;`   
another one is required  from the qspi-nor-flash@0 for testing.

Could  you look at https://github.com/zephyrproject-rtos/zephyr/pull/51926

FRASTM, 2022-11-03:
Note that `west build -p auto -b nucleo_h743zi samples/subsys/shell/fs ` does not report any build error

Laczen, 2022-11-03:
@jpowen898, @FRASTM, I don't think it is a good idea to use littlefs on the internal flash of a stm32h743zi: the flash blocks are 128kB and it will need at least 2 blocks to provide any power loss resilience. So at least the storage size should be doubled. Even in that case I am unsure if a littlefs configuration with only 2 eraseable blocks is a good idea.

FRASTM, 2022-11-03:
> @jpowen898, @FRASTM, I don't think it is a good idea to use littlefs on the internal flash of a stm32h743zi: the flash blocks are 128kB and it will need at least 2 blocks to provide any power loss [...]

Do you mean simply preventing that platform (and maybe others with _large_ sector-size) to run that samples on the **internal** flash mem ?
(Then this is to clarify in the README of that sample)

Laczen, 2022-11-03:
I don't want to prevent running the sample on that platform, but it should somehow be made clear to the user that there are limitations arising from using it on devices with large erase-block-size (and limited erase-block-count). I think it would be good to add this to the documentation of the platform. It should be straight forward that any power loss resilient storage solution needs at least 2 erase-blocks (where would the data be stored during an erase ?).

There are users that will be able to work around limitations (e.g. because it is used as read-only, or by adapting the partitions to their needs). But by just enabling the samples users might think there is no limitation at all (because the sample runs), and then after some time get stuck because in real world usage the limitations jump up.

Trying to clarify all limits of a (external) subsystem in a sample IMO seems the wrong place.

jpowen898, 2022-11-04:
> Since the nucleo_h743zi.overlay is deleting the node (of the flash0) ` /delete-node/ &storage_partition;` another one is required from the qspi-nor-flash@0 for testing.
> 
> Could you look at #51926

I'm confused by this PR. It looks like they are implying that there is a qspi flash chip on the nucleo_h743zi to which they are mounting the littlefs file system. From what I can tell looking at schematics for this board there is no qspi chip on it. Is the assumption that you need to buy and wire up a breakout board for that qspi flash chip in order to get the example to work?

closed 2022-11-05 by carlescufi