zephyrproject-rtos/zephyr #51874: Zephyr 3.1 bosch,bme280 device is in the final DTS and accessible, but DT_HAS_BOSCH_BME280_ENABLED=n

chalecampb, 2022-11-02:
**Describe the bug**
Following the steps,

Starting with zephyr 3.1, installed via Nordic's toolchain

In device tree (eg, app.overlay for me)
1. Add the following to add the bme280 sensor to the bus
&i2c1 {
	status = "okay";
	bme280@77 {
		compatible = "bosch,bme280";
		reg = <0x77>;
		status = "okay";
	};
};

Build and confirm in zephyr.dts that the device made it in, and the status is okay. 

Build a quick test with the following to confirm we can access the device at compile time (this was successful)

#if !DT_NODE_EXISTS(DT_PATH(soc, peripheral_50000000, i2c_9000, bme280_77))
#error "whoops"
#endif

In kconfig
1. Enable I2C
2. Enable SENSORS
3. Enable BME280

This is where the bug seems to appear. The BME280 cannot be instantiated because DT_HAS_BOSCH_BME280_ENABLED=n

Searching for DT_HAS_BOSCH_BME280_ENABLED yields several results, but they are all the same. 

DT_COMPAT_BOSCH_BME280 := bosch,bme280
config DT_HAS_BOSCH_BME280_ENABLED
	def_bool $(dt_compat_enabled,$(DT_COMPAT_BOSCH_BME280))

I was speaking with someone in the discord who pointed out that the dt_compat_enabled is evaluated by a python script which just searches for the compatibility strings. Mine is the same as above (above the config, and it matches the compatibility string in the overlay)

Digging deeper, it looks like autoconf.h shows #define CONFIG_DT_HAS_BOSCH_BME280_ENABLED 1. Since autoconf.h is downstream of kconfig in the build flowchart it is not clear to me how kconfig thinks this is false, and also uses a slightly different name (as a layperson there's no way I am going to be able to see the nuance between CONFIG_DT_HAS_BOSCH_BME280_ENABLED  and DT_HAS_BOSCH_BME280_ENABLED...). 

From my perspective the device tree parsing script has some issue which is preventing the compatible flag from being detected by kconfig. 

**To Reproduce**
See above

**Expected behavior**
1. Enable I2C, Sensors,
2. Define the device in the device tree
3. Enable the device in kconfig
4. Use the device

**Impact**
I can't get data from the sensor, so the project can't continue, although I will continue poking at it until I see something I missed. 

**Logs and console output**
Can you let me know what logs would be needed? I am not aware of anything besides the console dump. 

**Environment (please complete the following information):**
Windows 10
Nordic toolchain-installed zephyr

**Additional context**
N/A

nordicjm, 2022-11-02:
Cannot replicate this in zephyr or using latest main of ncs, it is possible that this is a bug in ncs (downstream, not upstream in zephyr). I would suggest trying this with the latest version of ncs (main). Closing as this is not a zephyr issue.

closed 2022-11-02 by nordicjm

chalecampb, 2022-11-02:
Hi,

I had just pulled the latest version of NCS (2.1.0) last week. It looks like there is a 2.1.1 but it doesn't list this as a fix or anything. I can give it a shot. 

Can you let me know the steps to perform this on the CLI outside of VS code, to reproduce in the same environment?

thedjnK, 2022-11-02:
> Can you let me know the steps to perform this on the CLI outside of VS code, to reproduce in the same environment?

Sure, instructions are at https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/nrf/gs_installing.html be sure to use the `main` one

chalecampb, 2022-11-05:
Hi,

Just letting you know, if I do make menuconfig on the CLI, BME280 is enabled. If I try to flash, there is an issue with JLink which I already submitted to nordic. So I think you are right, this is something in the IDE.

Thanks,
-Alex