zephyrproject-rtos/zephyr #51893: LSM303dlhc sensor example not compiling for nRF52840

CodingGhost, 2022-11-02:
the LSM303dlhc example does not compile for the nRF52840 target. tried on 2.7.1 and latest main, on 2.7.1 I get
devicetree.h:305:40: error: 'DT_N_INST_0_st_lsm303dlhc_magn_P_label' undeclared (first use in this function)
and in main I get:
util.h:78:55: error: size of unnamed array is negative

**To Reproduce**
follow this guide and use this command:  west build -b nrf52840dk_nrf52840 samples/sensor/lsm303dlhc
https://docs.zephyrproject.org/latest/samples/sensor/lsm303dlhc/README.html

**Impact**
showstopper currently, as Im unable to implement the sensor.

**Environment (please complete the following information):**
 - OS: Fedora 15
 - Toolchain west

henrikbrixandersen, 2022-11-02:
Well, the `nrf52840dk_nrf52840` board does not have an onboard LSM303dlhc sensor. If you added one you need to make a devicetree overlay describing how this sensor is connected to the board.

CodingGhost, 2022-11-02:
oh. that makes sense. I will try that. I guess I can take the devicetree of the stm board as example?

henrikbrixandersen, 2022-11-02:
> I guess I can take the devicetree of the stm board as example?

Sure.