zephyrproject-rtos/zephyr #51912: Add textual LCD display driver support

thedjnK, 2022-11-02:
**Is your feature request related to a problem? Please describe.**
At present, zephyr has a display driver which is very much suited to graphical LCDs whereby e.g. images can be displayed, however it is lacking for support of textual-based displays, e.g. HD44780-based displays, or those with an SPI/I2C interface that support limited commands, essentially for moving a cursor, writing text or clearing the whole display. It would be beneficial to have a display class that supports the functionality of these basic textual displays, either as a new dedicated interface or by expanding the existing display interface to support the required functionality

**Describe the solution you'd like**
The driver would need to have the following functionality:

- Display init
- Display de-init
- Display on/off
- Display cursor on/off
- Display blinking position on/off
- Display write text
- Display set cursor position (row, column)
- Display get details (width and height in characters, character set)
- Display clear

Optional nice to have functionality:

- Display set brightness
- Display custom character set
- Display custom command (some displays have a bit more functionality e.g. setting font size, enabling automatic text scroll, etc., this could allow sending those commands without needing new functions for each of them)

The display configuration would be set via device tree, the type of driver and interface would be needed alongside the number of characters for width, height and likely the character set that the display supports

**Describe alternatives you've considered**
Use the existing display driver, but the required functionality is just not present

nuertey, 2022-11-04:
@thedjnK 
FYI, see my implementation for an LCD16x2 here: https://github.com/nuertey/ZephyrOS-LCD16x2

I tested it on the component and it is working. Pictures here (scroll down): https://github.com/nuertey/ZephyrOS-WeatherStation
Regards
Nuertey

nuertey, 2022-11-04:
And I forgot to mention--and this is most relevant to your feature request--that I based my implementation on the official sample: 

https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/drivers/lcd_hd44780

nuertey, 2022-11-04:
Understood.

gmarull, 2022-11-08:
One misc driver that could likely be converted to such class: https://github.com/zephyrproject-rtos/zephyr/tree/main/drivers/misc/grove_lcd_rgb

lawrencek52, 2022-11-23:
I think what @thedjnK is really looking for is a package like ncurses above the character display. ncurses has been around a longtime and a subset of the basic ncurses APIs are probably reasonable for this project.

The list of functions @thedjnK  suggested did not include:
- query functions:
    - the size of the display in rows and columns. 
    - the display capabilities, can it draw in colour? invert the character? flash the cursor, underline or block cursor, etc.
 
I don't think the ncurses underlying termcap database is required. The Zephyr equivalent is probably a set of macros and defines that describe each display's capabilities and allow unusable features to be compiled out making the code as small as possible. Possibly the terminal capabilities could be embedded in the device tree entry for the device.
 
ncurses also includes a whole set of input functions, but these are probably not necessary for a character display. ncurses is also capable of handling multiple windows this would be useful on a system with multiple character displays.

thedjnK, 2022-11-23:
> I think what @thedjnK is really looking for is a package like ncurses above the character display

I'm not.

lawrencek52, 2022-11-23:
> > I think what @thedjnK is really looking for is a package like ncurses above the character display
> 
> I'm not.

Every function you asked for is covered by the ncurses API.

henrikbrixandersen, 2022-11-23:
> Every function you asked for is covered by the ncurses API.

This has nothing to do with ncurses.

It's a driver API for a specific class of display hardware: text based displays (as opposed to graphical/frame buffer displays, which are already supported by Zephyr).

ncurses is a library for creating text-based user interfaces (TUIs). Now, one might want something along the lines of ncurses in Zephyr, but that would still need to operate on top of a low-level hardware driver API like the one suggested here - and it is out of scope for this PR.

closed 2023-05-26 by carlescufi