zephyrproject-rtos/zephyr #51931: Failing unit test re. missing PERIPHERAL_ISO_SUPPORT KConfig selection

erbr-ot, 2022-11-03:
Compilation of tests/bluetooth/init/test_ctlr_peripheral_iso is failing after modification to ull_peripheral_iso.c. 
In the output (https://github.com/zephyrproject-rtos/zephyr/actions/runs/3368517840/jobs/5587194250) is seen:
warning: BT_ISO_PERIPHERAL (defined at subsys/bluetooth/Kconfig.iso:19) was assigned the value 'y'
but got the value 'n'. Check these unsatisfied dependencies: (!BT_CTLR ||
BT_CTLR_PERIPHERAL_ISO_SUPPORT) (=n). See
http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_BT_ISO_PERIPHERAL and/or look up
BT_ISO_PERIPHERAL in the menuconfig/guiconfig interface. The Application Development Primer, Setting
Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be helpful
too.

This is similar to issues re. BSIM tests noticed and handled originally in PR:
https://github.com/zephyrproject-rtos/zephyr/pull/46583

I confirmed (by temp revert of my change in ull_peripheral_iso.c) that this issue has been present in the above mentioned test since this PR was introduced, but was not noticed, because build did not fail, until change introduced by https://github.com/zephyrproject-rtos/zephyr/pull/51807.

erbr-ot, 2022-11-03:
I'm not sure how is the correct way to resolve this, ie hot to correctly add the correct KConf selection to the build of this test

Thalley, 2022-11-03:
I guess you can apply the same solution as we did for the BSIM tests, right? 

Alternatively, since we are adding more testing for this, perhaps it's time we start thinking about being able to set `BT_CTLR_PERIPHERAL_ISO_SUPPORT=y` as an experimental feature in some way?

erbr-ot, 2022-11-03:
> I guess you can apply the same solution as we did for the BSIM tests, right?
Well yeah, the thing is just that I don't know how to ;-)

erbr-ot, 2022-11-03:
Turned out the actual failure was due to missing macros, and not resolved by resolving the KConfig, but rather by adding macros, that then 'magically' appeared by a different PR;-)
I believe this 'issue' is still present, however it has no impact on unit test

closed 2022-11-03 by erbr-ot