zephyrproject-rtos/zephyr #51951: Zephyr Getting Started steps fail with Python v3.11

DerekSnell, 2022-11-03:
**Describe the bug**
Not all required Python modules install now that Python v3.11 is released.  When following Zephyr's Getting Started steps on Windows 10, installing Python using Chocolately will install the latest v3.11.  The following modules in Zephyr's Requirements.txt files will fail to install:
- gcovr
- pyocd

**To Reproduce**
- follow the [Zephyr Getting Started](https://docs.zephyrproject.org/latest/develop/getting_started/index.html) steps for Windows 10
- on a fresh system, the step `choco install python` will install the latest Python v3.11
- This step will fail, caused by the modules listed above:
`pip install -r %HOMEPATH%\zephyrproject\zephyr\scripts\requirements.txt`

**Impact**
Severe! New users to Zephyr will not be able to complete the install steps.  Also, `west flash` commands fail without pyOCD installed, even when using a different runner like JLink.

**Environment:**
Tried on Windows 10

**Additional context**
Issue #51780 fixed module windows_curses for Python v3.11.  But even after this fix, the other modules listed above still fail.

stephanosio, 2022-11-03:
This is definitely not a high priority bug. There is nothing forcing users to use Python 3.11.

Technically, this is not really a Zephyr bug eiterh, it is just that `cmsis-pack-manager` and `lxml` packages have not provided the wheels for Python 3.11 yet, so this issue should really be reported to those projects.

stephanosio, 2022-11-03:
Also FYI, you can install Python 3.10 using Chocolatey as follows:

```
choco install python --version 3.10
```

DerekSnell, 2022-11-03:
Thanks, @stephanosio ,
I understand your feedback, and while these are not Zephyr bugs, they become issues when following the Zephyr installation steps.   

One solution is to update the Getting Started steps with the workaround you shared, at least until Python v3.11 is fully supported:
```
choco install python --version 3.10
```

evgeniy-paltsev, 2022-11-04:
> pip install -r %HOMEPATH%\zephyrproject\zephyr\scripts\requirements.txt

Frankly speaking this thing fails almost every time for me :)

Failures depends on which python version and Zephyr version we have but In almost all cases I have to either disable something from requirements.txt or just ignore that some packages were not installed due to some issues (if I don't need to use them).

It's really unstable.

ClaCodes, 2022-11-11:
> Thanks, @stephanosio , I understand your feedback, and while these are not Zephyr bugs, they become issues when following the Zephyr installation steps.
>
> One solution is to update the Getting [...]

Had the same issue and agree with the suggestion to patch the "getting started"-guide for now.

dleach02, 2022-12-13:
Changing the priority to get some attention. This issue particularly hits new users doing fresh installs. It is very frustrating if the installation doesn't work as documented so if we know there is a version problem we need to document or fix it in our flow.

stephanosio, 2022-12-13:
> if we know there is a version problem we need to document or fix it in our flow.

As noted in https://github.com/zephyrproject-rtos/zephyr/issues/51951#issuecomment-1302660301, this is not really a problem on the Zephyr side . The problem is that some third-party Python packages that Zephyr depends on have not released the "wheels" (aka. prebuilt binary packages) for Python 3.11 yet.

The only way for us to "fix" the problem is to either:

1. host our own PyPI package index that contains the up-to-date wheels for all Zephyr Python dependencies (impractical due to enormous maintenance overhead)
2. include a Python installation with all required packages in Zephyr SDK (not really sure if we want to go down that path)
3. designate a specific Python version to install in the Getting Started Guide.

cc @carlescufi @mbolivar-nordic

mbolivar-nordic, 2022-12-14:
> cc @mbolivar-nordic

4. add a note in the GSG recommending to avoid 3.11 on windows for the time being

carlescufi, 2022-12-27:
> > cc @mbolivar-nordic
> 
> 4. add a note in the GSG recommending to avoid 3.11 on windows for the time being

This sounds like a very good idea.

gmarull, 2022-12-28:
Looks like this issue should be fixed, cmsis-pack-manager supports Python 3.11 now. https://github.com/pyocd/cmsis-pack-manager/releases/tag/v0.5.1

gmarull, 2022-12-28:
I've just gone through the GSG on a Windows 11 machine with Python 3.11 and everything works fine now. Closing.

closed 2022-12-28 by gmarull

DerekSnell, 2022-12-29:
> I've just gone through the GSG on a Windows 11 machine with Python 3.11 and everything works fine now. Closing.

Thank you, @gmarull ,
I also went through the GSG on a Windows 10 machine with Python 3.11, and it worked great.  Thanks