zephyrproject-rtos/zephyr #51977: newlib integration: _unlink isn't mapped to unlink

tenllado, 2022-11-04:
**Describe the bug**

Using the newlib libc in Zephyr, the ISO C remove function calls the _unlink_r funtion which requires the _unlink function, but this function is missing and the compilation fails at link time.

I have tested this while trying to create a lua module for Zephyr. The lua/loslib.c file includes a call to remove:

static int os_remove (lua_State *L) {
  const char *filename = luaL_checkstring(L, 1);
  return luaL_fileresult(L, remove(filename) == 0, filename);
}

I compiled the project with the zephyr-sdk-0.15.0 for a disco_l475_iot1 board, and get the following error:

/ld.bfd: ... zephyr-sdk-0.15.0/arm-zephyr-eabi/arm-zephyr-eabi/lib/thumb/v7e-m/nofp/libc.a(lib_a-unlinkr.o): in function `_unlink_r':
unlinkr.c:(.text._unlink_r+0xc): undefined reference to `_unlink'
collect2: error: ld returned 1 exit status

The project compiles fine if I provide the following implementation for _unlink:

int _unlink(const char *pathname)
{
	return fs_unlink(pathname);
}

But this is probably not what I should be doing.

**To Reproduce**

I prepared a hello_world example available at: https://github.com/tenllado/zephyr_hello_world_lua

The README.md file contains instructions to compile it as a west workspace. The project includes a file src/missing_stubs.c the definition of _unlink (and also _times, which is also missing in Zephyr). If you comment the _unlink definition you will obtain the error mentioned above.

**Expected behavior**
Zephyr should be including a definition of the _unlink function for the newlib libc library.

**Impact**
This absence of this function breaks the implementation of the ISO C remove function. It is possible to replace it by fs_unlink.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: zephyr-sdk-0.15.0
 - Commit SHA: sha for the zephyr project is 8db84cfa6a954e466d3c7d416afeb1d638df6c69

cfriedt, 2023-07-08:
I'm currently unable to reproduce this with the testsuite, which does exercise both newlib and `unlink()`
```shell
twister -T tests/posix/fs/
```

@tenllado - you may need to that you build with some of the configurations defined in `tests/posix/fs/prj.conf` such as
```
CONFIG_FILE_SYSTEM=y
CONFIG_LOG=y
CONFIG_FAT_FILESYSTEM_ELM=y
CONFIG_DISK_DRIVER_RAM=y
CONFIG_DISK_RAM_VOLUME_SIZE=80
CONFIG_POSIX_API=y
CONFIG_POSIX_FS=y
CONFIG_ZTEST=y
CONFIG_ZTEST_NEW_API=y
CONFIG_MAIN_STACK_SIZE=4096
CONFIG_EVENTFD=n
```

Feel free to reopen if you think the issue is still not resolved.

closed 2023-07-08 by cfriedt

GeekOffTheStreet, 2024-04-18:
FWIW, I also hit this.  I can build the tests/posix/fs using my prj.conf and same west command, but hit this in my project:
```
[133/139] Linking C executable zephyr/zephyr_pre0.elf
FAILED: zephyr/zephyr_pre0.elf zephyr/zephyr_pre0.map /workspace/repos/gr716-zephyr-apps/gbec/test/zephyr-build/zephyr/zephyr_pre0.map 
: && ccache /opt/zephyrproject/zephyr-sdk-0.16.5/sparc-zephyr-elf/bin/sparc-zephyr-elf-gcc  -gdwarf-4 zephyr/CMakeFiles/zephyr_pre0.dir/misc/empty_file.c.obj -o zephyr/zephyr_pre0.elf  zephyr/CMakeFiles/offsets.dir/./arch/sparc/core/offsets/offsets.c.obj  -fuse-ld=bfd  -T  zephyr/linker_zephyr_pre0.cmd  -Wl,-Map=/workspace/repos/gr716-zephyr-apps/gbec/test/zephyr-build/zephyr/zephyr_pre0.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/sparc/core/libarch__sparc__core.a  zephyr/lib/libc/newlib/liblib__libc__newlib.a  zephyr/lib/libc/common/liblib__libc__common.a  zephyr/lib/posix/liblib__posix.a  zephyr/subsys/fs/libsubsys__fs.a  zephyr/drivers/interrupt_controller/libdrivers__interrupt_controller.a  zephyr/drivers/console/libdrivers__console.a  zephyr/drivers/serial/libdrivers__serial.a  zephyr/drivers/timer/libdrivers__timer.a  modules/fatfs/libmodules__fatfs.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  -L"/opt/zephyrproject/zephyr-sdk-0.16.5/sparc-zephyr-elf/bin/../lib/gcc/sparc-zephyr-elf/12.2.0/leon3/soft"  -L/workspace/repos/gr716-zephyr-apps/gbec/test/zephyr-build/zephyr  -lgcc  ../../build/libunit_tests.a  ../../build/libgtest.a  zephyr/arch/common/libisr_tables.a  -msoft-float  -mcpu=leon3  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn  -lstdc++  -Wl,-no-pie  -lm  -Wl,-lc  -L"/opt/zephyrproject/zephyr-sdk-0.16.5/sparc-zephyr-elf/sparc-zephyr-elf"/lib/leon3/soft  -u_printf_float  -Wl,-lgcc  -lc && cd /workspace/repos/gr716-zephyr-apps/gbec/test/zephyr-build/zephyr && /usr/bin/cmake -E true
/opt/zephyrproject/zephyr-sdk-0.16.5/sparc-zephyr-elf/bin/../lib/gcc/sparc-zephyr-elf/12.2.0/../../../../sparc-zephyr-elf/bin/ld.bfd: /opt/zephyrproject/zephyr-sdk-0.16.5/sparc-zephyr-elf/sparc-zephyr-elf/lib/leon3/soft/libc.a(lib_a-unlinkr.o): in function `_unlink_r':
unlinkr.c:(.text._unlink_r+0xc): undefined reference to `_unlink'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

Git-qisen, 2024-12-04:
> FWIW, I also hit this. I can build the tests/posix/fs using my prj.conf and same west command, but hit this in my project:FWIW，我也击中了这个。我可以使用我的 prj.conf 和相同的 west 命令构建测试/posix/fs，但在我的项目中点击这个：
> 
> ```
> [133/139] Linking C executable zephyr/zephyr_pre0.elf
> FAILED: zephyr/zephyr_pre0.elf zephyr/zephyr_pre0.map /workspace/repos/gr716-zephyr-apps/gbec/test/zephyr-build/zephyr/zephyr_pre0.map 
> : && ccache /opt/zephyrproject/zephyr-sdk-0.16.5/sparc-zephyr-elf/bin/sparc-zephyr-elf-gcc  -gdwarf-4 zephyr/CMakeFiles/zephyr_pre0.dir/misc/empty_file.c.obj -o zephyr/zephyr_pre0.elf  zephyr/CMakeFiles/offsets.dir/./arch/sparc/core/offsets/offsets.c.obj  -fuse-ld=bfd  -T  zephyr/linker_zephyr_pre0.cmd  -Wl,-Map=/workspace/repos/gr716-zephyr-apps/gbec/test/zephyr-build/zephyr/zephyr_pre0.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/sparc/core/libarch__sparc__core.a  zephyr/lib/libc/newlib/liblib__libc__newlib.a  zephyr/lib/libc/common/liblib__libc__common.a  zephyr/lib/posix/liblib__posix.a  zephyr/subsys/fs/libsubsys__fs.a  zephyr/drivers/interrupt_controller/libdrivers__interrupt_controller.a  zephyr/drivers/console/libdrivers__console.a  zephyr/drivers/serial/libdrivers__serial.a  zephyr/drivers/timer/libdrivers__timer.a  modules/fatfs/libmodules__fatfs.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  -L"/opt/zephyrproject/zephyr-sdk-0.16.5/sparc-zephyr-elf/bin/../lib/gcc/sparc-zephyr-elf/12.2.0/leon3/soft"  -L/workspace/repos/gr716-zephyr-apps/gbec/test/zephyr-build/zephyr  -lgcc  ../../build/libunit_tests.a  ../../build/libgtest.a  zephyr/arch/common/libisr_tables.a  -msoft-float  -mcpu=leon3  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn  -lstdc++  -Wl,-no-pie  -lm  -Wl,-lc  -L"/opt/zephyrproject/zephyr-sdk-0.16.5/sparc-zephyr-elf/sparc-zephyr-elf"/lib/leon3/soft  -u_printf_float  -Wl,-lgcc  -lc && cd /workspace/repos/gr716-zephyr-apps/gbec/test/zephyr-build/zephyr && /usr/bin/cmake -E true
> /opt/zephyrproject/zephyr-sdk-0.16.5/sparc-zephyr-elf/bin/../lib/gcc/sparc-zephyr-elf/12.2.0/../../../../sparc-zephyr-elf/bin/ld.bfd: /opt/zephyrproject/zephyr-sdk-0.16.5/sparc-zephyr-elf/sparc-zephyr-elf/lib/leon3/soft/libc.a(lib_a-unlinkr.o): in function `_unlink_r':
> unlinkr.c:(.text._unlink_r+0xc): undefined reference to `_unlink'
> collect2: error: ld returned 1 exit status
> ninja: build stopped: subcommand failed.
> ```

I also encountered the same problem, have you solved it？

Git-qisen, 2024-12-04:
@GeekOffTheStreet I also encountered the same problem like u posted, have you solved it？