zephyrproject-rtos/zephyr #51978: newlib integration: _times isn't mapped to _times_r

tenllado, 2022-11-04:
**Describe the bug**

Using the newlib libc in Zephyr, the clock() and/or time() functions call the _times_r function which requires the _times function to be supplied, but this function is missing and the compilation fails at link time.

I have tested this while trying to create a lua module for Zephyr. Lua uses both the clock() and the time() function and the compilation fails with the following error:

... zephyr-sdk-0.15.0/arm-zephyr-eabi/arm-zephyr-eabi/lib/thumb/v7e-m/nofp/libc.a(lib_a-timesr.o): in function `_times_r':
timesr.c:(.text._times_r+0x2): undefined reference to `_times'
collect2: error: ld returned 1 exit status

**To Reproduce**

I prepared a hello_world example available at: https://github.com/tenllado/zephyr_hello_world_lua

The README.md file contains instructions to compile it as a west workspace. The project includes a file src/missing_stubs.c a definition of _times (and also _unlink, which is also missing in Zephyr). If you comment the _times definition you will obtain the error mentioned above.

**Expected behavior**
Zephyr should be including a definition of the _times function for the newlib libc library.

**Impact**
The absence of this function prevents the full support of the lua library for the zephyr project.

**Environment (please complete the following information):**
- OS: Linux
 - Toolchain: zephyr-sdk-0.15.0
 - Commit SHA: sha for the zephyr project is 8db84cfa6a954e466d3c7d416afeb1d638df6c69

stephanosio, 2022-11-05:
`times` is a POSIX function and it is currently not available in the Zephyr POSIX subsystem.

stephanosio, 2022-11-05:
Converted to enhancement since newlib cannot map a POSIX function to the POSIX subsystem when it is not implemented by the POSIX subsystem.

stephanosio, 2022-11-05:
I did a brief investigation and the ISO C function `clock()` does depend on the POSIX `times()`, which is indeed a problem. I will provide a fix such that `clock()` does not depend on `times()`.

p.s. ISO C `time()` does not require POSIX `times()`; instead, it requires `_gettimeofday()`, which is already provided by the newlib libc-hooks. That aside, the libc hook implementation depending on the POSIX `gettimeofday()` should not be the case.

cfriedt, 2022-11-22:
Just added the modules label, as this blocks the Lua module

cfriedt, 2023-05-02:
Just dropping this link here for myself mainly. 
https://github.com/zephyrproject-rtos/newlib-cygwin/blob/zephyr-newlib-3.3.0/newlib/libc/time/clock.c

I'm trying to recall how I tied this into `sysconf()` - perhaps it's not as related as I thought.

cfriedt, 2023-06-01:
@stephanosio - do you still want to provide a fix for this, or would #57800 work?

Also, not sure if we should consider this an enhancement or a bug.

tenllado, 2023-06-01:
I think that [#57800](https://github.com/zephyrproject-rtos/zephyr/pull/57800) would fix the problems I am having.