zephyrproject-rtos/zephyr #51984: Central misses connection parameters update request from a connected peripheral

ahmedmoheb-nordic, 2022-11-05:
## Introduction

The use case includes a Central that connects to 4 peripherals

### Problem description

The Central uses the following procedure to connect to the peripherals
1 - Central starts a scanning procedure
2 - When a device is found, central attempt to connect to it 
3 - When the 'connected' callback is received, a signal is set to start a new scanning procedure from the main thread
4 - When a peripheral is connected, it sends a connection parameters update request

Connecting to the 4 peripherals is working as expected, except that one of the connection update request is missed at the central side.

Even after increasing the following configuration parameters, issue is still exist.

Please check the following draft PR which shows the case and can be executed using BabbleSim
https://github.com/zephyrproject-rtos/zephyr/pull/51983

A shell script can be found under ~/zephyrproject/zephyr/tests/bluetooth/bsim_bt/bsim_test_conn_stress/tests_scripts/conn_stress.sh

ahmedmoheb-nordic, 2022-11-05:
@jori-nordic @alwa-nordic please check this

cvinayak, 2022-11-07:
Zephyr Controller supports LE Connection Parameter Request Procedure, which generates LL_REJECT_EXT_IND PDU with error code Unsupported Link Layer Parameter Value (0x20) when the Controller is busy with another Connection Update procedure.

I have shared a PR #52006 that I use locally for testing LE Connection Parameter Request across multiple simultaneous to reschedule connections of similar connection interval in a non-overlapping schedule.

PR #52006 (2022-11-08): Bluetooth: Host: Add auto conn param update retry count and timeout
Add implementation to retry the automatic peripheral
perferred connection parameter request with a configurable
retry countdown and retry back-off timeout.

Fixes #51984.
closed 2022-11-08 by carlescufi