zephyrproject-rtos/zephyr #52018: Change QEMU Cortex M3 machine to more flexible one

yashi, 2022-11-07:
**Is this request related to a missing driver support for a particular hardware platform, SoC or board? Please describe.**

Currently we are using `ti_lm3s6965` as `qemu_cortex_m3` but it seems [it's not flexible for SRAM size](https://github.com/zephyrproject-rtos/zephyr/pull/40437).  For emulating Cortex M3 with QEMU, I believe most of users don't really care which board we are running but does care the fact that it's a Cortex M3 machine and is capable running users application in emulation mode.

qemu_cortex_m3 can't really support running network applications due to limited SRAM size.  With the current origin/main:

```shell
$ git describe
zephyr-v3.2.0-482-g2666702cd10
$ west build -t run -b qemu_cortex_m3 zephyr/samples/net/sockets/echo_server -- -DOVERLAY_CONFIG=overlay-qemu_cortex_m3_eth.conf
    :
    :
[291/302] Linking C executable zephyr/zephyr_pre0.elf
FAILED: zephyr/zephyr_pre0.elf zephyr/zephyr_pre0.map /home/yashi/src/zephyr-project/build/zephyr/zephyr_pre0.map 
: && ccache /home/yashi/zephyr-sdk-0.15.1/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc  -gdwarf-4 zephyr/CMakeFiles/zephyr_pre0.dir/misc/empty_file.c.obj -o zephyr/zephyr_pre0.elf  zephyr/CMakeFiles/offsets.dir/./arch/arm/core/offsets/offsets.c.obj  -fuse-ld=bfd  -Wl,-T  zephyr/linker_zephyr_pre0.cmd  -Wl,-Map=/home/yashi/src/zephyr-project/build/zephyr/zephyr_pre0.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/arm/core/aarch32/libarch__arm__core__aarch32.a  zephyr/arch/arch/arm/core/aarch32/cortex_m/libarch__arm__core__aarch32__cortex_m.a  zephyr/lib/libc/minimal/liblib__libc__minimal.a  zephyr/soc/arm/ti_lm3s6965/libsoc__arm__ti_lm3s6965.a  zephyr/subsys/net/libsubsys__net.a  zephyr/subsys/net/l2/ethernet/libsubsys__net__l2__ethernet.a  zephyr/subsys/net/ip/libsubsys__net__ip.a  zephyr/subsys/net/lib/config/libsubsys__net__lib__config.a  zephyr/subsys/net/lib/conn_mgr/libsubsys__net__lib__conn_mgr.a  zephyr/subsys/random/libsubsys__random.a  zephyr/drivers/console/libdrivers__console.a  zephyr/drivers/serial/libdrivers__serial.a  zephyr/drivers/ethernet/libdrivers__ethernet.a  zephyr/drivers/timer/libdrivers__timer.a  modules/mbedtls/libmodules__mbedtls.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  -L"/home/yashi/zephyr-sdk-0.15.1/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.1.0/thumb/v7-m/nofp"  -L/home/yashi/src/zephyr-project/build/zephyr  -lgcc  zephyr/arch/common/libisr_tables.a  -no-pie  -mcpu=cortex-m3  -mthumb  -mabi=aapcs  -mfp16-format=ieee  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn && cd /home/yashi/src/zephyr-project/build/zephyr && /usr/bin/cmake -E echo
/home/yashi/zephyr-sdk-0.15.1/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.1.0/../../../../arm-zephyr-eabi/bin/ld.bfd: zephyr/zephyr_pre0.elf section `noinit' will not fit in region `RAM'
/home/yashi/zephyr-sdk-0.15.1/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.1.0/../../../../arm-zephyr-eabi/bin/ld.bfd: region `RAM' overflowed by 18208 bytes
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /home/yashi/src/zephyr-project/build --target run
```

What do you think if I propose changing to another board with much bigger SRAM size or flexible with `CONFIG_SRAM_SIZE` or [DT](https://github.com/zephyrproject-rtos/zephyr/issues/17624).

**Describe why you are asking for this support?**

I'd like qemu_cortex_m3 to be more flexible machine so that we can all use it with our own requirements.

stephanosio, 2022-11-07:
FWIW, we already have `mps2_an385`, which is a Cortex-M3 board that runs on QEMU with 4MiB "flash" and RAM.

yashi, 2022-11-07:
Indeed, the following command just works with `zeth` described at https://docs.zephyrproject.org/3.2.0/connectivity/networking/qemu_eth_setup.html
```
$ west build -t run -b mps2_an385 zephyr/samples/net/sockets/echo_server -- -DOVERLAY_CONFIG=overlay-smsc911x.conf
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /home/yashi/src/zephyr-project/zephyr/samples/net/sockets/echo_server
-- Found Python3: /usr/bin/python3.10 (found suitable exact version "3.10.8") found components: Interpreter 
-- Cache files will be written to: /home/yashi/.cache/zephyr
-- Zephyr version: 3.2.99 (/home/yashi/src/zephyr-project/zephyr)
-- Found west (found suitable version "0.14.0", minimum required is "0.7.1")
-- Board: mps2_an385
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.15.1 (/home/yashi/zephyr-sdk-0.15.1)
-- Found toolchain: zephyr 0.15.1 (/home/yashi/zephyr-sdk-0.15.1)
-- Found Dtc: /usr/bin/dtc (found suitable version "1.6.1", minimum required is "1.4.6") 
-- Found BOARD.dts: /home/yashi/src/zephyr-project/zephyr/boards/arm/mps2_an385/mps2_an385.dts
-- Generated zephyr.dts: /home/yashi/src/zephyr-project/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /home/yashi/src/zephyr-project/build/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: /home/yashi/src/zephyr-project/build/zephyr/dts.cmake

warning: QEMU_ICOUNT_SHIFT (defined at boards/Kconfig:67) was assigned the value '7' but got the
value ''. Check these unsatisfied dependencies: QEMU_ICOUNT (=n). See
http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_QEMU_ICOUNT_SHIFT and/or look up
QEMU_ICOUNT_SHIFT in the menuconfig/guiconfig interface. The Application Development Primer, Setting
Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be helpful
too.

Parsing /home/yashi/src/zephyr-project/zephyr/samples/net/sockets/echo_server/Kconfig
Loaded configuration '/home/yashi/src/zephyr-project/zephyr/boards/arm/mps2_an385/mps2_an385_defconfig'
Merged configuration '/home/yashi/src/zephyr-project/zephyr/samples/net/sockets/echo_server/prj.conf'
Merged configuration '/home/yashi/src/zephyr-project/zephyr/samples/net/sockets/echo_server/overlay-smsc911x.conf'
Configuration saved to '/home/yashi/src/zephyr-project/build/zephyr/.config'
Kconfig header saved to '/home/yashi/src/zephyr-project/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 12.1.0
-- The CXX compiler identification is GNU 12.1.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/yashi/zephyr-sdk-0.15.1/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
CMake Warning at /home/yashi/src/zephyr-project/zephyr/subsys/random/CMakeLists.txt:11 (message):
  

      Warning: CONFIG_TEST_RANDOM_GENERATOR is not a truly random generator.
      This capability is not secure and it is provided for testing purposes only.
      Use it carefully.

CMake Warning at /home/yashi/src/zephyr-project/zephyr/CMakeLists.txt:809 (message):
  No SOURCES given to Zephyr library: drivers__net

  Excluding target from build.

-- Configuring done
-- Generating done
-- Build files have been written to: /home/yashi/src/zephyr-project/build
-- west build: running target run
[1/309] Preparing syscall dependency handling

[8/309] Generating include/generated/version.h
-- Zephyr version: 3.2.99 (/home/yashi/src/zephyr-project/zephyr), build: zephyr-v3.2.0-482-g2666702cd105
[298/309] Linking C executable zephyr/zephyr_pre0.elf

[302/309] Linking C executable zephyr/zephyr_pre1.elf

[308/309] Linking C executable zephyr/zephyr.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:      150640 B         4 MB      3.59%
             RAM:       85200 B         4 MB      2.03%
        IDT_LIST:          0 GB         2 KB      0.00%
[308/309] To exit from QEMU enter: 'CTRL+a, x'[QEMU] CPU: cortex-m3

uart:~$ *** Booting Zephyr OS build zephyr-v3.2.0-482-g2666702cd105  ***
[00:00:04.340,000] <err> eth_smsc911x: net_recv_data: -22
[00:00:04.350,000] <inf> net_config: Initializing network
[00:00:04.350,000] <inf> net_config: IPv4 address: 192.0.2.1
[00:00:04.460,000] <inf> net_config: IPv6 address: 2001:db8::1
[00:00:04.460,000] <inf> net_config: IPv6 address: 2001:db8::1
[00:00:04.460,000] <inf> net_echo_server_sample: Run echo server
[00:00:04.460,000] <inf> net_echo_server_sample: Network connected
[00:00:04.460,000] <inf> net_echo_server_sample: Starting...
[00:00:04.460,000] <inf> net_echo_server_sample: Waiting for TCP connection on port 4242 (IPv6)...
[00:00:04.460,000] <inf> net_echo_server_sample: Waiting for TCP connection on port 4242 (IPv4)...
[00:00:04.460,000] <inf> net_echo_server_sample: Waiting for UDP packets on port 4242 (IPv6)...
[00:00:04.460,000] <inf> net_echo_server_sample: Waiting for UDP packets on port 4242 (IPv4)...
[00:01:29.870,000] <dbg> net_echo_server_sample: process_udp: (udp4): UDP (IPv4): Received and replied with 1 bytes
[00:01:29.870,000] <dbg> net_echo_server_sample: process_udp: (udp4): UDP (IPv4): Received and replied with 6 bytes
[00:01:29.870,000] <dbg> net_echo_server_sample: process_udp: (udp4): UDP (IPv4): Received and replied with 4 bytes
[00:01:29.870,000] <dbg> net_echo_server_sample: process_udp: (udp4): UDP (IPv4): Received and replied with 26 bytes
[00:01:29.870,000] <dbg> net_echo_server_sample: process_udp: (udp4): UDP (IPv4): Received and replied with 1280 bytes
[00:01:29.870,000] <dbg> net_echo_server_sample: process_udp: (udp4): UDP (IPv4): Received and replied with 1 bytes
[00:01:29.870,000] <dbg> net_echo_server_sample: process_udp: (udp4): UDP (IPv4): Received and replied with 256 bytes
uart:~$ 
```

nashif, 2024-02-10:
based on the comments above, seems to be addressed already.

closed 2024-02-10 by nashif