zephyrproject-rtos/zephyr #52055: Bluetooth: Controller: Broadcast scheduling issues

Casper-Bonde-Bose, 2022-11-08:
### Describtion
For a multi BIS BIG there seems to be issues with the scheduling. It is unknown if all issues are related, but I managed to reproduce the issues using the broadcast_audio_sample application with a modified stream_sent_cb where every 40th PDU is not written to the controler. I know this is not normal behaviour for an audio streaming application, but at high CPU load it can occur that the host is too late writing a PDU to the controller. (The real application where these issues were observed only occationally saw the issues).
To reproduce the issue there is a patch to the sample-application and controller config below.

**A general issue, that occurs regardless of the data flow issue:**
[image]
The AUX_ADV_IND causes a full BIG event to be dropped. This is even though the allowed interval is 1000ms to 1200ms - hence 200ms jitter can be allowed, and that there is plenty of space between the BIG events. In the ideal case they should be placed in between the BIG events, and if there is not enough room, they should corrupt only the number of BIS subevents that the duration of the ADV requires - in this case that would be one bis sub-event - and preferably the last in the BIG event as that will favor receivers that do not need retransmissions. (This might be a different issue than the following - hence I'm fine if it is moved to a separate case, but reporting as I suspect it is all related)

**When the data flow issues present them self I observe:**
**Real application:**
[image]

**Reproduction:**
Deliberately discarding every 40th PDU on the host:
[image]
Deliberately discarding every 4ooth PDU on the host:
[image]

- Some BIS events do not carry any data (in the reproduction this is way more sever than in the real application). This seems to cause ISO host credits to leak, and eventually the host stops sending data to the controller.
- The empty BIS events seems to trigger and offset of the BIG timing - note how there is no room for payload after the empty event in the circle. Timing references are hard in an air-trace, but the last image shows that it might actually be the empty PDUs that are timed wrong.
- A non Zephyr controller is able to sync to the broadcast, but not receive any data. For example the nrfconnect/sdk-nrf nrf5340_audio application modified to receive the 4 broadcasts. If I replace the Zephyr controller with the controller image from nrfconnect/sdk-nrf, then it is able to receive the data - which points to a bug in the Zephyr controller.

If none of the data from the host is discarded the issue do not seem to occur for the reproduction, but in the real application it does - also a BIS might not have data for one or more BIS events - it will depend on the application and type of data broadcasted - for example air-port announcements might only have rare occurrences of actual data - hence the controller should be able to handle that.

### Setup:
Commit 261025a86ee0b03e3efd5a6279a13489b218a954
+ patches below to configure for multiple BISes.
HW: nRF5340 DK version 2.0.0
**BIG config:** 
BIS Count: 4
RTN: 0
SDU: 251
Interval: 7.5ms
Controller build:
`west build -b nrf5340dk_nrf5340_cpunet -p -d build/nrf_rpmsg_ctrl zephyr/samples/bluetooth/hci_rpmsg -- -DCONF_FILE=nrf5340_cpunet_iso_broadcast-bt_ll_sw_split.conf`
Host build:
`west build -b nrf5340dk_nrf5340_cpuapp -p -d build/nrf_broadcast_test zephyr/samples/bluetooth/broadcast_audio_source`

**Observations**
To allow 251 bytes of ISO payload over the HCI interface the CONFIG_BT_CTLR_ISO_TX_BUFFER_SIZE needs to be set to a value higher than 251 bytes (as that is the value returned in read_buffer_size_v2). but the name suggests the HCI buffer limit is also used as ISO TX limit, which seems odd and wrong - the MAX ISO TX PDU is 251 bytes.
If this value is set to 251 it forces the host to fragment the 251 byte ISO data payload and thereby use 2 credits per ISO PDU.

**Expected behavior**
- AUX_ADV_IND should not block a full BIG event
- Rare missing data for a single bis should not cause no data for multiple BISes (and potentially leak host credits)
- Rare missing data from the host should not cause timing issues for when in the time the BIS events occur.
- The Broadcast should be receivable by non-Zephyr controllers.

**Impact**
When the issue occurs a non-zephyr controller cannot receive any of the broadcasted data (not tried to use a Zephyr based controller).
When the issue occurs the host stops transmitting data after a while.
When the issue occurs a large amount of the data is not transmitted causing very bad audio quality on the receiver side.

**Logs and console output**
Ellisys trace from cloud: {5fb3d6c7-8f83-4759-b235-8328ae77f01d} - this is discarding every 400th PDU from the host as done in the patch.

**Patches**
`From 8c247bd92b783169e1baf526c1bedee70aeae0c4 Mon Sep 17 00:00:00 2001
From: Casper Bonde <casper_bonde@bose.com>
Date: Tue, 8 Nov 2022 10:28:20 +0100
Subject: [PATCH 1/2] Bluetooth: Controller: Add support for 4 BIS BIGs and 251
 byte PDUs

Allow up to 4 BISes
Allow transmission of 251 bytes of pauload without fragmentation
over the HCI interface.

---
 .../nrf5340_cpunet_iso_broadcast-bt_ll_sw_split.conf       | 7 ++++---
 1 file changed, 4 insertions(+), 3 deletions(-)
[... 87 lines cut ...]
 static struct bt_audio_stream streams[CONFIG_BT_AUDIO_BROADCAST_SRC_STREAM_COUNT];
 static struct bt_audio_broadcast_source *broadcast_source;
 
@@ -55,13 +69,19 @@ static void stream_sent_cb(struct bt_audio_stream *stream)
 		return;
 	}
 
+	if ((sent_cnt % 400) == 2) {
+		seq_num++;
+		sent_cnt++;
+		return;
+	}
+
 	buf = net_buf_alloc(&tx_pool, K_FOREVER);
 	if (buf == NULL) {
 		printk("Could not allocate buffer when sending on %p\n",
 		       stream);
 		return;
 	}
-
+	mock_data[0] = sent_cnt & 0xff;
 	net_buf_reserve(buf, BT_ISO_CHAN_SEND_RESERVE);
 	net_buf_add_mem(buf, mock_data, preset_16_2_1.qos.sdu);
 	ret = bt_audio_stream_send(stream, buf, seq_num++,
`

cvinayak, 2022-11-09:
> To allow 251 bytes of ISO payload over the HCI interface the CONFIG_BT_CTLR_ISO_TX_BUFFER_SIZE needs to be set to a value higher than 251 bytes (as that is the value returned in [...]
If this value is set to 251 it forces the host to fragment the 251 byte ISO data payload and thereby use 2 credits per ISO PDU.

@Thalley Host should not be fragmenting a 251 byte SDU when LE Read Buffer Size v2 returns 251 bytes, is there a bug in use of `send_frag`?

@Casper-Bonde-Bose I had to fix your update to the sample to have a catchup of `stream_sent_cb` each time you skip the call. This is not Controller scheduling issue, but the sample not catching up to keep the call to `stream_sent_cb` sustained. Find attached the simulation trace of the changes I made to have non-overlapping advertising intervals, which are here: #52093

[broadcast_audio_source_gh_52055_202211090747.zip](https://github.com/zephyrproject-rtos/zephyr/files/9968153/broadcast_audio_source_gh_52055_202211090747.zip)

Thalley, 2022-11-09:
> @Thalley Host should not be fragmenting a 251 byte SDU when LE Read Buffer Size v2 returns 251 bytes, is there a bug in use of `send_frag`?

Could this be a case where the 4-8 octet header in the ISO packet is not taken into account? 

If you want to send 251 octets of ISO data, then the ISO packet sent to the controller will be 255-259 octets, depending on whether the packet contains the timestamp or not.

Casper-Bonde-Bose, 2022-11-10:
> Could this be a case where the 4-8 octet header in the ISO packet is not taken into account?
>
> If you want to send 251 octets of ISO data, then the ISO packet sent to the controller will be 255-259 [...]

Exactly - this is what I was trying to communicate. The payload is 251 octets, but header and optional timestamp means the HCI buffer needs to be larger than 251. But perhaps the values from read_buffer_size_v2 is used wrong in the host - but how will the host know what the HCI limit is - or can it just assume that it is OK to send 259 bytes ISO packets over HCI (251 bytes of Data_Load)?

So there seem to be a bug in send_frag (or if send_frag is still shared between ACL and ISO there might be s difference in buffer limits?)

Casper-Bonde-Bose, 2022-11-10:
> @Casper-Bonde-Bose I had to fix your update to the sample to have a catchup of `stream_sent_cb` each time you skip the call. This is not Controller scheduling issue, but the sample not catching up [...]

@cvinayak The point is not to catch up. If there is no data for a specific BIS event to be sent one cannot simply catch up. I understand that in the sample application data flow will stop after 18*400 iterations - but in the real application the transmission of new data is not triggered by the call-back, but availability of new data from the audio subsystem.

Thalley, 2022-11-10:
> Exactly - this is what I was trying to communicate. The payload is 251 octets, but header and optional timestamp means the HCI buffer needs to be larger than 251. But perhaps the values from [...]

The value read from `read_buffer_size_v2` does not contain the HCI ISO Data packet header. `5.4.5 HCI ISO Data packets` defines the header to only be the 4 octets that contain this:

[image]

The timestamp (4 octets) are part of the ISO_Data_load. 

So if the buffer read is 251, then you can send 251 octets of ISO_Data_Load, which means 251 octets for the ISO_SDU_Fragment if not supplying a timestamp, or 247 ISO_SDU_Fragment with timestamp. 

If you are sending 251 octets of SDU data _with_ 4 octets timestamp (total of 255) and it fragments it, then it is working as intended. 
If you are sending 251 octets of SDU data _without_ the timestamp and it fragments it, then there is a bug in the host.

cvinayak, 2022-11-10:
> > @Casper-Bonde-Bose I had to fix your update to the sample to have a catchup of `stream_sent_cb` each time you skip the call. This is not Controller scheduling issue, but the sample not catching up to keep the call to `stream_sent_cb` sustained. Find attached the simulation trace of the changes I made to have non-overlapping advertising intervals, which are here: #52093
> 
> @cvinayak The point is not to catch up. If there is no data for a specific BIS event to be sent one cannot simply catch up. I understand that in the sample application data flow will stop after 18*400 iterations - but in the real application the transmission of new data is not triggered by the call-back, but availability of new data from the audio subsystem. (this will of cause be changed at some point when I have a mechanism to do ASRC of the audio to match BT controller timing)

I am referring to the lost `stream` pointer in calls to `stream_sent_cb` when the callback does not call `bt_audio_stream_send`. I.e. everytime `bt_audio_stream_send` is skipped will reduce the successive number of `stream_send_cb` callback returning.

The controller is not using sequence number yet, hence controller does not delay SDUs based on sequence number. Integration with ISOAL is work in progress #52034.

cvinayak, 2022-11-10:
> If you are sending 247 octets of SDU data _without_ the timestamp and it fragments it, then there is a bug in the host.

@Thalley Application is sending 251 bytes, hence the controller needs to return `ISO_Data_Packet_Length`  as 255 (for without `Time_Stamp`) to support 251 byte `ISO_SDU_Fragment` in the `ISO_Data_Load` field. Host is correct in its implementation, controller needs Kconfig to be used correctly to return 259 so that it is capable of receiving `ISO_Data_Load` with and without the `Time_Stamp` from host.

Thalley, 2022-11-10:
> @Thalley Application is sending 251 bytes, hence the controller needs to return `ISO_Data_Packet_Length` as 255 (for without `Time_Stamp`) to support 251 byte `ISO_SDU_Fragment` in the [...]

Agreed. 

I also think we can improve the documentation in the host w.r.t. this; it isn't clear that a ISO buffer size of e.g. 251 means that "only" 243/247 (depending on timestamp) SDU size can be used.

Casper-Bonde-Bose, 2022-11-14:
@cvinayak  Just to be sure the main point of this issue is not missed. The adjustment of the ISO Buffer was just needed to support the "real application" that requires 251 byte PDU's - I do not suspect that adjustment is causing the issues.
The main issues (the 3 bullet points) are that the Zephyr controller seems to skip some TX events, use wrong timing and seems to transmit on wrong channels.

cvinayak, 2022-11-15:
Attached is the simulation trace of your patch, and I do not observe any BIG event timing issue. Overlapping advertising event are present (#52093 has changes to adv interval to avoid overlaps, I am not using it to simulate your behavior) and due to bug in your patch to skip calls to  `bt_audio_stream_send` which looses generation of `stream_sent_cb`, eventually Controller gets no new ISO data to transmit from your patched sample.

[broadcast_audio_source_gh_52055_202211140348.zip](https://github.com/zephyrproject-rtos/zephyr/files/10008694/broadcast_audio_source_gh_52055_202211140348.zip)

Casper-Bonde-Bose, 2022-11-15:
@cvinayak I'll be happy to send you the air-trace - I just don't see a way to attach an air-trace to a github issue. I have tried to reach you on Discord to share it that way, but it seems I need you to accept a friend request to be able to send you a message with the trace?
The issue happens already after 2 BIG events - hence you don't need it to run for long - and I think your fix in my patch will hide the issue - as I mentioned the "real" application never runs out of data to send, but it seems that the trigger of the bug is when some BIG events do not have data to transmit for all BISes - and your fix will ensure there is always data, just that data arrives to the controller a few us later.
Also unless you validate frequency and time in the simulator you will not see the issue.

Casper-Bonde-Bose, 2022-11-15:
I just noticed that you attached an air-trace in your comment, I was looking for a button to upload a file - you just need to drag the file into this window :-)
Here the trace:
[Zephyr_crtl_broadcast_issue.zip](https://github.com/zephyrproject-rtos/zephyr/files/10011036/Zephyr_crtl_broadcast_issue.zip)

I see your trace do not have the same "scrambled" PDU order as when I run it - which is kind of strange. Would it be possible for you to share your hci_rpmsg build for me to test if we build the BT-controller FW differently or the host application differently?

cvinayak, 2023-02-16:
A `hci_rpmsg` sample can be build upstream by supplying `-DCONF_FILE=nrf5340_cpunet_iso-bt_ll_sw_split.conf` to add ISO support.

Closing the issue. Can be re-opened if original issue is not solved in latest upstream (v3.3.0-rc3)

closed 2023-02-16 by Thalley