zephyrproject-rtos/zephyr #52056: Bluetooth: Missing LL data length update callback on Central and Peripheral sides

ahmedmoheb-nordic, 2022-11-08:
## Introduction

Updating LL data length fails

### Problem description

After establishing a connection between a Central and a Peripheral, trying to update the LL data length fails.
Also, the update callback isn't called.

Please check the following draft PR which shows the case and can be executed using BabbleSim
https://github.com/zephyrproject-rtos/zephyr/pull/51985

A shell script can be found under ~/zephyrproject/zephyr/tests/bluetooth/bsim_bt/bsim_test_conn_stress/tests_scripts/conn_stress.sh

ahmedmoheb-nordic, 2022-11-08:
@jori-nordic @alwa-nordic please check this

ahmedmoheb-nordic, 2022-11-15:
No longer exist after rebasing.

It seemed to be related to the interval between receiving the "connected" callback and starting the next scanning procedure.
Before rebasing:

- The connection update request from the second connected peripheral was missed which is related to this issue https://github.com/zephyrproject-rtos/zephyr/issues/51984
- By introducing a delay between receiving the "connected" callback after connecting to the first peripheral and starting the scanning procedure, the central got the connection update request and LL data length update callback was received as expected

closed 2022-11-15 by ahmedmoheb-nordic

jori-nordic, 2022-11-17:
not reproducible anymore after @ahmedmoheb-nordic upgraded from ubuntu 20.04 as host system to ubuntu 22.04.
Probably this is due to the different compiler/linker in the previous system.

ahmedmoheb-nordic, 2022-11-19:
I tried to reproduce the issue with a VM running Ubuntu 20.04, but I couldn't.

Adding a delay before starting the second scanning procedure allows the 2 expected connection parameters update requests from the connected peripherals to be received, then the LL data length update callbacks are received normally after initiating a LL data length update.

Also, removing the waiting condition and without a delay before the second scanning procedure, an error code is returned upon requesting to update the LL data length.

Please find the attached screen shots showing the 2 cases.

[image]

[image]